earlephilhower/ESP8266Audio #2: Attempting to use the library results in an error.

tretyakovsa, 2017-09-28:
In file included from C:\Users\user\Documents\Arduino\libraries\ESP8266Audio\src\libflac\private/bitreader.h:38:0,

                 from C:\Users\user\Documents\Arduino\libraries\ESP8266Audio\src\libflac\bitreader.c:40:

C:\Users\user\Documents\Arduino\libraries\ESP8266Audio\src\libflac\private/cpu.h:39:20: fatal error: config.h: No such file or directory

 #include <config.h>

                    ^

compilation terminated.

earlephilhower, 2017-09-28:
Sorry to hear that.  What IDE and ESP8266 sdk are you using?  I'm unable to reproduce it on GIT head plugin and 1.8.1 Arduino IDE.

It's a simple path problem.  There is a config.h file in the src/libflac directory that should be picked up for you but isn't.  A simple replacement of
#include <config.h>
with
#include "config.h"
(i.e. greater than/less than to quotes) may fix it for you, but I'd like to make it fail on my end and test a fix for committing myself...

Thx
-EFP3

earlephilhower, 2017-09-28:
I was just able to reproduce the failure.  Looks to only happen under Windows, not Linux (my main dev. environment).

A simple change from <config.h> to "../config.h" in the subdirectory headers seems to have fixed it under Windows without causing any trouble under Linux, so I've pushed this new patch to github.

Can you give it a try and if it works please close this issue?

If not, send your output again with some more info on your setup.

Thx
-EFP3

tretyakovsa, 2017-09-29:
Everything worked. Thank you for your work.

commit 663be686 (2018-05-24): Fix 8->16 bit output conversion overflow error

The I2S outputs were converting samples in the buffer passed in
from u8->s16 on every call.  When the I2S was full, however, the
sample would not get pushed and the output function would be
called at a later time with the same buffer...which was already
at 16 bits.  It'd be converted again and turned into a high
frequency noise on the output.

Now only modify a copy of the input sample in I2S and I2SNoDAC.
Fixes the conversion problem competely.

Fixes #2
commit 0987b12e (2018-05-24): Fix 8->16 bit output conversion overflow error

The I2S outputs were converting samples in the buffer passed in
from u8->s16 on every call.  When the I2S was full, however, the
sample would not get pushed and the output function would be
called at a later time with the same buffer...which was already
at 16 bits.  It'd be converted again and turned into a high
frequency noise on the output.

Now only modify a copy of the input sample in I2S and I2SNoDAC.
Fixes the conversion problem competely.

Fixes #2
closed 2017-09-29 by tretyakovsa