earlephilhower/ESP8266Audio #3: SD error compilation.

renat2985, 2017-09-29:
Please tell me what I do wrong? Thanks for ESP8266Audio!

[image]

earlephilhower, 2017-09-29:
Howdy,

Looks like you have a custom SD library file in ~/Arduino/libraries/SD.  That won't work at all with the ESP8266 ide/sdk.  Remove that and let Arduino use the ESP8266 specific SD card libs and I think you'll be OK.

-EFP3

renat2985, 2017-09-29:
Thanks. Work.
Do you have plans to make a simple example with a web interface where can add new internet radio? :)

earlephilhower, 2017-09-29:
No plans now, sorry.  I'm always happy to get pull requests, though. :)

earlephilhower, 2017-10-17:
Closing as this was resolved.

closed 2017-10-17 by earlephilhower