earlephilhower/ESP8266Audio #4: Web interface

renat2985, 2017-09-30:
I'm trying to implement a web interface with my friend. But your library refuses to work correctly. We will be very happy if you can help me.

https://github.com/tretyakovsa/ESP_WIFI_RADIO

earlephilhower, 2017-09-30:
What exactly is the problem?  Basically you set it up with the httpsreader then in your main loop() you need to pump the mp3->loop() call one or more times to have it generate samples for the audio output.

earlephilhower, 2017-09-30:
Also, just for sanity, have you been able to run one of the simple PROGMEM examples like PlayWAVFromPROGMEM and hear it?  If not, there's some HW issue and you'll need to check your wiring...

earlephilhower, 2017-12-01:
Closing this as I think you're consolidating everything in bug #13.

closed 2017-12-01 by earlephilhower