earlephilhower/ESP8266Audio #5: Slow/Choppy audio with a mp3 of  bitrate of 32Kbps

Uksa007, 2017-10-15:
Hi,

I’m using an ESP8266 - Wemos D1 mini with your awesome library, ESP8266Audio.

If I use the sample pno-cs.mp3 included in the example program“PlayMP3FromSPIFFS” it play perfectly fine in fact it sounds amazing for abusing the I2S. I’m using 160Mhz for the CPU and a transistor to drive a small speaker, as per the instructions.

If I use a the attached force.mp3 it is very choppy and sounds slow.
force.mp3 is from google translator, I don’t have any control over the format.
I thought the issue maybe bitrate/sample rate related, but the force.mp3 is significantly lower. 
Pno-cs.mp3  Bitrate: 128kbps  Samplerate48Khz
Force.mp3  Bitrate: 32 Kbps  Samplerate24Khz
[force.zip](https://github.com/earlephilhower/ESP8266Audio/files/1385893/force.zip)

I also have choppy playback using the example StreamMP3FromHTTP.ino but I suspect that may be network delay, as there doesn’t’ seem to be any buffer?

If I use the exampe StreamMP3FromHTTP to stream just a single mp3 file from a server(my ultimate goal), it seems to crash at the end of the file. I suspect it's not detecting the end of the file. 

Is there an issue with the force.mp3 that causes the slow/choppy playback?
Is there anything I can do to the Arduino code/library to make it play correctly, as I can’t change the mp3 from google.
How can I stop the StreamMP3FromHTTP from crashing at the end of the file?

Thanks for your time.

earlephilhower, 2017-10-16:
Howdy,

In these cases what I like to do is use the SPIFFSWAV output and then load it up in audacity to see if it's a matter of timeouts or actually the data itself going weird.  The WAV file doesn't care if it's made at 1/2 realtime so if it's OK then it's a speed problem.  If it's weird, then it's a logic bug.

In this case, take a look at the waveform for force.mp3:
[image]

So it looks like libmad isn't generating 1/2 of each MP3 frame in this case.  I remember during porting there was a special, low-bitrate codepath which I didn't go into much detail on, but I would bet it's a bug I introduced there.  I probably won't be able to look at it until this weekend, so don't hold your breath, but it's in the queue.

As for the streaming HTTP, you could be quite right.  I only ever tried it on a single web radio stream (i.e. no EOF) and don't use it myself as the ESP8266 is right on the hairy edge of its performance actually doing any web request processing while playing audio.  The web stack also seems quite variable in performance, and I've had the sample run flawlessly for hours and then on the next reboot it comes up and stutters even though there's been no internet connectivity or code changes.

The code for the streaming is quite simple, though, so I bet you could take a look and find the bug easily since you have working testcase already.  I'm always happy to get pull requests!  Basically when you find the socket is closed, just return EOF in the read() method and everything else should just work..

Uksa007, 2017-10-17:
Hi,

Thanks for the reply, and adding the issue to the queue.

With regards to the crashing at the EOF, I have been looking at the library, in particular in the "AudioFileSourceHTTPStream.cpp". I'm not really that well versed in how it all works, but from what I can see the following seems to be main reading function. 
I have tried a number of different thing, but none seems to have worked apart from causing it to crash sooner. I'm not sure where the "int len" comes from, maybe it's not correct for a file?

Can you please let me know if I'm looking in the right place, and point me in the right direction.

`uint32_t AudioFileSourceHTTPStream::read(void *data, uint32_t len)
{
  if (!http.connected()) return 0;

  WiFiClient *stream = http.getStreamPtr();
  while (stream->available() < (int)len) yield();

  size_t avail = stream->available();
  if (!avail) return 0;
  if (avail < len) len = avail;

  int read = stream->readBytes(reinterpret_cast<uint8_t*>(data), len);
  pos += read;
  return read;
}`

Thanks!

earlephilhower, 2017-10-17:
Yes, that's probably it.  You'll note that nowhere is there a check that the WiFiClient isn't closed or disconnected.  My thinking is that copious checks for those kinds of conditions while waiting for data and returning it would get to the bottom of things fast...

Uksa007, 2017-10-17:
I would have thought that "if (!http.connected()) return 0;" would check that it was connected and if not abort.
I'm struggling to find a way to kill it, return 0 doesn't seem to do it, and it only want a int, so you can't return EOF. 

Thoughts?

earlephilhower, 2017-10-17:
Ah, I didn't see that.  I shouldn't do any code reviews before coffee in the AM.

There may still be a race issue there (where it's connected on the check, but during the wait to get data it barfs), so I'd throw in some more checking in the while() loop.

To signal "EOF" just do a "return 0;"  The upper level stack assumes readlen==0 ==> EOF (like STDIO).

One more thing that'd be helpful.  If you can do a "ESP8266 Exception Decode" (in the Arduino menus) on the crash stack dump, that would be very useful in pinpointing the exact crash spot.  It might *not* be where we're looking here.

Thx!
-EFP3

Uksa007, 2017-10-17:
Hi,
 I added some more checking but still no change. 

There doesn't seem to be stack trace on the serial monitor, it just stops doing anything in void loop.

eg my loop looks  like:
`
void loop()
{
  yield();
  if (mp3->isRunning()) {
    if (!mp3->loop()) mp3->stop();
  } else {
    Serial.printf("MP3 done\n");
    delay(1000);
  }
  Serial.print("I'm still alive: ");
  Serial.println(millis());
}
`
The serial output stops at EOF, no crash dump?

I'm still alive: 10337
I'm still alive: 10338
I'm still alive: 10340
I'm still alive: 10342

Uksa007, 2017-10-17:
So some more investigating, and I think it's this while loop that seems to be offender, causing it to yield and do nothing else, if I remove it, it detects the EOF file, but seems unstable and does crash with illegal instruction. But some progess:

  while (stream->available() < (int)len) yield();
  size_t avail = stream->available();
  if (!avail) return 0;
  if (avail < len) len = avail;

earlephilhower, 2017-10-18:
Because the ESP8266 uses a cooperative multitasking setup, and a watchdog timer, you need the yield() in any busy-wait loops to a) allow the WiFi stack/etc. to run and actually handle the network and get data to be used, and b) feed the WDT or else the system will hard reset.

len is the amount of data needed for the next MP3 frame.  It's not the problem here as there is no general EOF indicator available or needed in MP3 streams.

The issue is probably here:
> while (stream->available() < (int)len) yield();

Can you add a check that it's still connected to the while()?
> while ((stream->available() < (int)len) && (stream->connected()) ) yield();

That should make the while() abort if the http connection is closed and the available should be 0 at that point and the rest of the logic should just work(tm).

Uksa007, 2017-10-18:
Hi,

I had tried something similar in the part, your while loop doesn't make any difference, so I guess that means it thinks the stream is still connected?

I also tried the following with no result:
  while ((stream->available() < (int)len) && (http.connected()) ) yield();

Whatever is holding the connection open, it's not wanting to close it!

earlephilhower, 2017-10-18:
Well, it's not pretty but there's a simple way around that...timeouts:

 uint32_t AudioFileSourceHTTPStream::read(void *data, uint32_t len)
{
if (!http.connected()) return 0;

WiFiClient *stream = http.getStreamPtr();
**int start = millis();**
while ((stream->available() < (int)len) && **(millis()-start <500)**) yield();
...

That'll give a ~500ms maximum wait for data, then it'll return EOF...

earlephilhower, 2017-10-22:
Your MP3 playback is fixed with the latest commit.  Basically I was running the synthesizers 2x as long as they should have been for your low bitrate mono file.  A simple fix, and all contained in my own new wrapper code.

Uksa007, 2017-10-22:
Hi,
Thanks, I can confirm that the playback quality works great now! 
Thanks for fixing it!

I'm still struggle with the http connection not closing. 

The post above re 500ms time out, sort of works, but often last parts of the playback are lost, and it still often does not return to the main loop!

Is there a more robust approach to solving this issue?

Thanks again!

earlephilhower, 2017-10-22:
Check the push this AM.  I was able to reproduce the hang on a local file and fix it.

Basically, the web server sends the length in a HTTP header and doesn't close the TCP connection because it may be reused.  So you need to track the read length vs. the expected sending length. 

The commit now has a 500ms or reat-to-expected-size timeout.  Only about 10 lines to fix, and seems to work for both files and streaming just fine.

earlephilhower, 2017-11-27:
Closing as no response and it's fixed as far as me and the other users can tell.

closed 2017-11-27 by earlephilhower