earlephilhower/ESP8266Audio #7: Compilation Issue

armSeb, 2017-11-01:
Hello, 

Thanks for this library which I try to run on NodeMCU 1.0.

I intalled the library but when I try to compile an example, I get the following message : 

`In file included from /Users/armseb/Documents/Arduino/libraries/ESP8266Audio-master/src/AudioFileSourceFastROMFS.cpp:21:0:
/Users/armseb/Documents/Arduino/libraries/ESP8266Audio-master/src/AudioFileSourceFastROMFS.h:25:30: fatal error: ESP8266FastROMFS.h: No such file or directory
 #include <ESP8266FastROMFS.h>`

I can't find any information about this header file, where can I find it ?

Thank you.

Sebastian.

armSeb, 2017-11-01:
Sorry, I didn't noticed that the missing library is here https://github.com/earlephilhower/ESP8266FastROMFS

earlephilhower, 2017-11-01:
Thanks for the heads-up!  I thought the IDE used mkdepends to figure out which files to compile, but it seems like it just compiles all files in the directory.  I'll need to think of a way to split these up without ending up with a dozen different libraries, ugh.

armSeb, 2017-11-03:
Hi, 

Thanks for your feedback. Just a little notice on the documentation seems to be perfect to begin.
Did you plan to add some SPI RAM support to increase the buffer size ? 
I am using the NoDAC output but I bought an I2S DAC to have stereo and better output quality.
To finish, what do you think about adding a capacitor between the transistor base and audio output to avoid some DC current which can damage the speaker ?

Regards, 

Sebastian

earlephilhower, 2017-11-03:
Howdy,

I'll throw a note in the readme for now.  Unfortunately this inclusion is a bigger problem:  it includes the SD library as well which has a ~1000byte static member that takes up precious space even if you don't instantiate any SD interface.  

For the cap, wouldn't it go from the collector to the speaker and not base?  If you're going for safety, a base resistor may also make sense to ensure the base current can't get high enough to cause damage if the speaker shorts out or something.

As for SPI RAM, I wasn't thinking about it but you're the 2nd person this week to suggest it.  The problem is that in the nonos-sdk that SRAM was only used as a compressed source buffer (i.e. to hold TCP packets to smooth out streaming performance).  Most clicks we're getting today are, I think, related to the ESP8266 doing a WiFi process for longer than the internal buffers (11ms @44KHz) can cover before the app can refill them.  In this case, the external SRAM isn't going to help because the app can't gain control to decompress it and reload the DMA buffers.

Input buffering is simple, to do, though, and only requires probably 40 lines in a new AudioInputFileBuffer class that takes a AudioInputFile as source, reads it and stuffs it into SPI RAM, and on the output just reads out of the FIFO.  The AudioOutputBuffered is almost what you'd need, actually, just on the inputs and w/SPI RAM instead of ESP RAM usage.

What I want to do is increase the I2S buffer (i.e. decoded) and that's more complicated and needs ISR changes.  For now, I've just submitted a PR ( https://github.com/esp8266/Arduino/pull/3790 )  to allow increasing the internal RAM buffers.  That'll help a little, but there's not that much free RAM to use so it's limited.  Rewriting the ISR to use the SPI (if that's even possible) to pull directly into DMA buffers would be the real answer.

-EFP3

armSeb, 2017-11-03:
Yep the SD library conflicts with an other library which I had on my local lib directory.

Regarding the capacitor, I suggest it to avoid a continuous transistor conduction. If for some mysterious reason the output gpio stays to high state, this avoid the speaker to be damaged by DC current.

earlephilhower, 2017-11-03:
Ah, mybad.  You'd have the cap between I2S out and the base which would let you use electrolytic as well since no -ve inductance-caused spikes.

I'll make a note on that when I update the readme as a start.  For real field deployments you'd want it in, but for most Arduino folks the fewer the # of parts, the better.  At least I'm not suggesting driving the speaker directly off of the GPIO pin like on the PICs. :)

armSeb, 2017-11-05:
I drive successfully an earphone (32ohm) with a simple 1K resistor and 10nF capacitor as filter. 
The sound is good but I get cracks and noise at regular rythm when streaming MP3 from HTTP.
When playing a MOD file, the NoDAC output is saturated.

It's an amazing and powerful device, thanks a lot for your work, I coupled it with Wifi Manager library to be able to change the wifi credentials at runtime.

earlephilhower, 2017-11-05:
Cool!  At 32ohms I can imagine you're saturating.  You could hook up to the 3.3V source instead of the 5V source to lower the amplitude of the wave coming out (assuming you're using the transistor...if not, ignore this obviously).

I just started adding a 2nd level buffer class to AudioFileSource which would allow extra RAM to buffer SPIFFS, IP, etc. which may help here.  I'm not 100% convinced it will help because of the whole 1-threaded nature of the ESP8266 Arduino.  The idea is while you're decoding the MP3 you're also buffering in the next packet into RAM...but if there's only 1 execution context you can't really do it "behind the scenes."  We'll see, it should be simple enough to write, in any case.

armSeb, 2017-11-06:
For an earphone I don't need the transistor, as the output level is high. The resistor acts as low pass filter (with the capacitor) and as amplitude limiter (to protect earphone, ear and ESP8266 output).

Regarding the extra ram, the goal is to avoid sound losses caused by network jitter.

armSeb, 2017-11-13:
Hi!

I just received my I2S DAC, and tried id, the sound quality is awesome! 
The DAC is a PCM5102.

earlephilhower, 2017-11-14:
Great.  They're worth the price and it takes much less CPU to send out the raw samples than to do the SW DAC (~8 instructions per bit IIRC) which can help with stuttering in some cases.

I've also just pushed a new class, AudioFileSourceBuffer() which uses system RAM to double-buffer any input source.  You instantiate it with the real source (i.e. a new HTTPStreamFile) and then pass in this buffer object to the mp3 decoder.  I'll do a sample (and more testing) probably this weekend.

armSeb, 2017-11-14:
Hello,

Great to hear that, I'll test it soon.

I have to investigate on my side regarding the MP3 decoding, as I get only mono instead of stereo. Is there some special configuration to have to be able to decode stereo ?

earlephilhower, 2017-11-14:
There's nothing but a stereo mode in I2S.  You have to shift out left then right samples (hence the LRCLOCK).

Hate to be silly, but have you checked the obvious things like the headphone being fully engaged, any config pins on the DAC (or if it's I2C controlled only you'll have to hack that yourself), and that it's actually a stereo DAC, and of course that it's a stereo MP3?  The adafruit DAC+amp is only mono, FWIW, and has control pins that let you select left, right, or sum channels.

As a simple sanity test you can run a stereo WAV test instead of MP3.

armSeb, 2017-11-14:
I have only the choice between the i2s or left justified mode. 

This is a stereo mp3, headphones are plugged correctly. I'll try with a wav file.

Here is the DAC datasheet : http://www.ti.com/lit/ds/symlink/pcm5102.pdf

armSeb, 2017-11-14:
I'm afraid to say you that my raspberry pi is too old for that (first version), I have to take another pi.

So, I decided to hack the AudioOutputI2SDAC.cpp file. If replace sample[RIGHTCHANNEL] or sample[LEFTCHANNEL] with 0x0000, I get no sound on right earphone (or left earphone). So, the DAC works fine. I guess the channels are not correctly separated somewhere.

armSeb, 2017-11-14:
I tried the following in AudioGeneratorMP3.cpp : 

`// If we're here, we have one decoded frame and sent 0 or more samples out
  if (samplePtr < synth.pcm.length) {
    sample[AudioOutput::LEFTCHANNEL ] = synth.pcm.samples[0][samplePtr];
    sample[AudioOutput::RIGHTCHANNEL] = 0x0000; //synth.pcm.samples[1][samplePtr];
    samplePtr++;
  } else {
    samplePtr = 0;

    switch ( mad_synth_frame_onens(&synth, &frame, nsCount++) ) {
        case MAD_FLOW_STOP:
        case MAD_FLOW_BREAK:
          return false; // Either way we're done
        default:
          break; // Do nothing
    }
    // for IGNORE and CONTINUE, just play what we have now
    sample[AudioOutput::LEFTCHANNEL ] = synth.pcm.samples[0][samplePtr];
    sample[AudioOutput::RIGHTCHANNEL] = 0x0000; //synth.pcm.samples[1][samplePtr];
    samplePtr++;
  }
`

Instead of getting muted right channel, I get mono output. channels seems to be mixed.

earlephilhower, 2017-11-14:
Good debugging.  You found an uninitialized class variable bug in the I2S output routine!  Since it's uninitialized, it could work one time and fail the next....

I've pushed the 1-line fix (init the MONO flag to false), so pull the latest and try again (after removing your forced-0, of course) and I think it should get you going.

armSeb, 2017-11-14:
Yeah this is exactly what I found, after commenting the stereo to mono routine in the AudioOutputI2SDAC.cpp I got real stereo :)

armSeb, 2017-11-14:
So, I added the AudioFileBuffer class, with 8192 bytes to begin, it seems to work fine.

earlephilhower, 2017-11-15:
Was there any change in the amount of hiccups while playing back web streams?  I've not run it in person, only remotely, so can't actually hear any differences...

armSeb, 2017-11-15:
It seems to be better, but I can't add more than 8192bytes buffer without stability issues. I guess the free RAM amount is very limited.

I purchased some 23LC1024 chips to help to implement external SPIRAM buffer. With 128KByte buffer it may run smoother.

Another interesting points are the following ; 

- Ability to have some software mixer which allows to adjust the output volume. I modified the AudioOutputI2SDAC in this way: 
`  uint32_t s32 = ((sample[RIGHTCHANNEL]/5)<<16) | ((sample[LEFTCHANNEL]/5) & 0xffff);`
The goal is to make the output volume controllable by GPIO interrupts.

- Song title (i.e ICY meta) decoding ( for displaying them on oled display for example).

armSeb, 2017-11-15:
I confirm that with only 4096 Bytes buffer, I avoid many hiccups, the program is leff sensitive to network jitter. But the buffer seems consumes some CPU. Without buffer I was able to listen 192kbps webradio, with the buffer I can hear some audio glitches. With lower bitrate (128kbps) It's perfect.

earlephilhower, 2017-11-15:
The only real spot to optimize is this snippet, where instead of doing some tedious math I crawl across the buffered data.  Everything else is pretty much a straight read-to-buffer or read-to-destination which is as fast as it's going to get:

---snip---
  // **Naively** pull from buffer until we've got none left or we've satisfied the request
  uint8_t *ptr = reinterpret_cast<uint8_t*>(data);
  while (len && length) {
    ...
  }
---snip---

Also, there are some commented out Serial.printfs.  You may want to uncomment this one:
> //  Serial.printf("read %d from buffer, %d avail in buff, %d remaining for request\n", bytes, length, len);

and see how much of that 4K buffer really is being used.  My quick testing didn't see > 700-800 bytes buffered at any time due to the WiFi stack not supplying data faster.

earlephilhower, 2017-11-15:
Oh, and gain control was one thing I wanted to do as well.  I'll be implementing a fixed point scaling since it's pretty cheap on this CPU whereas something like "float x = 0.95 * y;" can take a millisecond...

armSeb, 2017-11-15:
> read 1881 from buffer, 2214 avail in buff, 0 remaining for request
> read 1881 from buffer, 2214 avail in buff, 0 remaining for request
> read 1881 from buffer, 2214 avail in buff, 0 remaining for request
> read 1880 from buffer, 2215 avail in buff, 0 remaining for request
> read 1881 from buffer, 2214 avail in buff, 0 remaining for request
> read 1881 from buffer, 2214 avail in buff, 0 remaining for request
> read 1881 from buffer, 2214 avail in buff, 0 remaining for request
> read 1881 from buffer, 2215 avail in buff, 0 remaining for request
> read 1880 from buffer, 2215 avail in buff, 0 remaining for request
> read 1881 from buffer, 2214 avail in buff, 0 remaining for request
> read 1881 from buffer, 2214 avail in buff, 0 remaining for request
> read 1881 from buffer, 2214 avail in buff, 0 remaining for request
> read 1881 from buffer, 2214 avail in buff, 0 remaining for request
> read 1881 from buffer, 2214 avail in buff, 0 remaining for request
> read 1880 from buffer, 2215 avail in buff, 0 remaining for request
> read 1881 from buffer, 2214 avail in buff, 0 remaining for request
> read 1881 from buffer, 2214 avail in buff, 0 remaining for request
> read 1881 from buffer, 2214 avail in buff, 0 remaining for request
> read 1881 from buffer, 2214 avail in buff, 0 remaining for request
> read 1880 from buffer, 2215 avail in buff, 0 remaining for request
> read 1881 from buffer, 2214 avail in buff, 0 remaining for request
> read 1881 from buffer, 2214 avail in buff, 0 remaining for request
> read 1881 from buffer, 2214 avail in buff, 0 remaining for request
> read 1881 from buffer, 2214 avail in buff, 0 remaining for request
> read 1880 from buffer, 2215 avail in buff, 0 remaining for request
> read 1881 from buffer, 2214 avail in buff, 0 remaining for request
> read 1881 from buffer, 2214 avail in buff, 0 remaining for request
> read 1881 from buffer, 2214 avail in buff, 0 remaining for request
> read 1881 from buffer, 2214 avail in buff, 0 remaining for request
> read 1881 from buffer, 2214 avail in buff, 0 remaining for request
> read 1880 from buffer, 2215 avail in buff, 0 remaining for request
> read 1881 from buffer, 2214 avail in buff, 0 remaining for request
> read 1881 from buffer, 2214 avail in buff, 0 remaining for request

earlephilhower, 2017-11-16:
Re: the speed drop you were seeing, I've put up a less naive buffer manager which devolves into 1 or 2 memcpys per read instead of a byte-by-byte iteration.  Give it a go when you have the chance.

armSeb, 2017-11-16:
I'll try as soon as possible, thanks !

For now, I make some tests from my local network. I am playing a 320kbps mp3 file, without buffer function, which I serve from my mac with "python -m SimpleHTTPServer".

Audio file with ID3 version 2.4.0, contains:MPEG ADTS, layer III, v1, 320 kbps, 44.1 kHz, Stereo
It can be interesting to add an UDP file source, which is just an UDP socket listening on some port. A sort of "AirPlay" equivalent.

armSeb, 2017-11-20:
The new buffering system seems to be less stable than to original one, as I have some crashes.

Regarding the gain setting, I'll try it as it seems to create some audio distorsion.

armSeb, 2017-11-20:
OK, for the gain function, I found a workaround, but I don't understand why this happens.

I removed:
```
sample[0] = Amplify(sample[0]);
sample[1] = Amplify(sample[1]);
```
I modified:
`uint32_t s32 = ((sample[RIGHTCHANNEL])<<16) | (sample[LEFTCHANNEL] & 0xffff);`
to 
`uint32_t s32 = ((Amplify(sample[RIGHTCHANNEL]))<<16) | (Amplify(sample[LEFTCHANNEL]) & 0xffff);`

And no audio distorsion anymore (In fact this was not a distorsion, but a noise following the signal level, it's hard to describe).

armSeb, 2017-11-20:
Regarding the buffer issues, when it happens there is no crash, just a freeze (I  guess this is a blocking read function).

Here is the log : 

```
*WM: 
*WM: AutoConnect
*WM: Connecting as wifi client...
*WM: Already connected. Bailing out.
*WM: IP Address:
*WM: 192.168.1.215
19432
readend: toReadFromBuffer=2048, toReadToEnd=2048
read 2048 from buffer, 2048 avail in buff, 0 remaining for request
at mid: read 284
Decoding error 0x0235 (bad main_data_begin pointer) at byte offset 0
Decoding error 0x0235 (bad main_data_begin pointer) at byte offset 418
readend: toReadFromBuffer=1672, toReadToEnd=1672
read 1672 from buffer, 660 avail in buff, 0 remaining for request
at mid: read 0
readend: toReadFromBuffer=660, toReadToEnd=376
readhead: toReadFromBuffer=284, rp=0, wp=284
read 376 from buffer, 0 avail in buff, 1012 remaining for request
at end: read 3368
Decoding error 0x0101 (lost synchronization) at byte offset 4840
Decoding error 0x0235 (bad main_data_begin pointer) at byte offset 4974
readhead: toReadFromBuffer=1388, rp=284, wp=3652
read 0 from buffer, 1980 avail in buff, 0 remaining for request
at end: read 444
at begin: read 1016
```

After the last line (at begin: read 1016) there is no more output, the ESP8266 blue LED stays on, and stay in this state forever. I have to reset it. This can  happens at any time. 

WM is WifiManager library I use for wifi configuration, 19432 is free heap at the beginning.

armSeb, 2017-11-20:
I guess it's a typo ? 

`toReadFromBuffer -= toReadFromBuffer;`

earlephilhower, 2017-11-20:
Thanks for the suggestions!

For the noise problem, I can't see any reason why the results of your way or the original would be any different.  Are you sure it's made any difference?  Doing it in the s32 calculation's fine by me if it really changes things, but it's a real head scratcher.

And VERY good catch on that buffer problem!  Yes, you're obviously correct.  I swear I did a few tests before committing and it seemed fine, but I'm not sure how it ever worked as written.

Again, I appreciate the help!

armSeb, 2017-11-20:
Regarding the noise problem, I tested this my original "attenuation" code in this way: 

```
sample[0] = sample[0] / 10;
sample[1] = sample[1] / 10;
```

And I got noise again. I guess there is a problem with the variable assignment  or type, I need to make further debug. I confirm the buffer works well, I don't have freezes anymore.

earlephilhower, 2017-11-20:
Integer division will add some noise, by definition, since you're effectively rounding down answers from x.9999 to x.0001 to x.000.  But I don't think you're going to hear that unless the SNR is miniscule.

It could be that in division there's a hidden promotion from 16->32 and some signedness issues come into play for the division.  Each sample is represented as a signed 16-bit integer, but in reality is a fixed point 0.15 number (i.e. represents only +/- 1.0f).

I'll have do some paper and pencil work to really see if I can figure it out...

armSeb, 2017-11-20:
I tried this:

```
int16_t right = Amplify(sample[RIGHTCHANNEL]);
int16_t left = Amplify(sample[LEFTCHANNEL]);

uint32_t s32 = ((right<<16) | left & 0xffff);
```

And no noise. So I tried again this:

```
sample[RIGHTCHANNEL] = Amplify(sample[RIGHTCHANNEL]);
sample[LEFTCHANNEL] = Amplify(sample[LEFTCHANNEL]);

uint32_t s32 = ((sample[RIGHTCHANNEL]<<16) | sample[LEFTCHANNEL] & 0xffff);
```

And I get noise again. This is very light but audible noise.

earlephilhower, 2017-11-20:
sample[] is a int16_t, which means both your examples should be exactly the same except that the second one should have an add'l 2 memory stores to update sample[x].  So the one w/temp variables should be a little more efficient, but have no functional changes.

Can you get assembly output for both snippets?  I think that's the only way to see where the difference is coming from at this point.

armSeb, 2017-11-20:
I can manage to get the assembly output but it's difficult to spot the right portion of code.

armSeb, 2017-11-20:
In the meantime, I added this code:

```
  Serial.printf("1: %d\n", ESP.getCycleCount());
  sample[RIGHTCHANNEL] = Amplify(sample[RIGHTCHANNEL]);
  Serial.printf("2: %d\n", ESP.getCycleCount());
  uint16_t test = Amplify(sample[LEFTCHANNEL]);
  Serial.printf("3: %d\n", ESP.getCycleCount());
```
So, the result : 

```
1: -1577370209
2: -1577162148
3: -1576953978
1: -1576745611
2: -1576537550
3: -1576329372
1: -1576121001
2: -1575912950
3: -1575704772
1: -1575496409
2: -1575288346
3: -1575080172
```
It seems the second code option is longer than the first.

armSeb, 2017-11-21:
IT seems there is another issue in the buffer. If the buffer becomes empty during playing, it doesn't refill completely. The consequence is that the buffer stays low and the player becomes sensitive to the network latency. This causes hiccups after a certain amount of time.

It can be better to trigger a complete refill, with a risk of one hiccup instead of many.

armSeb, 2017-11-21:
When the buffer is empty, I trigger a complete refill. This causes a bug hiccup but works better than many little hiccups. I don't know is this is the better way...

```
  if (len) {
    // Still need more, try direct read from src
    Serial.printf("read %d from buffer, %d avail in buff, %d remaining for request\n", bytes, length, len);
    bytes += src->read(ptr, len);
    readPtr = 0;
    writePtr = 0;
    length = 0;
    filled=0;
    return bytes;
  }
```

earlephilhower, 2017-11-21:
I'll try it this weekend, but before that I'll just save the output to a WAV file and verify there really is something different.  The AudioOutputSPIFFSWAV class lets me debug this in Audacity on the PC.  It was very handy with the speech synthesizer debugging.

earlephilhower, 2017-11-21:
Just saw your other updates.  That may be the 48khz hiccup seen on the other issue.

I think your solution is a good way for dealing with this.  The infamous "Buffering..." issue.

I think we can do this in stages, though, and not try and fill all XXX KB (which would be very long) and instead spread it out over several passes through the method.  Instead of one hiccup, 4 or 5 smaller ones (or, if we're lucky, none at all if the stream catches up).

armSeb, 2017-11-21:
If the stream comes back quickly, the buffer refill is inaudible. But there is several times where the latency is too long and in this case the read statement is blocked. But there is not easy workaround as the buffer is too small. 

I started to implement the ICY part, but it's not a trivial task:
- The HTTP client must send a custom header (http.addHeader("Icy-MetaData", "1"))
- The HTTP client must catch the response header icy-metaint which is the amount of bytes to read before finding the icy block.
- When the icy block is found, must read the first byte which is the total block length, then read the block and remove it from mp3 stream (if these block remains in the stream, this will cause hiccups and mp3 decoding errors).

armSeb, 2017-11-21:
I have to investigate a lot ;)

```
read 1302 from buffer, 0 avail in buff, 370 remaining for request
Available buffer after refill: 4096
Decoding error 0x0101 (lost synchronization) at byte offset 63310842
read 1033 from buffer, 0 avail in buff, 638 remaining for request
Available buffer after refill: 0
read 0 from buffer, 0 avail in buff, 1254 remaining for request
Decoding error 0x0101 (lost synchronization) at byte offset 63386224
Decoding error 0x0101 (lost synchronization) at byte offset 63386725
Decoding error 0x0102 (reserved header layer value) at byte offset 63386816
Available buffer after refill: 4096
read 378 from buffer, 0 avail in buff, 1294 remaining for request
Available buffer after refill: 0
read 0 from buffer, 0 avail in buff, 418 remaining for request
MP3 done

Exception (0):
epc1=0x40217290 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

ctx: sys 
sp: 3ffffc60 end: 3fffffb0 offset: 01a0

>>>stack>>>
3ffffe00:  00000005 00000000 00000020 40101ae6  
3ffffe10:  4000437d 00000030 00000016 ffffffff  
3ffffe20:  60000200 00000008 24042403 80000000  
3ffffe30:  20000000 3fff6dc0 80000000 203fc120  
3ffffe40:  80000000 3fffc6fc 3ffe9d04 3fff6dc4  
3ffffe50:  000001d4 003fc120 60000600 00000030  
3ffffe60:  7fffffff 3ffea774 3ffea774 00000001  
3ffffe70:  4021fcc1 3ffef530 00000000 00000000  
3ffffe80:  40105718 4c88ed2f 3fff150c 00000000  
3ffffe90:  00000000 00000000 0000001f 401055d5  
3ffffea0:  4000050c 40217290 3fffc270 4000050c  
3ffffeb0:  40000f68 00000030 00000011 ffffffff  
3ffffec0:  40000f58 00000000 00000020 00000000  
3ffffed0:  00000007 4021be48 3ffe9d04 bfffffff  
3ffffee0:  ffffffff 3fff150c 3ffe9d04 3fffdab0  
3ffffef0:  00000000 3fffdcb0 3ffef0f0 00000030  
3fffff00:  00000000 400042db 00000064 60000600  
3fffff10:  40004b31 3fff6ca4 000002f4 003fc000  
3fffff20:  40105b3a 3ffef0d0 3ffee170 4010741c  
3fffff30:  4021b919 3ffee170 3ffef0d0 f490b24d  
3fffff40:  3fff6ca4 00001000 4021bdae 00000008  
3fffff50:  40218750 3ffe9d04 4021be5b 3ffee224  
3fffff60:  3ffef0d0 4c6d14ad 3ffef0d0 60000600  
3fffff70:  4022d51d 3ffee224 3ffef0d0 f490980d  
3fffff80:  4022d562 3fffdab0 00000000 3fffdcb0  
3fffff90:  3ffef0f8 00000000 40000f65 3fffdab0  
3fffffa0:  40000f49 000280d9 3fffdab0 40000f49  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

```
If I understand correctly the MP3 decoder stops when receiving 0 data ?

earlephilhower, 2017-11-21:
Yes, it checks a flag ("bool running" IIRC) right at the entrance of the ::loop() method and if it's not set returns immediately. 

You can get a more useful stack dump with the ESP Exception Decoder.  Just cut-n-paste that dump into the window and you'll get the cause and a stack trace, if it can figure it out.

For the ICY stuff, is it a 1-time thing, or is it randomly throughout the stream?  If it's 1-time, a simple thing would be to simply do that processing using the FileHTTPStream object in your main code and then just pass the object to the mp3 playback when you're done.

If it's in the stream, then a filter is probably your easiest thing (if not the most optimal).  Just like the Buffer object, pass through all calls and silently remove any ICY blocks.  I was thinking of doing the same for ID3, but since ID3v2 is now always @ the beginning of the file, the prior method is simpler and as correct.

earlephilhower, 2017-11-21:
Oh, and if you'd like to put the final changes into this repo, why not do a fork now before you get too many changes in?  That way you can send in a PR and I'll just click the "ok" button, and you'll be attributed in the history and "blame" files.

armSeb, 2017-11-21:
Yep I can do that :)

I already have my own git repo @home to be able to progress a little bit from the office (when I have some time at work). This repo englobes the whole arduino dir. 

For the ICY part, it's necessary to make a filter, as the blocks are placed periodically into the stream. For that, it's necessary to get the header containing the number of bytes between two blocks, as described here : 

https://cast.readme.io/v1.0/docs/icy

earlephilhower, 2017-11-21:
The disassembly of the two different versions intersped w/source (ugly because of -O3) is here.  I haven't had a chance to look at it in detail yet.
[disass.txt](https://github.com/earlephilhower/ESP8266Audio/files/1492532/disass.txt)

Had to replicate Amplify into 2 separate inline functions or else GCC was making it a function call, but they're just cut-n-pasted copies w/a different name.

earlephilhower, 2017-11-21:
So the only difference I see between the two disassemblies is that there's two additional s16i to update the stack sample[] array on the 2nd one (as expected).  So it'll be microscopically slower due to that, but the actual math seems identical.

ICY actually looks quite simple.  I'd subclass the HTTPStream object, add that metadataindex variable, add a user-specified callback on getting a metadata block, and then just keep track of bytes since last MD block.  When you're about to read the new MD block, send it instead to the CB, which should be a short async call (i.e. set a global flag or something) and reset the byte counter.  The upper layers wouldn't know a thing, and you'd be optimal as you'd not be adding a new filter that'd end up doing 1 more memcpy (and having 1 more buffer!).

earlephilhower, 2017-11-21:
Here you go with a good starting point.  The ICYStream class committed just now dumps the ICY metadata to Serial.printf, and doesn't do sanity checking, but I think it's a good start...

````
...Connecting to WiFi
...Connecting to WiFi
ICY metaint = 16000
Stream size: -1
Buffering...
Decoding error 0x0235 (bad main_data_begin pointer) at byte offset 0
Decoding error 0x0235 (bad main_data_begin pointer) at byte offset 418
Running for 20659 ms...
ICY metadata:
StreamTitle='Tiffany - Could've Been';
---end---
Running for 21660 ms...
Running for 22669 ms...
....
Running for 67726 ms...
Running for 68729 ms...
ICY metadata:
StreamTitle='Bad English - When I See You Smile';
---end---
Running for 69740 ms...
Running for 70741 ms...
Running for 71751 ms...
....
Running for 121812 ms...
Running for 122814 ms...
````

armSeb, 2017-11-21:
Your solution seems to be quite simple, but many of blocks are empty. The server doesn't always send the StreamTitle ;)

earlephilhower, 2017-11-21:
Yup, there are 100s of MD blocks of 0 size that you don't see above (they're silently eaten by the code as they do inject a single 0-byte into the stream).

So you just need to have a callback or update the stream title class variable when the MD block size >0 and it contains the header you want.  The UI either gets a callback, or once a second checks file->GetStreamTitle() and you're set, I think.

armSeb, 2017-11-21:
Wow, I need some time to understand your implementation, with the "friend" class. I'm not a C expert ;)

Let's try :)

armSeb, 2017-11-21:
On certain streams, it seems there is a problem with the blocks, as the mp3 output is corrupted.

I need to investigate.

earlephilhower, 2017-11-21:
Friend in this C++ context just lets the subclass use the parent class' private variables.

The virtual function modifier lets you override only specific functions and use the parent's class to do the rest of the work.  That way you don't have to replicate tons of code.  In this instance, the way it works is the plain HTTPSocket::read() can actually call either the HTTPSocket::readInternal() or the ICYSOcket::readInternal().  It doesn't care, it just jumps through the vtable and everything "just works."

The current ICYStream::readInternal() method needs error checking, and I'd suggest parsing the MD buffer on-the-fly and not reading the whole thing into a malloc()'d space due to memory constraints...if they send a MD of 4K, it'll kill the ESP8266 as there's not enough RAM to handle it + MP3 decode at the same time!  So, take the MD handling bit here w/a grain of salt and expect it to be seriously changed in its final form.

armSeb, 2017-11-21:
Thanks for explain !

I have a strange issue with it: 

```
ICY metaint = 16000
Stream size: -1
19704
Buffering...
Decoding error 0x0101 (lost synchronization) at byte offset 0
Decoding error 0x0104 (reserved sample frequency value) at byte offset 293
mdSize: 3, mdRet: 1
ICY metadata:
StreamTitle='P.DIDDY - Hello good morning ';
---end---
Decoding error 0x0101 (lost synchronization) at byte offset 19863
Decoding error 0x0101 (lost synchronization) at byte offset 20615
mdSize: 0, mdRet: 1
Decoding error 0x0101 (lost synchronization) at byte offset 36163
Decoding error 0x0101 (lost synchronization) at byte offset 36819
mdSize: 0, mdRet: 1
Decoding error 0x0101 (lost synchronization) at byte offset 51628
Decoding error 0x0101 (lost synchronization) at byte offset 52463
Decoding error 0x0104 (reserved sample frequency value) at byte offset 52844
mdSize: 0, mdRet: 1
Decoding error 0x0101 (lost synchronization) at byte offset 67928
Decoding error 0x0101 (lost synchronization) at byte offset 68759
mdSize: 0, mdRet: 1
Decoding error 0x0238 (Huffman data overrun) at byte offset 83810
Decoding error 0x0101 (lost synchronization) at byte offset 84228
Decoding error 0x0238 (Huffman data overrun) at byte offset 84941
Decoding error 0x0101 (lost synchronization) at byte offset 85359
Decoding error 0x0102 (reserved header layer value) at byte offset 85454
Decoding error 0x0104 (reserved sample frequency value) at byte offset 85474
mdSize: 0, mdRet: 1
Decoding error 0x0238 (Huffman data overrun) at byte offset 98857
Decoding error 0x0101 (lost synchronization) at byte offset 100111
mdSize: 0, mdRet: 1
Decoding error 0x0101 (lost synchronization) at byte offset 115575
```

This happens on this stream: http://ice.stream.frequence3.net/frequence3-128.mp3

earlephilhower, 2017-11-21:
That stream has other issues unrelated to ICY.  It's sending non-MP3 data with the encoder info in it.  
````
$ wget -q -O - http://ice.stream.frequence3.net/frequence3-128.mp3 | strings | grep LAME
LAME
LAME3.98.2
LAME
...
````
It doesn't seem to be ID3, just random text data if you look at it via XXD.  There are valid ADTS (i.e. mp3/AAC) frame headers following the text, 0xfffX (i.e. 0xff 0xfb) though.

earlephilhower, 2017-11-21:
Basically the MP3 code's get-data needs to filter out stuff that doesn't begin with 0xfffX.  Could you open an issue on this, as it's separate from the other stuff?

armSeb, 2017-11-21:
For testing purpose I just commented the following line:

`http.addHeader("Icy-MetaData", "1");`

So, the stream plays fine. The issue is in the ICY block parsing ;)

earlephilhower, 2017-11-21:
I think ICY is not the issue.  They're probably just streaming pre-encoded MP3 files, and started sending ones w/o the non-ADTS data in them.

My reasoning is that the wget command above doesn't send the request for ICY headers and it showed the invalid data coming over the line.

[32 more replies]

closed 2017-12-11 by earlephilhower