earlephilhower/ESP8266Audio #8: Restart/loop audio

eketjall, 2017-11-03:
It's probably just me don't understanding, but how do you actually restart or loop an audio file?
My guess was to put the mp3->begin again after the file is finished but didn't work.

```
#include <Arduino.h>
#include <ESP8266WiFi.h>
#include "AudioFileSourceSPIFFS.h"
#include "AudioGeneratorMP3.h"
#include "AudioOutputI2SDAC.h"

// To run, set your ESP8266 build to 160MHz, and include a SPIFFS of 512KB or greater.
// Use the "Tools->ESP8266 Sketch Data Upload" menu to write the MP3 to SPIFFS
// Then upload the sketch normally.  

// pno_cs from https://ccrma.stanford.edu/~jos/pasp/Sound_Examples.html

AudioGeneratorMP3 *mp3;
AudioFileSourceSPIFFS *file;
AudioOutputI2SDAC *out;

void setup()
{
  WiFi.forceSleepBegin();
  Serial.begin(115200);
  delay(1000);
  SPIFFS.begin();
  file = new AudioFileSourceSPIFFS("/microwave.mp3");
  out = new AudioOutputI2SDAC();
  mp3 = new AudioGeneratorMP3();
  mp3->begin(file, out);
}

void loop()
{
  if (mp3->isRunning()) {
    if (!mp3->loop()) mp3->stop();
  } else {
    Serial.printf("MP3 done\n");
    delay(2000);
    mp3->begin(file, out);
  }
}
```

earlephilhower, 2017-11-03:
After playback, the file position is at EOF.  I'm not sure it's safe to reuse the MP3 object, either, after it's completed.  

What I'd suggest is simply { f->close(); delete file; delete mp3; } and then re-open the file and recreate the MP3 generator and start that new object.

eketjall, 2017-11-03:
Thanks for the super quick answer.

f->close(); didn't work (it's not in the scope), but delete file; delete mp3; seems sufficient enough.

earlephilhower, 2017-11-27:
Closing as the response seemed to answer your question.

closed 2017-11-27 by earlephilhower