earlephilhower/ESP8266Audio #12: HTTP streaming : MP3 is fine, but AAC has "artefacts"

gerardwr, 2017-11-22:
Thanks to @earlephilhower and @armSeb sorting out my troubles I can use this great library for streaming HTTP streams (ref. issue #9 ).

In issue #9 I already mentioned my remaining issue with playing AAC streams,  but issue #9 was about "connection errors" and the AAC issue was more or less "lost".

The issue is:
- MP3 streams play with good quality, often for many hours without loosing connection to the stream, with only occasional tiny hickups probably due to "network errors".
- AAC streams also play with good quality, also often for many hours without loosing connection to the stream. But every 5-10 seconds the sound is disturbed by a disturbing "whistling hiss" lasting for 1-2 seconds. It sounds as if there's occasionally another sound "mixed" with the original stream. Imperfect combining stereo to mono, maybe?

I tested using the Morow Radio stream, that's available both as MP3 stream and AAC stream.
http://stream.morow.com:8080/morow_med.aacp
http://stream.morow.com:8000/morow.mp3
Both streams play fine in my browser.

Could it be that the AAC decoding software has a remaining issue?

I created the AAC sketch below adapting the StreamMp3FromHTTP example (both with a 4096 buffer).
```
#include <Arduino.h>
#include <ESP8266WiFi.h>
#include "AudioFileSourceHTTPStream.h"
#include "AudioGeneratorAAC.h"
#include "AudioOutputI2SNoDAC.h"
#include "AudioFileSourceBuffer.h"

// To run, set your ESP8266 build to 160MHz, update the SSID info, and upload.

/*
21-11-2017. AAC Plays "OK" on Wemos D1 clone with artefacts every 5-10 secs
*/

// Enter your WiFi setup here:
const char *SSID = "....";
const char *PASSWORD = "....";

// Randomly picked URL
const char *URL="http://stream.morow.com:8080/morow_med.aacp";                     // Morow Radio, 128kbps 44.1 kHz : works OK, artefacts every 5-10 sec

AudioGeneratorAAC *aac;
AudioFileSourceHTTPStream *file;
AudioFileSourceBuffer *buff;
AudioOutputI2SNoDAC *out;

void setup()
{
  Serial.begin(115200);
  delay(1000);
  Serial.println("Connecting to WiFi");

  WiFi.disconnect();
  WiFi.softAPdisconnect(true);
  WiFi.mode(WIFI_STA);
  
  WiFi.hostname("melody");
  
  byte zero[] = {0,0,0,0};
  WiFi.config(zero, zero, zero, zero);

  WiFi.begin(SSID, PASSWORD);

  // Try forever
  while (WiFi.status() != WL_CONNECTED) {
    Serial.println("...Connecting to WiFi");
    delay(1000);
  }

  file = new AudioFileSourceHTTPStream(URL);
  buff = new AudioFileSourceBuffer(file, 4096);     // you can increase default 2048 bufffersize to 4096 (seems better) or 8192 (seems much worse)
  out = new AudioOutputI2SNoDAC();
  aac = new AudioGeneratorAAC();
  aac->begin(buff, out);
}

void loop()
{
  if (aac->isRunning()) {
    if (!aac->loop()) aac->stop();
  } else {
    Serial.printf("AAC done\n");
    delay(1000);
  }
}

armSeb, 2017-11-22:
I can't get your AAC stream to play !

```
Connecting to WiFi
...Connecting to WiFi
...Connecting to WiFi
...Connecting to WiFi
Buffering...
AAC done
AAC done
AAC done
AAC done
AAC done
AAC done
AAC done
AAC done
```

gerardwr, 2017-11-22:
Really?? It's playing fine (with artefacts) here ATM, I swear !!

I ran into trouble after the "lwip2 MSS experiment" so I reinstalled yesterdays backup of ESP8266. Compiled the ACC sketch with lwip Prebuilt Source (gcc), and it ran!

Sketch output is rather boring because no errors to report.
```...Connecting to WiFi
...Connecting to WiFi
...Connecting to WiFi
Buffering...

gerardwr, 2017-11-22:
I assume you checked the URL in the browser to make sure the AAC stream is "up" in your part of the world?

armSeb, 2017-11-22:
Yep, it plays fine in VLC on my mac.

gerardwr, 2017-11-22:
I am stumped!

gerardwr, 2017-11-22:
To document the artefacts I recorded 1 minute of the AAC stream currently playing on my ESP8266 with Quicktime Player on my Mac via the builtin microphone, so recording quality is not great.

But you can hear the artefacts clearly, "whistles like a bird" and "hisses", mainly at:
20-25 sec
40-59 sec

Here's the 2.4M file : [Naamloos.m4a.zip](https://github.com/earlephilhower/ESP8266Audio/files/1497718/Naamloos.m4a.zip)

armSeb, 2017-11-23:
It seems to be a data corruption. It can be interesting to download a part of this AAC stream and look for strings like ICY blocks.

gerardwr, 2017-11-23:
Challenges, challenges ;-)

Used Mplayer to play stream, this is the complete output after 10 min streaming:
- when started some "mismatches" but while playing no more errors
- only ICY reports every few minutes when new song is playing (see end of listing"
- so the artefacts I hear every few seconds is not from ICY

```
imac-van-gerard:~ gerard$ mplayer http://stream.morow.com:8080/morow_med.aacp
MPlayer 1.3.0-4.2.1 (C) 2000-2016 MPlayer Team

Playing http://stream.morow.com:8080/morow_med.aacp.
Resolving stream.morow.com for AF_INET6...

Couldn't resolve name for AF_INET6: stream.morow.com
Resolving stream.morow.com for AF_INET...
Connecting to server stream.morow.com[94.23.2.201]: 8080...

Name   : MOROW
Genre  : Prog Rock
Website: http://www.morow.com
Public : yes
Bitrate: 64kbit/s
Cache size set to 320 KBytes
Cache fill:  0.00% (0 bytes)   
ICY Info: StreamTitle='Rush - Limelight';
Cache fill: 15.00% (49152 bytes)   

AAC file format detected.
==========================================================================
Opening audio decoder: [ffmpeg] FFmpeg/libavcodec audio decoders
libavcodec version 57.24.102 (internal)
[aac @ 0x110196b90]element type mismatch 1 != 0
AUDIO: 44100 Hz, 2 ch, floatle, 0.0 kbit/0.00% (ratio: 0->352800)
Selected audio codec: [ffaac] afm: ffmpeg (FFmpeg AAC (MPEG-2/MPEG-4 Audio))
==========================================================================
2017-11-23 12:46:55.417 mplayer[772:25868] 12:46:55.416 WARNING:  140: This application, or a library it uses, is using the deprecated Carbon Component Manager for hosting Audio Units. Support for this will be removed in a future release. Also, this makes the host incompatible with version 3 audio units. Please transition to the API's in AudioComponent.h.
AO: [coreaudio] 44100Hz 2ch floatle (4 bytes per sample)
Video: no video
Starting playback...
[aac @ 0x110196b90]element type mismatch 1 != 0
[aac @ 0x110196b90]element type mismatch 1 != 0
[aac @ 0x110196b90]element type mismatch 1 != 0
[aac @ 0x110196b90]element type mismatch 1 != 0
[aac @ 0x110196b90]element type mismatch 1 != 0
[aac @ 0x110196b90]element type mismatch 1 != 0
[aac @ 0x110196b90]element type mismatch 1 != 0
[aac @ 0x110196b90]element type mismatch 1 != 0
[aac @ 0x110196b90]element type mismatch 1 != 0
[aac @ 0x110196b90]element type mismatch 1 != 0
[aac @ 0x110196b90]element type mismatch 1 != 0
[aac @ 0x110196b90]element type mismatch 1 != 0
[aac @ 0x110196b90]element type mismatch 1 != 0
[aac @ 0x110196b90]element type mismatch 1 != 0
[aac @ 0x110196b90]element type mismatch 1 != 0
[aac @ 0x110196b90]element type mismatch 1 != 0
A:   0.0 (00.0) of 0.0 (unknown)  0.6% 27% 
Audio device got stuck!
A:   0.0 (00.0) of 0.0 (unknown)  0.6% 32% 
Audio device got stuck!
A: 105.8 (01:45.7) of 0.0 (unknown)  0.7% 38% 
ICY Info: StreamTitle='M Kelly - M Kelly';
A: 109.9 (01:49.8) of 0.0 (unknown)  0.7% 38% 
ICY Info: StreamTitle='Steven Wilson - Permanating (New)';
A: 316.8 (05:16.8) of 0.0 (unknown)  0.7% 38% 
ICY Info: StreamTitle='Circus Maximus - Sane to more';
A: 548.3 (09:08.2) of 0.0 (unknown)  0.7% 38% 
```

earlephilhower, 2017-11-23:
ICY corruption would occur at every 8000 or 16000 bytes, because it's adding a 0-byte to the stream at some frequency to indicate no new metadata.  So it's not that.

The Stereo->Mono downmix has no history, so it wouldn't be that.

1-2 seconds of hiss is more than a frame's worth of data so it's not a normal corruption.

There could be MP4 ATOMs in the stream w/metadata included, but that should be filtered/ignored by any stream code in Helix. 

It could just be that the Helix player code doesn't grok some new AAC extension they're using.  If you can find/make an AAC file (maybe binary recording via wget of the stream) that would help debug.  Playing that stream back on a PC app using the Helix code and hearing or not/hearing it would be the best way to narrow that down.

earlephilhower, 2017-11-23:
Oh, just to see if they're sending ICY even if not requested, can you change
> #include "AudioFileSourceHTTPStream.h"
to
> #include "AudioFileSourceICYStream.h"
and
>   file = new AudioFileSourceHTTPStream(URL);
to
>   file = new AudioFileSourceICYStream(URL);

gerardwr, 2017-11-23:
@earlephilhower : Did you have a chance to listen to the analog recording I uploaded above? There seems to be no regularity in the timing of the artefects.

Inspired by a remark from @armSeb I used wget to download the AAC stream but my Mac refuse to see it as a valid AAC file. There are no readable strings in the downloaded file (like "ICY") so I abandoned this route and switched to Mplayer. Any suggestion for an free AAC-file analyser for Mac, Googled for it but did not find much.

I'll give changing to AudioFileSourceICYStream a go, and see what this brings.

For "daily use" I switched to the MP3 stream of Morow Radio, but getting the AAC steam working also is a nice challege :-)

armSeb, 2017-11-23:
In the meantime, I tried to read the FLAC streaming from Frequence 3 (just to test). 

http://hd.stream.frequence3.net/frequence3.flac

But the stream is packed into ogg container. I'll try from my own server to see if I can have reliable quality.

earlephilhower, 2017-11-23:
@gerardwr no, I've not listened to the sound, but do believe you.  

If you can find a HelixPlayer binary that runs on your OS, please try that with the stream and report.  It'll isolate the decode (their code) or the streaming (my code)...

Other thing is it may be SBR encoded on those frames.  There's not enough RAM on the chip to do SBR decode per their document (>64K needed, only 40KB free for apps on the ESP).

gerardwr, 2017-11-23:
@earlephilhower 

No Helixplayer for Mac found :-(

I think I remember seeing "SBR" somewhere, but having checkd a mllion things, I forgot if, when and where.

For now i'm going to try another AAC stream and see if this has artefacts too. Will let you know.

gerardwr, 2017-11-23:
@earlephilhower 

Just remembered where I saw "SBR".

When I play the morow AAC stream in VLC and inspect the "Codec details" it lists (sorry it's in Dutch):
[image]

Do I understand you correctly that this means that the ESP is too low on RAM to play this stream correctly and that could cause the artefacts?

earlephilhower, 2017-11-23:
https://www.helixcommunity.org/projects/datatype/2005/aacfixptdec

2. MPEG4-LC HE-AAC (SBR enabled, "high quality" mode)

    ROM (const globals) = 37580
    RAM (heap, max 2 channels) = 79580
    Code size - x86 (MSVC++ 6.0) = 47822
    Code size - ARM (ADS ARM9TDMI) = 41764
    Additional RAM required per channel = 17198 

So yup, AAC+SBR is not going to happen on the ESP8266.

gerardwr, 2017-11-23:
That's really a shame, the artefact sounded like some kind of decoding "glitch", apart from that the qulaity of the stream was good.

In fact I think apart from the artefact the AAC stream seemed more stable than the MP3 stream. The overnight stream was the AAC stream.

Can't win them all, thx for enlightening me.

So when wil you adapt the library for the ESP32. Has 2 cores @ 240Mhz cores and 500KB Ram ;-)
So much to do, so little time!

earlephilhower, 2017-11-24:
I actually just received a ESP32 board last week but haven't plugged it in yet.  It's a real SOC, with dual cores and a 3rd, ultra low-power CPU for deep sleep operation and 1/2MB of RAM.  No challenge at all to use, you could probably just compile the Helix sources as-is with full options and it'd "just work."  With threading you don't even have to worry about loops and splitting work up, just spin up a thread and let it sleep on I2S mutex full...

The biggest challenge seems to be that it's even more poorly documented than the ESP8266, if I'm reading the forums right. :(

earlephilhower, 2017-11-24:
For fun I added the SBR code back and made it compile.  The data struct it needs to handle SBR is  *50788* bytes.  Ouch!

gerardwr, 2017-11-24:
@earlephilhower 

Always fun to do an (SBR) experiment, too bad you missed out on the joy of success this time ;-)

re. ESP32:
I have an ESP32 lying around, but the Arduino-ESP code is given me cc1plus errors caused by the older CPU in my Mac (acc. iggr), and the build process seems to require an update (ref.https://github.com/espressif/arduino-esp32/issues/764).

In a German Computer magazine was a interesting project building a ESP32 WAV player:
- hardware like this
[image]
- If you want a peek at the Arduino code
[espsoundmachine.zip](https://github.com/earlephilhower/ESP8266Audio/files/1501369/espsoundmachine.zip)

Can send you the entire article in pdf but it's in German. Know that you are learning to understand Dutch radio, but not sure how's you German ;-)

earlephilhower, 2017-11-24:
I know c't magazine, they've been around for ages.

This example is a waste of the ESP32 IMO.  You can do the same w/the ESP8266 (a user already is using a SD card for file source) *and* play MP3 and AAC and FLAC and WAV files. And no DAC required, either. :)

gerardwr, 2017-11-24:
Suprised that you know c't magazine, have been reading it from the beginning, and still do.

I agree that an ESP32 is "overdone" for the c't use case, but I think it is an good "proof of concept" example on how to create parallels task in the ESP32 across the 2 cores.

AND I agree that your library is extremely flexibel in uses cases, and the c't example is a just "one trick pony".

Nevertheless ESP32 looks like an appealing alternative for playing internet radio streams:
- 1 core for communication, 1 core for sound reproduction
- lots of RAM for communication buffers and extensive code (SBR?)
- has builtin 2 DAC (8bit) channels, so no more delta sigma processing required
- it's a new challenge ;-)

Found my ESP32 in a drawer today, and uploaded my first .bin to the Flash using esptool. Should be able to run the infamous Arduino "Blink" sketch shortly, at 240Mhhz ;-)

armSeb, 2017-11-25:
For SBR it can be tricky to use SPIRAM for requirements, is it possible ?

earlephilhower, 2017-11-26:
Theoretically it is possible, since you can do anything with a Turing machine and enough memory.

Practically, I highly doubt it.  There are many individual arrays and values in that structure, and they're combined in lots of places to generate the add'l spectral bands.  Even if you separated every memory access to the PSInfoSBR structure, an involved task, I really, really doubt it would operate fast enough given the SPI overhead.

earlephilhower, 2017-11-27:
As it looks like SBR is the issue here, and it's not doable on the ESP8266, is there any reason to leave this open?  We're tracking the SPIRAM buffer and error handling in other active issues.

gerardwr, 2017-11-27:
Agreed, I have closed this issue.

closed 2017-11-27 by gerardwr