earlephilhower/ESP8266Audio #15: Cannot find AudioFileSourceFastROMFS.h

daitangio, 2017-11-26:
Hi, I was unable to try the midi example because the Arduino complains about a missed AudioFileSourceFastROMFS.h
I tried to install your FastROMFS filesystem with no luck.
How can I fix it?

earlephilhower, 2017-11-27:
Howdy,

I pulled that one out because I haven't cleaned up the new filesystem to my liking.  You can either replace it with a SPIFFS or SD file in the example, or you can get the files from just before I pulled them at
git commit 551a8ac0bb857c03b16cdb7ce553db476ba30728 .  The metadata in the filesystem is going to change to do better wear leveling, but OTW it's reasonably stable.

If you do use it, you'll need the ESP8266FastROMFS library as well as the IDE plugin (check my profile for it) to actually generate and upload a filesystem to the chip.

If you try SPIFFS you may not be able to do as high polyphony, and may need to use a simpler MIDI file and smaller SF2 soundfont to get reasonable speed.  SPIFFS is great for doing small sequential IOs, but random blocks, which is what wavetables need, are extremely slow.

daitangio, 2017-11-30:
Thank you. By the way this library is wonderful! I was able to get esp8266 running a small wav using a poor piezo buzzer (I will use a transitor soon, as you suggested).

earlephilhower, 2017-12-01:
Updating the title to remind me to change the MIDI demo to SPIFFs when I get a chance.

daitangio, 2017-12-02:
I can take care of it, only give me some time. I will send you a pull request, okey?

earlephilhower, 2017-12-02:
Great, whenever you get a chance.  No rush!

daitangio, 2017-12-03:
I have done a first implementation with the NoDac version (I have no a DAC to try with).
I got a core dump shortly after booting.
I checked files are correctly on flash. Where am I wrong?

earlephilhower, 2017-12-04:
Nowhere, there was a bug introduced when I added a ::loop member to all objects to support HTTP buffering.  The passed in object pointer for the File wasn't being saved, and we were calling a garbage data.  That File is wrapped in the MIDI setup with a custom buffering routine, so it was fine when checked in a while back, and I didn't build this example before checking in because I'd removed the FastROMFS stuff from the repo...

earlephilhower, 2018-01-21:
Closing, the merge was done a while back.

closed 2018-01-21 by earlephilhower