earlephilhower/ESP8266Audio #18: Use of printf_P ?

dagnall53, 2017-12-16:
Hi, and thanks for your ESP8266 audio library... However, I have an issue that is certainly down to my system, but I was hoping you could shed some light on it: 
I am getting an error regarding "printf_P"

Arduino: 1.8.5 (Windows 10), Board: "NodeMCU 1.0 (ESP-12E Module), 80 MHz, 115200, 4M (3M SPIFFS)"

C:\Users\Richard\Documents\Arduino\libraries\ESP8266Audio-master\src\AudioFileSourceBuffer.cpp: In constructor 'AudioFileSourceBuffer::AudioFileSourceBuffer(AudioFileSource*, int)':

C:\Users\Richard\Documents\Arduino\libraries\ESP8266Audio-master\src\AudioFileSourceBuffer.cpp:30:23: error: 'class HardwareSerial' has no member named 'printf_P'

   if (!buffer) Serial.printf_P(PSTR("Unable to allocate AudioFileSourceBuffer::buffer[]\n"));

                       ^
I have looked and note you use this function a lot in the code, so some simple deletes did not allow me to make the examples work...

Presumably there is a Library that you are using that defines "Serial.printf_P" ... do you know which one it is? 

I have just deleted my whole esp8266 library sets and clean re-installed arduino 1.8.5 hoping it was a recent library addition ...   I can see others have the code working so I assume I have not (re?) installed something crucial! Is it something defined in your fork of the esp8266 code? 

Many thanks
Dagnall

armSeb, 2017-12-16:
Hello, 

Please use the ESP8266 library v2.4 rc2:

https://github.com/esp8266/Arduino/releases

earlephilhower, 2017-12-17:
Thanks, @armSeb .  I've included the JSON link for installing 2.4.0-rc2 in the README.md.  Closing, as this will cover it.

closed 2017-12-17 by earlephilhower