earlephilhower/ESP8266Audio #19: possible memory leak?

dagnall53, 2017-12-18:
Hi, 
I have a simple repetitive loop, playing a chuff sound. (for a loco sound module I am working on).
After 727 repeats it reliably fails, with the exception shown below.  Its not dependant on the wav file, using a much longer "orchestra" wav, it also crashed at 727 loops. 
Is the crash an error in the way I set up repeated wav files, or is it a memory leak elsewhere?- I'm afraid the exception decode means little to me.

To try and test a theory, I added curly brackets and added "delete file; file=Null;}"  after the "wav->stop();" 
(I have commented this out in the copy below).  
Using this code seems to cure or hide the problem...Its run faultlessly more than 1704 repeats so far.. but I thought your code already closed the file as part of stop?

Many thanks 
Dagnall

`Exception (29):
epc1=0x40207180 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

ctx: cont 
sp: 3ffeffc0 end: 3fff01d0 offset: 01a0

>>>stack>>>
3fff0160:  40104c70 13591d92 3ffeeefc 3ffef1a0  
3fff0170:  3fffdad0 3ffe8900 3ffeeb58 401004d8  
3fff0180:  3fffdad0 3ffef1a0 402091f0 40209100  
3fff0190:  40201356 00000001 00000000 40206cd5  
3fff01a0:  3fffdad0 3ffef0e0 3ffeeb58 40206dc5  
3fff01b0:  3fffdad0 00000000 3ffef198 4020923c  
3fff01c0:  feefeffe feefeffe 3ffef1b0 4010070c  
<<<stack<<<`
This decodes `as:ecoding 9 results
0x40104c70: ets_timer_arm_new at ?? line ?
0x401004d8: malloc at C:\Users\Richard\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc2\cores\esp8266\umm_malloc/umm_malloc.c line 1668
0x402091f0: loop_task at C:\Users\Richard\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc2\cores\esp8266/core_esp8266_main.cpp line 57
0x40209100: String::operator=(String const&) at C:\Users\Richard\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc2\cores\esp8266/WString.cpp line 220
0x40201356: delay at C:\Users\Richard\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc2\cores\esp8266/core_esp8266_wiring.c line 54
0x40206cd5: BeginPlay(char const*) at C:\Users\Richard\AppData\Local\Temp\arduino_modified_sketch_25729/testWAV_sounds.ino line 14
0x40206dc5: loop at C:\Users\Richard\AppData\Local\Temp\arduino_modified_sketch_25729/testWAV_sounds.ino line 36
0x4020923c: esp_schedule at C:\Users\Richard\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc2\cores\esp8266/core_esp8266_main.cpp line 57
0x4010070c: cont_norm at C:\Users\Richard\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc2\cores\esp8266/cont.S line 109
`

Copy of main part of my program: 

`void BeginPlay(const char *wavfilename){
  file = new AudioFileSourceSPIFFS(wavfilename);
  wav->begin(file, out);
}
void setup()
{
  Serial.begin(115200);
  delay(1000);
  out = new AudioOutputI2SDAC();
  wav = new AudioGeneratorWAV();
  BeginPlay("/initiated.wav");
  x=0;
}
void loop()
{
  if (wav->isRunning()) {
    if (!wav->loop()) {wav->stop();
    }    //;delete file; file = NULL;}  // test to see if clearing file has any effect...
  } else {
    Serial.println(x);
    x=x+1;
    ;Serial.printf("WAV done\n");
    delay(1);
    BeginPlay("/chuff.wav");
      }
}`

earlephilhower, 2017-12-18:
Well, if you do a "new" then you better do a "delete" or else bad stuff like this will happen.  My examples, sadly, are quite poor and most likely don't do it.

Yes, please delete the file and wav
````
....
if (!wav->loop()) {
   wav->stop();
   delete wav;
   delete file;
}....
````

Your code *should* "delete wav; wav = new AudioGeneratorWAV;" before doing wav->begin(), but it should work OK as there is no state needed in the WAV class.  For other classes, like MP3 and AAC, there is and it won't work w/o deleting and recreating the object.

dagnall53, 2017-12-18:
FAST!
Thanks, 
Certainly deleting "file" seemed the cure, (its the "file" that I think repeatedly create "new) 
Do I need to delete the "wav"  and or  "out" as well? 
I modified the code to "delete file; delete wav;" (without NULL, ) and it crashed on first run. 
Just having "delete file;" works, but I have not waited 727 loops yet.... 

Dagnall

earlephilhower, 2017-12-18:
Changed the title to remind me the examples should really clean up after themselves.

earlephilhower, 2017-12-18:
Yes, the out and wav in your code should be fine.  Just delete the file before new()ing another one and you should be fine.

dagnall53, 2017-12-18:
Many thanks.
You can close this if you like!

ErichHeinemann, 2018-02-18:
... a .rewind() - function for the Audio-Generator would be cool to play the last loadded WAV-Sounds and file again and again without the need to setup all again.