earlephilhower/ESP8266Audio #20: Is it possible to change the playback speed/rate?

dagnall53, 2017-12-18:
HI I thought I would add this as a separate "issue".
I would like to change the playback speed of my "chuff" wav.
 
I had assumed something like 
out->SetRate((wav->sampleRate)*My_Rate_Modifiier);  .. except of course that does not compile!..

Huge thanks
Dagnall

earlephilhower, 2017-12-18:
Howdy,

With the SW DAC I2C you could do this, but you'll need to cast to an (int) in that call, and it will increase the sample rate (i.e. it'll sound higher pitched).  You will need at least one pass through the loop() method first, because the codec may not set the I2S frequency until some data goes through.

For a real DAC, they often are coded to only recognize specific frequencies (and include special digital filters optimized for that rate) and you can use the FilterAndDecimate() object in the output path and set any rational decimation ratio as well as filter to get rid of artefacts, if you're so inclined.

dagnall53, 2017-12-18:
Hmm, looks like I need to work harder on this than I thought.. 
I initially had a simple version of ESP8266Wavplay up and running with a DAC that hacked the wavplay.cpp with " i2s_set_rate(wProps.sampleRate*wavspeed);  "   - It worked, over a range of about 0.75 to 2 times the sample rate (I was using 11k sample rate) and using adafruit i2sDAC).... but there was something else in that code that did not run the LRC line correctly so there was random noise, even when I did not modify the playback rate. . Yours seems much better behaved so I was hoping I could hack the is2 set rate somewhere in the same way. Perhaps as a "pitch" function. 

For train sounds, adjusting the playback rate of a the "chuff" is remarkably effective, and the pitch change also seems "correct" when you listen to it.

Dagnall

dagnall53, 2017-12-20:
I tried the simple approach by adding I2S.h and then using i2s_set_rate(11025*Pitch); 
As you noted, this only "sort of" works and it introduces interesting (annoying) faults in the playback.
I tried it with NODAC versions and they sounded worse. 

I am going to try to adjust the "chuff" periodicity by making my sample longer and then truncating it when the next "chuff" is needed. But this introduces another issue. I would like to be able to play a whistle or other sounds whilst the chuff continues: Have you given any thought as to how the library might support mixing sounds...? (perhaps a new issue topic?) 
Ideally I was wondering if there is any way I can run two wav samples asynchronously into the "out" function(s). I have tried the obvious, declaring an additional audio generator (wav2) and file source (file2) and sending this to the same output, along with a second "wav2->loop" for the second audio generator (as well as to a second output), but this, not surprisingly, does not work...
Have you a simple solution to mix two sources on the fly? 
Could be very useful for other applications ...

Thanks
Dagnall

closed 2017-12-22 by dagnall53

earlephilhower, 2017-12-22:
Howdy,

One way to change speed is by decimation.  There is a FIR+decimation filter in the repo you can try as well.  If I ever find some free time I'll write up an example as it is quite powerful and can let you do interesting effects like reverb or LPF/HPF to change tone quality.

dagnall53, 2017-12-22:
Thanks.. 
I looked at the decimation stuff but decided it was a bit complex for my needs. 
I have now rewritten my sound stuff to shorten the "chuff" sounds when the loco is running faster. This seems to be the same approach as used in commercial sound units and gives excellent results. 
My program plays one file at  time, switching, for example, to a whistle instead of the repetitive chuff when I call for a whistle. What would be great is if I could somehow mix two sound sources (chuff and selectable others such as whistle) to give a single output. 
I have started a new thread to ask this question. 
Cheers
Dagnall

rshashikanth, 2023-07-05:
Any help regarding the change of speed in playback, is really helpful in some of my projects, like audio books and stuff

Please post the code, if you are successful in achieving this