earlephilhower/ESP8266Audio #21: Web radio crashes in loop (entered a m3u url)

tueddy, 2017-12-18:
Hi,

i'm testing your webradio and accidentialy entered a m3u playlist url instead of a mp3 url.
Demo crashes on every startup.

Steps to reproduce:
Enter this URl in webradio:
http://dl-ondemand.radiobremen.de/bremeneins.m3u

Serial output dump:

...Connecting to WiFi
Connected
Go to http://192.168.178.28/ to control the web radio.
Resuming stream from EEPROM: http://dl-ondemand.radiobremen.de/bremeneins.m3uRunning for 7 seconds...Free mem=42128
Changing URL to: http://dl-ondemand.radiobremen.de/bremeneins.m3u

Soft WDT reset

ctx: cont 
sp: 3fff0740 end: 3fff0a90 offset: 01b0

>>>stack>>>
3fff08f0:  402144ae 00000030 00000013 ffffffff  
3fff0900:  402144ab 3fff3c08 3fff3c48 402061ec  
3fff0910:  3fff718a 3fff718b 000000e0 ffffffe0  
3fff0920:  000000c0 3fffc6fc 00000001 3fff3bcc  
3fff0930:  00000000 3fff3c48 3fff3c08 00000030  
3fff0940:  01930467 00000000 3fff9f9c 3fff335c  
3fff0950:  3ffef442 3ffef9a8 4021134c 00000030  
3fff0960:  3fff1cf0 00000000 3fff1cd8 3ffef344  
3fff0970:  3ffef344 00000031 3ffe8304 401004d8  
3fff0980:  3fff718b 00000186 3fff19bc 40210c45  
3fff0990:  3ffe84ec 00000000 00000000 4021a9e7  
3fff09a0:  00000000 3fff19bc 3fff19a4 4021a9be  
3fff09b0:  00000000 3fff711c 00001bd6 3fff3c08  
3fff09c0:  3fff3c48 3fff3c08 3fff3c48 402064e5  
3fff09d0:  40236c76 00000000 3fff19bc 3fff6fcc  
3fff09e0:  00000000 00000000 3fff3bcc 402144ab  
3fff09f0:  55555555 00000001 3fff3bcc 3fff6fcc  
3fff0a00:  3fff3bdc 3fff70cc 3fff3bcc 4021460f  
3fff0a10:  c024bfd7 3fff335c 3ffef3f0 40211ae3  
3fff0a20:  40216860 00000000 00000000 3fff0a40  
3fff0a30:  0000a490 3ffef3f0 3ffef2f8 40211d09  
3fff0a40:  24160000 3ffef41c 3ffef9a8 3ffefa68  
3fff0a50:  3fffdad0 3ffef41c 7fffffff 40211942  
3fff0a60:  3ffe9048 1cb2a8c0 feefeffe 3ffefa68  
3fff0a70:  3fffdad0 00000000 3ffefa60 40217a54  
3fff0a80:  feefeffe feefeffe 3ffefa70 40100710  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v00000000
~ld

EXP exception tracer:

Decoding 18 results
0x402144ae: AudioGeneratorMP3::DecodeNextFrame() at C:\Users\D.Carstensen\Documents\Arduino\libraries\ESP8266Audio\src/AudioGeneratorMP3.cpp line 266
0x402144ab: AudioGeneratorMP3::DecodeNextFrame() at C:\Users\D.Carstensen\Documents\Arduino\libraries\ESP8266Audio\src/AudioGeneratorMP3.cpp line 266
0x402061ec: mad_header_decode at C:\Users\D.Carstensen\Documents\Arduino\libraries\ESP8266Audio\src\libmad/frame.c line 354
0x4021134c: StatusCallback(void*, int, char const*) at C:\Users\D8310~1.CAR\AppData\Local\Temp\arduino_modified_sketch_674620/WebRadio.ino line 226
0x401004d8: malloc at C:\Users\D.Carstensen\Documents\Arduino\hardware\esp8266com\esp8266\cores\esp8266\umm_malloc/umm_malloc.c line 1668
0x40210c45: _malloc_r at C:\Users\D.Carstensen\Documents\Arduino\hardware\esp8266com\esp8266\cores\esp8266/heap.c line 15
0x4021a9e7: _strdup_r at /Users/igrokhotkov/e/newlib-xtensa/xtensa-lx106-elf/newlib/libc/string/../../../.././newlib/libc/string/strdup_r.c line 11
0x4021a9be: strdup at /Users/igrokhotkov/e/newlib-xtensa/xtensa-lx106-elf/newlib/libc/string/../../../.././newlib/libc/string/strdup.c line 11
0x402064e5: mad_frame_decode at C:\Users\D.Carstensen\Documents\Arduino\libraries\ESP8266Audio\src\libmad/frame.c line 446 (discriminator 1)
0x40236c76: i2s_write_sample_nb at C:\Users\D.Carstensen\Documents\Arduino\hardware\esp8266com\esp8266\cores\esp8266/core_esp8266_i2s.c line 192
0x402144ab: AudioGeneratorMP3::DecodeNextFrame() at C:\Users\D.Carstensen\Documents\Arduino\libraries\ESP8266Audio\src/AudioGeneratorMP3.cpp line 266
0x4021460f: AudioGeneratorMP3::loop() at C:\Users\D.Carstensen\Documents\Arduino\libraries\ESP8266Audio\src/AudioGeneratorMP3.cpp line 266
0x40211ae3: StartNewURL() at C:\Users\D8310~1.CAR\AppData\Local\Temp\arduino_modified_sketch_674620/WebRadio.ino line 296 (discriminator 2)
0x40216860: EspClass::getFreeHeap() at C:\Users\D.Carstensen\Documents\Arduino\hardware\esp8266com\esp8266\cores\esp8266/Esp.cpp line 409
0x40211d09: loop at C:\Users\D8310~1.CAR\AppData\Local\Temp\arduino_modified_sketch_674620/WebRadio.ino line 361
0x40211942: setup at C:\Users\D8310~1.CAR\AppData\Local\Temp\arduino_modified_sketch_674620/WebRadio.ino line 275
0x40217a54: loop_wrapper at C:\Users\D.Carstensen\Documents\Arduino\hardware\esp8266com\esp8266\cores\esp8266/core_esp8266_main.cpp line 57
0x40100710: cont_norm at C:\Users\D.Carstensen\Documents\Arduino\hardware\esp8266com\esp8266\cores\esp8266/cont.S line 109

Best regards
[image]

Dirk

tueddy, 2017-12-18:
appendix:

i have tested before with this URL.

http://rb-bremeneins-live.cast.addradio.de/rb/bremeneins/live/mp3/128/stream.mp3

This url works fine in browser but plays nothing in webradio. Another URL works fine..

earlephilhower, 2017-12-23:
Thanks for the info.  It should barf, or do nothing, but not crash.  Having a repeatable crash is a good thing and makes it easier to debug than an actual streaming radio station.  The error handling in the MP3 decoder is less than optimal (mostly my doing because I had to turn the whole libmad decoder inside-out to make it work on a per-32-sample chunk to fit in memory)...

earlephilhower, 2017-12-23:
So the inner decode loop in the MP3 generator had a potential infinite loop that the short read caused.  I've pushed a fix for that.

Trying the URL I found in the M3U file in the StreamMP3FromHTTP (again, after fixing a minor PROGMEM string issue), however, I was able to play it fine:
````
const char *URL="http://rb-bremeneins-live.cast.addradio.de/rb/bremeneins/live/mp3/128/stream.mp3";
````

Gives...
````
Connecting to WiFi
...Connecting to WiFi
...Connecting to WiFi
...Connecting to WiFi
...Connecting to WiFi
...Connecting to WiFi
Connected
Running for 21929 ms...
STATUS(buffer) '2' = 'Refilling buffer'
METADATA(ICY) 'StreamTitle' = 'Eric Clapton - I Shot The Sheriff'
Running for 22930 ms...
Running for 23934 ms...
Running for 24943 ms...
Running for 25950 ms...
....
Running for 126587 ms...
Running for 127593 ms...
Running for 128604 ms...
Running for 129606 ms...
Running for 130612 ms...
METADATA(ICY) 'StreamTitle' = 'Tom Jones - Sexbomb'
Running for 131620 ms...
Running for 132626 ms...
Running for 133636 ms...
Running for 134643 ms...
````

earlephilhower, 2017-12-23:
The problem is your URL is too long for the buffer in the WebRadio.ino app.  Here's the output:
````
Running for 90 seconds...Free mem=35008
Changing URL to: http://rb-bremeneins-live.cast.addradio.de/rb/bremeneins/live/m, vol=47
Before stop...Free mem=35008
After stop...Free mem=35008
````
Note the truncation.  I increased the URL limit to 96 chars in the main app, but had a hardcoded 64-char limit in the HTTP handler.  I've pushed a change to fix that (1-liner) and now can see WebRadio.ino starting fine on that station...

tueddy, 2017-12-23:
Hi,

thanks for the very fast fix! The URL works fine now!

tueddy, 2017-12-23:
Hi,

sorry to reopen:
The commit 6c3064a fixes the url issue but breaks the MP3 decoder now!
You can reproduce it with MySong.mp3 from issue #23. This file does not play any longer.
Error message is:
stream.error != mad_Err_bufflen

Best regards 
Dirk

earlephilhower, 2017-12-24:
Ack, you're correct.  Very weird as only non-streaming ones seem to barf.  Even the crashing MP3 stops immediately.

If only there was an actual CI testing setup on real HW or an emulator.  Even the Arduino ESP8266 repo only compiles things, and can't test them.

earlephilhower, 2017-12-24:
Give the latest commit a go.  I've run the MP3 from SPIFFS and the WebRadio (well, for 320 seconds so far) without incident.

tueddy, 2017-12-25:
Works totally fine now, thank's!!!

closed 2017-12-25 by tueddy