earlephilhower/ESP8266Audio #22: WebRadio demo: crash with AudioOutputI2SNoDAC & change the volume

tueddy, 2017-12-18:
Hi,

i'm testing webradio with AudioOutputI2SNoDAC as output. i can play a mp3 stream, but when is change the volume the esp crashes with this staktrace:
(so far as i can see it's this line:
out->SetGain
)

Exception 28: LoadProhibited: A load referenced a page mapped with an attribute that does not permit loads
Decoding 18 results
0x4021157f: HandleVolume(WiFiClient*, char*) at C:\Users\D8310~1.CAR\AppData\Local\Temp\arduino_modified_sketch_145799/WebRadio.ino line 321
0x40211578: HandleVolume(WiFiClient*, char*) at C:\Users\D8310~1.CAR\AppData\Local\Temp\arduino_modified_sketch_145799/WebRadio.ino line 321
0x40211c05: loop at C:\Users\D8310~1.CAR\AppData\Local\Temp\arduino_modified_sketch_145799/WebRadio.ino line 384
0x40101db2: trc_NeedRTS at ?? line ?
0x401021f2: wDev_ProcessFiq at ?? line ?
0x40101f8c: wDev_ProcessFiq at ?? line ?
0x401021f2: wDev_ProcessFiq at ?? line ?
0x401068c4: millis at C:\Users\D.Carstensen\Documents\Arduino\hardware\esp8266com\esp8266\cores\esp8266/core_esp8266_wiring.c line 68
0x402179bb: loop_wrapper at C:\Users\D.Carstensen\Documents\Arduino\hardware\esp8266com\esp8266\cores\esp8266/core_esp8266_main.cpp line 57
0x402179bb: loop_wrapper at C:\Users\D.Carstensen\Documents\Arduino\hardware\esp8266com\esp8266\cores\esp8266/core_esp8266_main.cpp line 57
0x40212c90: WiFiClient::~WiFiClient() at C:\Users\D.Carstensen\Documents\Arduino\hardware\esp8266com\esp8266\libraries\ESP8266WiFi\src/WiFiClient.cpp line 92
0x40213328: WiFiServer::available(unsigned char*) at C:\Users\D.Carstensen\Documents\Arduino\hardware\esp8266com\esp8266\libraries\ESP8266WiFi\src/WiFiServer.cpp line 106
0x40211b7a: loop at C:\Users\D8310~1.CAR\AppData\Local\Temp\arduino_modified_sketch_145799/WebRadio.ino line 373
0x402179b8: loop_wrapper at C:\Users\D.Carstensen\Documents\Arduino\hardware\esp8266com\esp8266\cores\esp8266/core_esp8266_main.cpp line 57
0x40100710: cont_norm at C:\Users\D.Carstensen\Documents\Arduino\hardware\esp8266com\esp8266\cores\esp8266/cont.S line 109

Best regards
Dirk

earlephilhower, 2017-12-23:
My hunch is this is memory related.  I'm running with -NDEBUG and it's iffy even so.  Will take a look-see.

earlephilhower, 2018-01-22:
The new valgrind host-based tests and preallocating all of the app buffers seems to have allowed for pretty good runtimes now.  There may still be times when the ESP8266 OS/TCP go OOM, but there's not much I can do there.  Closing unless there's something else.  Thx! -EFP3

closed 2018-01-22 by earlephilhower

tueddy, 2018-01-22:
Thank's earlephilhower for taking care of this issue!