earlephilhower/ESP8266Audio #23: PlayMP3FromSPIFFS: Crash with specific MP3 - Problem with ID3 reader?

tueddy, 2017-12-22:
Hi,

the "PlayMP3FromSPIFFS" example works fine. I have replaced the sample mp3 with MySong.mp3.
After uploading i get this crash (decoded here):

Decoding 11 results
0x4020d0b4: AudioOutputI2SNoDAC::~AudioOutputI2SNoDAC() at C:\Users\D.Carstensen\Documents\Arduino\libraries\ESP8266Audio\src/AudioOutputI2SNoDAC.cpp line 73
0x4020c7b2: AudioFileSourceID3::read(void*, unsigned int) at C:\Users\D.Carstensen\Documents\Arduino\libraries\ESP8266Audio\src/AudioFileSourceID3.cpp line 46
0x4020e7bc: Print::printf(char const*, ...) at C:\Users\D.Carstensen\Documents\Arduino\hardware\esp8266com\esp8266\cores\esp8266/Print.cpp line 107
0x4020f2c4: SPIFFSFileImpl::flush() at C:\Users\D.Carstensen\Documents\Arduino\hardware\esp8266com\esp8266\cores\esp8266/spiffs_api.h line 315
0x4020c90a: AudioFileSourceID3::read(void*, unsigned int) at C:\Users\D.Carstensen\Documents\Arduino\libraries\ESP8266Audio\src/AudioFileSourceID3.cpp line 46
0x4020cd13: AudioGeneratorMP3::ErrorToFlow() at C:\Users\D.Carstensen\Documents\Arduino\libraries\ESP8266Audio\src/AudioGeneratorMP3.cpp line 266
0x4020cebf: AudioGeneratorMP3::GetOneSample(short*) at C:\Users\D.Carstensen\Documents\Arduino\libraries\ESP8266Audio\src/AudioGeneratorMP3.cpp line 266
0x40201809: mad_frame_init at C:\Users\D.Carstensen\Documents\Arduino\libraries\ESP8266Audio\src\libmad/frame.c line 100
0x4020c200: setup at C:\Users\D.Carstensen\Documents\Arduino\libraries\ESP8266Audio\examples\PlayMP3FromSPIFFS/PlayMP3FromSPIFFS.ino line 49
0x4020ed48: preloop_update_frequency() at C:\Users\D.Carstensen\Documents\Arduino\hardware\esp8266com\esp8266\cores\esp8266/core_esp8266_main.cpp line 57
0x40100710: cont_norm at C:\Users\D.Carstensen\Documents\Arduino\hardware\esp8266com\esp8266\cores\esp8266/cont.S line 109

Then i stripped the ID3 tags from the specific file and uploaded again. The MP3 plays fine now.
So i assume there is a problem in ID3 tag reader? I see a special character "ß" in the title..

files to reproduce:
[MySong.zip](https://github.com/earlephilhower/ESP8266Audio/files/1582858/MySong.zip)

Hope this helps to find the issue.

Thanks & best regards &

Merry Christmas
Dirk

tueddy, 2017-12-22:
stacktrace of ESP exception decoder:
[image]

earlephilhower, 2017-12-22:
Thanks for the example failing MP3, that should make it simpler to track down.  The stack trace is handy, but sometimes not so correct as there aren't well defined stack frames (to save memory, I assume) so the decoder doesn't always know the exact spot.

earlephilhower, 2017-12-22:
Wow, looks like that song's ID3 tag has both UNICODE and ASCII fields.  I probably assumed one or the other throughout...
````
00000000: 4944 3302 0000 0000 1209 5454 3200 0012  ID3.......TT2...
00000010: 0054 6974 656c 736f 6e67 202d 2045 6e64  .Titelsong - End
00000020: 6500 5450 3100 000f 004c 656f 204c 6175  e.TP1....Leo Lau
00000030: 7365 6d61 7573 0054 434d 0000 1b01 fffe  semaus.TCM......
00000040: 4800 6500 6900 6b00 6f00 2000 5200 fc00  H.e.i.k.o. .R...
00000050: 7300 7300 6500 0000 5441 4c00 004f 01ff  s.s.e...TAL..O..
00000060: fe46 006f 006c 0067 0065 0020 0031 0030  .F.o.l.g.e. .1.0
00000070: 0020 002d 0020 0057 0069 006c 006c 0020  . .-. .W.i.l.l. 
00000080: 006e 0069 0063 0068 0074 0020 0064 0072  .n.i.c.h.t. .d.r
00000090: 0061 0075 00df 0065 006e 0020 0073 0070  .a.u...e.n. .s.p
000000a0: 0069 0065 006c 0065 006e 0000 0054 524b  .i.e.l.e.n...TRK
000000b0: 0000 0500 362f 3600 5450 4100 0005 0031  ....6/6.TPA....1
000000c0: 2f31 0054 5945 0000 0600 3230 3134 0054  /1.TYE....2014.T
000000d0: 434f 0000 1000 5370 6f6b 656e 2026 2041  CO....Spoken & A
000000e0: 7564 696f 0043 4f4d 0000 1000 656e 6769  udio.COM....engi
000000f0: 5475 6e50 4741 5000 3000 0054 454e 0000  TunPGAP.0..TEN..
00000100: 1200 6954 756e 6573 2031 322e 362e 312e  ..iTunes 12.6.1.
00000110: 3235 0043 4f4d 0000 6800 656e 6769 5475  25.COM..h.engiTu
00000120: 6e4e 4f52 4d00 2030 3030 3030 3231 4520  nNORM. 0000021E 
00000130: 3030 3030 3030 3030 2030 3030 3031 3933  00000000 0000193
00000140: 4620 3030 3030 3030 3030 2030 3030 3046  F 00000000 0000F
00000150: 4134 4520 3030 3030 3030 3030 2030 3030  A4E 00000000 000
00000160: 3035 3643 3420 3030 3030 3030 3030 2030  056C4 00000000 0
00000170: 3030 3130 3235 4520 3030 3030 3030 3030  001025E 00000000
00000180: 0043 4f4d 0000 8200 656e 6769 5475 6e53  .COM....engiTunS
00000190: 4d50 4200 2030 3030 3030 3030 3020 3030  MPB. 00000000 00
000001a0: 3030 3032 3130 2030 3030 3030 3738 4320  000210 0000078C 
000001b0: 3030 3030 3030 3030 3030 3335 4237 3634  000000000035B764
000001c0: 2030 3030 3030 3030 3020 3030 3043 3238   00000000 000C28
000001d0: 3345 2030 3030 3030 3030 3020 3030 3030  3E 00000000 0000
000001e0: 3030 3030 2030 3030 3030 3030 3020 3030  0000 00000000 00
000001f0: 3030 3030 3030 2030 3030 3030 3030 3020  000000 00000000 
00000200: 3030 3030 3030 3030 0043 4f4d 0000 4700  00000000.COM..G.
00000210: 656e 6769 5475 6e65 735f 4344 4442 5f31  engiTunes_CDDB_1
00000220: 0035 3530 4141 3630 362b 3230 3436 3131  .550AA606+204611
00000230: 2b36 2b31 3530 2b36 3132 342b 3530 3833  +6+150+6124+5083
00000240: 322b 3935 3538 382b 3134 3733 3331 2b31  2+95588+147331+1
00000250: 3938 3632 3400 434f 4d00 001e 0065 6e67  98624.COM....eng
00000260: 6954 756e 6573 5f43 4444 425f 5472 6163  iTunes_CDDB_Trac
00000270: 6b4e 756d 6265 7200 3600 0000 0000 0000  kNumber.6.......
````

tueddy, 2017-12-22:
Hi,

thank's for looking into this issue! This is a real world example, i've imported the file from audio cd with iTunes on Mac.
stack trace of ESP-exception decoder is not reliable? Good to know! Debugging caps. are very limited here..
Let me know if you need more info or testing.
P.S.: I really like your library, very clean design, thanks! 

Dirk

earlephilhower, 2017-12-23:
Yes, the Exception decoder does its best, but there is no way, for example, that ID3::read() is ever going to call I2SDAC::anything.

Anyway, the problem is that I've implemented ID3V2.*4*.0 per the https://id3.org docs, but you've got a ID3V2.*2*.0 ("obsolete" per the website, but obviously still in use!).  It has a different format with 3-byte fields, not 4.  When the decoder sees it, it determines the ID3 header is 1.2MB in length (again, different format) and it times out the WDT going through it looking for "valid" frames.

I'll spend some time putting in a V2.2.0 (and check if v2.3.0 matches 2.4 or 2.2) this weekend...

earlephilhower, 2017-12-23:
Try the latest commit.  ID3V2.2.0 and non-unsync'd ID3 tags should now work.  Your sample file ran OK with this commit.

````
Sample MP3 playback begins...
ID3 callback for: Performer = 'Leo Lausemaus'
ID3 callback for: Album = 'Folge 10 - Will nicht drau�en s��h�'
ID3 callback for: Year = '2014'
MP3 stop, len==0
MP3 done
MP3 done
````

tueddy, 2017-12-23:
Wow, that did the trick!
i get the ID3 information now, the german sz/ß looks a bit wierd but i assume it's an ASCII / encoding / wrong codepage problem. I have to investigate.
Thanks
Dirk

closed 2017-12-23 by tueddy