earlephilhower/ESP8266Audio #25: Play 1Kbyte buffer PCM(wav) Audio

elahroodi, 2017-12-26:
Hi Dear earlephilhower. thanks for this useful library.
I want to play 1Kbyte buffer PCM(wav) Audio without any header received by UDP, transferd by android mobile microphone.
my problem is just to play 1kByte audio for exampled"unsigned char buf[1023]" filled with PCM data with this library.
can you suggest any example code or help me to do that?

Many thanks
Moradi

earlephilhower, 2017-12-26:
Howdy

I'm traveling so apologies for any typos.

There are two ways of doing what you want. The fast way and the" correct" way.

The fast way I'd to write the UDP listener in your loop(). Then you'll just create an I2SDAC or I2SNODAC object and do a consumesample() for every byte received.  You'll meet to do the setup calls to set the frequency and volume, but that's a one time thing and pretty simple.  At some point consumesample will return false and till need to delay() or wait until the next loop call to send more.

The correct way would be to write a AudioFileSourceUDP and a simple AudioGeneratirRaw.  It's pretty simple to do if you look at the existing code. Then you can use the standard framework stuff.

imranaalam, 2019-10-16:
@elahroodi i am facing the similar problem. can you share some guidlines to achieve it?

closed 2020-08-08 by earlephilhower