earlephilhower/ESP8266Audio #26: Undefined reference to `memcpy_P'

Yveaux, 2018-01-05:
After updating to include https://github.com/earlephilhower/ESP8266Audio/commit/d06dcf11231c811e4c6f39d130d2eed637b30c0f compilation on Arduino Esp8266 2.4.0-rc2 fails with:

Linking everything together...
...
C:\Users\xxx\AppData\Local\Temp\arduino_build_948531\libraries\ESP8266Audio\libmad\layer3.c.o: In function `III_imdct_l':

c:\Users\xxx\Documents\Arduino_ESP8266Audio\libraries\ESP8266Audio\src\libmad/layer3.c:2202: undefined reference to `memcpy_P'

C:\Users\xxx\AppData\Local\Temp\arduino_build_948531\libraries\ESP8266Audio\libmad\layer3.c.o: In function `III_imdct_s':

c:\Users\xxx\Documents\Arduino_ESP8266Audio\libraries\ESP8266Audio\src\libmad/layer3.c:2219: undefined reference to `memcpy_P'

collect2.exe: error: ld returned 1 exit status

Seems related to https://github.com/esp8266/Arduino/issues/3726

earlephilhower, 2018-01-05:
Thanks. The API seems to change on a daily basis. Did you try the #include change they suggested in that other issue report?  I remember seeing comments about adding extern C blocks in the headers, but IIRC, I was already manually doing that around the include itself.

Yveaux, 2018-01-05:
I attempted to implement the #include change, but realised it wasn't clear to me what was meant...
However, I wondered how you were able to compile with 2.4.0-rc2.

earlephilhower, 2018-01-05:
I always run GIT head, and the extern C changes they did were about a week ago.  Like I said, it changes on a daily basis and I've not re-built in a week or so...

earlephilhower, 2018-01-05:
I just pulled things down and did a clean rebuild in the Arduino IDE and am getting to no errors or warnings.

Can you try pulling git head?  Maybe -rc2 was broken off at a weird spot and some files were not checked in or something...

Yveaux, 2018-01-07:
Apparently 2.4.0 has just been released. I upgraded and build completes now without any issues!
Thanks!

closed 2018-01-07 by Yveaux