earlephilhower/ESP8266Audio #29: NodeMCU - PlayMP3FromSPIFFS.ino - undefined reference to `memcpy_P'

vksalian, 2018-01-10:
I get following error when I tried to install on NodeMCU.
**Sketch that I used:** PlayMP3FromSPIFFS.ino from examples
**Error:**
Arduino: 1.8.4 (Linux), Board: "NodeMCU 1.0 (ESP-12E Module), 80 MHz, 921600, 4M (3M SPIFFS)"

Archiving built core (caching) in: /tmp/arduino_cache_653286/core/core_esp8266com_esp8266_nodemcuv2_CpuFrequency_80,UploadSpeed_921600,FlashSize_4M3M_969f2f37cb57bce10f8f8743aaef779a.a
libraries/ESP8266Audio-master/libmad/layer3.c.o: In function `III_imdct_l':
Multiple libraries were found for "SD.h"
 Used: /home/vksalian/Software-Downloaded/Arduino/arduino-1.8.4/hardware/esp8266com/esp8266/libraries/SD
 Not used: /home/vksalian/Software-Downloaded/Arduino/arduino-1.8.4/libraries/SD
/home/vksalian/Software-Downloaded/Arduino/arduino-1.8.4/libraries/ESP8266Audio-master/src/libmad/layer3.c:2202: undefined reference to `memcpy_P'
libraries/ESP8266Audio-master/libmad/layer3.c.o: In function `III_imdct_s':
/home/vksalian/Software-Downloaded/Arduino/arduino-1.8.4/libraries/ESP8266Audio-master/src/libmad/layer3.c:2219: undefined reference to `memcpy_P'
collect2: error: ld returned 1 exit status
exit status 1
Error compiling for board NodeMCU 1.0 (ESP-12E Module).

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

earlephilhower, 2018-01-10:
2.4.0 was just released a couple days ago.  Can you please update your ESP8266 SDK/libs and try again?

Some 2.4.0-rc* versions did not have memcpy_P defined for some reason, which was rectified in the final release.

earlephilhower, 2018-01-18:
Closing as this was 2.4rc1 only.

closed 2018-01-18 by earlephilhower