earlephilhower/ESP8266Audio #30: Click on completion of playing wav file

dagnall53, 2018-01-10:
Earle, I think I have found a "bug" in your wav audio generator.
In my application I have a number of files that I play, and I discovered that some have a distinct "click" when they end.
To cut a long story short, the ones that "click" have metadata that describes what/who the sounds /artists are.  The attached zip has two identical audio files,  one has metadata, and clicks on completion, and the other does not, and is silent on completion. 

Easy answer is to remove all the metadata, but I have found another post from 2014 that seems to suggest that at least one other wav program was not accounting for this metadata and producing similar results by effectively having the wrong data sample length, it also provided some code to answer this issue (at least for the other software.?) . 
Hopefully this may help improve your library??, it looks to me that you have a slightly different way to get the sample length after "data" to the one in the "solution" example.. 

[F7 F8.zip](https://github.com/earlephilhower/ESP8266Audio/files/1618562/F7.F8.zip)

This was the site with the possible solution?
> https://gamedev.stackexchange.com/questions/71571/how-do-i-prevent-clicking-at-the-end-of-each-sound-play-in-openal
<

For those following these issues: this is a video of a test WITHOUT metadata.. https://youtu.be/Vd0HbV_MXVI
All the best
Dagnall

earlephilhower, 2018-01-10:
Yup, it's a simple bug.  Here's the comment and line in error:
https://github.com/earlephilhower/ESP8266Audio/blob/2e348d2b1f26745491f863c631bbaba8f0c68dad/src/AudioGeneratorWAV.cpp#L160
Just pushed a commit that I think should handle this.  Give it a whirl and let me know.

dagnall53, 2018-01-10:
I downloded the zip and copied ---wav.cpp and --wav.h and it seems to have solved the problem!..
I wish I actually understood the edits to the code..!

Many thanks.
I will be putting my "ESPMQTTRocnetSound" up on Github now.. (with properly attributed sound clips!..) 
Thanks
Dagnall

closed 2018-01-10 by dagnall53