earlephilhower/ESP8266Audio #31: Enhancement request: support ESP32

tueddy, 2018-01-10:
Hi earlephilhower,

ESP-32 is the successor of ESP-8266, has two i2s ports and bluetooth audio support.
It should be possible to support this platform too.
I would make a pull request but this is a bit out of my capabilities..

I have installed the ESP-32 support in Arduino IDE and tried to modify the sources. Some hints for porting:

Replace include declarations with this:

`#ifdef` ESP32
  #include <HTTPClient.h>
#else
  #include <ESP8266HTTPClient.h>
#endif`

AudioFileSourceSPIFFS.cpp & AudioOutputSPIFFSWAV.cpp:
`#ifdef ESP32
#include "SPIFFS.h"
#endif`

AudioOutputI2SNoDAC.cpp:

`#ifdef` ESP32
  #include "driver/i2s.h"
#else
  #include <i2s.h>
#endif`

I2S support requires some more modification,
here is a demo for playing wav via i2s on ESP-32:
http://www.iotsharing.com/2017/07/how-to-use-arduino-esp32-i2s-to-play-wav-music-from-sdcard.html

Replace the functions

`i2s_begin();
i2s_end();
i2s_set_rate();
i2s_write_sample();`

with
`i2s_driver_install();
i2s_set_pin();
i2s_set_sample_rates();
i2s_driver_uninstall();
i2s_set_sample_rates();
i2s_write_bytes();`

Porting MP3 decoder seems harder and i failed to get it to compile. But it should be possible. Here is a libmad port for ESP-32:
https://github.com/MrBuddyCasino/ESP32_MP3_Decoder/tree/master/components/mad

What do you think?
Best
Dirk

earlephilhower, 2018-01-11:
Cool, thanks.

I have an ESP-32 sitting idle on my desk now, actually.  Never installed the Arduino SDK for it since I got pulled into the ESP8266 one (I2S and the HTTPS server I submitted) and building and operating my new CNC machine.  Only so many free hours in the week, unfortunately.

As for the libmad, I'd honestly rather just start with the original sources.  On the ESP32 there's so much RAM and CPU it's almost trivial to build and run things like this.  Starting with a modified version is usually a bad beginning. :(  I spent a long time getting the AAC Helix code into working shape because it's no longer available from what's left of RealNetworks, so I ended up forking a repo of someone who had a copy running on another embedded CPU.  Bad idea, lots of things fixed/selected for use case X which don't work on your own.

Please feel free to fork the repo and start hacking.  I'm always happy to get PRs...means less work for me!

tueddy, 2018-01-16:
Hi earlephilhower,
i will send you the pull request tomorrow. Got some things to work, playing wav on dac work's on both systems now..
Best Dirk

tueddy, 2018-01-17:
Hi,

just submitted PR #33..

earlephilhower, 2018-01-18:
Closing this as we're all set now.

closed 2018-01-18 by earlephilhower

dagnall53, 2019-01-26:
Just found this thread. see my question in #163 , but I think its possible that the "driver/is2.h" is not sufficiently universal..? I put my full paths to i2s.h in the other comment in case it helps..
Sorry for multiple cross-linked posts.. I will stop exploring and await a reply...