earlephilhower/ESP8266Audio #32: Is Gpio02 (D4) used in I2SNoDAC ?

dagnall53, 2018-01-13:
I am very happy using the no dac solution, I added a 1k resistor in the transistor base and can upload new sketches easily (with noise whilst uploading!).  

However, I have an issue: I see that D4 /GPIO02 is oscillating fast with circa 150us period, (exactly as it does with the I2SDAC code). 
Is this something left over from the I2SDAC code, or deliberate and essential? 
If just "left over", can you stop using it please so I can use the pin   ;-)
 

(still waiting eagerly for some mixing code..)

earlephilhower, 2018-01-15:
Howdy,

The I2S hardware takes over output control on the BLCK, LRCLK and I2SDATA pins.  So you can't use them for outputs, sorry.

You *may* be able to use them as inputs.  After starting the I2S, set the pindirection to INPUT and try reading an external value.  Depending on the way the pads are built, you *might* be able to read the input values just fine (but I haven't tried nor looked at the chip's IO driver schematic).

If you do try this, can you let me know if it works or fails? I can add a note in the readme to that effect, saving grief for the next person looking to get back a pin or two...

Thx
-EFP3

As for the mixing, it's just fallen into the backlog.  I've been reworking the stdc library and that's been taking my spare computer time pretty effectively.  It's now working and in a PR, so this is back in the queue.  Turns out it's not as simple as I'd thought, but still relatively straightforward.

dagnall53, 2018-01-16:
Hi again.. 
You are correct: I set `pinMode(2, INPUT_PULLUP);` in my setup and the port is now available (and works) as an input to sense when the pin is pulled low. (I assume input will also work, but have not tested that explicitly). 
This also turns off the LED on the ESP8266, (which was part of the reason I wanted to do it). 
A simple test for the port being pulled low in my "loop" code then works exactly as expected: 
`if (!digitalRead(2)){BeginPlay("/F5.wav");} //Play sound effect `

Many thanks
Sorry the mixing is not as simple as expected.. Would limiting all the samples to the same samplerate make it easier? That would be a small price to pay to be able to get the effect right.

earlephilhower, 2018-01-18:
Updated the readme w/this info, thanks!

For the mixer, I was already assuming the same sample rates.  The difficulty (well, just work but not hard thought) is that the mixer needs to generate a stub AudioOutput for each input it sinks which would then pass-through to the mixer w/some add'l info identifying which branch it is.  Housekeeping, not groundbreaking.

closed 2018-01-18 by earlephilhower

dagnall53, 2018-01-30:
Just thought I would add a little more: 
I did another check to see if D4 could be reused as an Output, and that was NOT successful. 
It seems the NoDAC code outputs a very fast clock on this pin if the pin is an output. 
SO: Summary, You can (re)use D4 as an Input with Pull up with NoDAC, but that's about it. 
(My codes do not use plain "Input" so I have not tested that)

earlephilhower, 2018-01-30:
Yup, that's pretty much it.  I2SNoDAC uses all the pins because the HW itself thinks it is driving an I2SDAC.  In fact, while testing I end up doing it a lot.  The sound is garbage, but you can kind of make out what's being played as a function of the volume of noise.  So any I2S pins other than the I2SOUT can be used as INPUT or INPUT_PULLUP, but none as outputs.

bigujun, 2018-02-17:
I made some tests and the pins (D4/GPIO2 and D8/GPIO15) work perfectly as output at the same time as I2SNoDAC, but you have to set them as output after initialize AudioOutputI2SNoDAC.

earlephilhower, 2018-02-17:
Very nice to know!  So after it's started, do a pinMode(D4, OUTPUT); pinMode(D8, OUTPUT); and then you can use standard digitalWrites on them?

Thx!