earlephilhower/ESP8266Audio #34: ESP32 Support Update/Fixes

earlephilhower, 2018-01-19:
Tracking bug for ESP32 issues brought up by @tueddy, @reaper7, others, and myself.

**I2S non-blocking**
The I2S functions now hard-block, meaning that once you start writing samples it will stick in side the mp3::loop and never return to your main app until the mp3 ends or an error occurs.  The i2s_write_bytes seems to always block, even when the timeoput is set to 0.

Workaround is to make a worker thread who only pushes samples passed in to it.  The thread will set a flag the main arduino code can look at (bool willBlock) and either say "sorry, out of space" or put the data into the worker thread's to-be-xmitted fifo.  Ugly.

**Enable onboard DAC**
. Remove the global, hard coded i2s settings now #if ESP32'd.
. Add a constructor, #ifdef ESP32'd, I2SDAC::I2SDAC(bclk,lrclk,dout) => Set these as the I2S Pins
. Modify I2SDAC()::I2SDAC() to default to something sane on the ESP32 (existing settings?)
. Add a constructor, #ifdef ESP32, I2SDAC::I2SDAC(INTERNAL_DAC_[0|1] ), which routes the I2S data to internal 8-bit DAC
.Modify I2SNoDac::I2SNODAC() to use sane defaulrs on ESP32
. Add a constructor, #ifdef ESP32'd, I2SNODAC::I2SNODAC(bclk,lrclk,dout) => Set these as the I2S Pins

earlephilhower, 2018-01-22:
So I2S nonblocking was a red herring.  This was a configuration problem in the I2S codec, so the ugly #1 is not needed.

Still need to enable onboard DAC as an option,. as well as allowing both I2s interfaces to drive a stereo delta-sigma amp...

earlephilhower, 2018-01-26:
Latest commit allows specifying the ports, and is significantly cleaned up from prior revs for ESP32 and ESP8266.

Built-in DAC still not implemented.

reaper7, 2018-01-26:
I see that AudioOutputI2S have **bool builtInDAC** nice :)

but esp32 has two possibility of direct output:
first _I2S_MODE_DAC_BUILT_IN_
and second (as people wrote - most better) _I2S_MODE_PDM_

like we can see in this project:
https://github.com/MrBuddyCasino/ESP32_MP3_Decoder/blob/master/components/audio_renderer/audio_renderer.c#L30

so, config mode and communication_format must be initialized differently 
and in this case simple bool is insufficient.
Maybe instead will be better use predefined values in header like:
```
#define I2S_MODE_EXTERNAL_DAC 0 //default value for external DAC
#define I2S_MODE_INTERNAL_DAC 1
#define I2S_MODE_INTERNAL_PDM 2
```
and replace bool with some wider variable:
AudioOutputI2S::AudioOutputI2S(int port, **uint8_t i2sMode**)

---
also pushing sample are different for MODE_DAC_BUILT_IN and MODE_PDM

earlephilhower, 2018-01-30:
I need to read a bit more, but it looks like PDM===Sigma Delta, so there's no need for the NoDAC on the device at all.  That would be cool, and since you have 2 I2S interfaces you could produce stereo output, to boot.

Of bigger concern is that several open bugs are reporting that simple things like setting the I2S rate don't work properly, so I'm going to try and focus on that when I get time first.  There's not a lot of space for my own code to go wrong here (all I do is call the built-in i2s_set_rate function), so I may need to workaround a bug or something in the SDK/HDK here.  Or maybe the audio PLL is not safe to use when WiFi is operating (noise induced on its power supply causing it to lock onto a bad frequency)?  Maybe just avoid the issue and always drive off of the digital logic clock instead.

reaper7, 2018-03-21:
@earlephilhower - tell mi, current lib version support PDM mode on esp32? or still only builtinDAC ?

earlephilhower, 2018-03-21:
Still only the built in DAC, sorry.  Been busy writing a new SSL driver for the ESP8266 and other I2S stuff, sorry.  Always happy for a PR, though. :)

reaper7, 2018-03-21:
this is **AudioOutputI2S.cpp** with hardcoded version of PDM mode (if I am not wrong with consume sample, it's too easy :) someone has to check it out )
Switched by (un)comment #define ESP32PDM in line 29

the differences are in:
- settings .mode and .communication_format
- ConsumeSample

initialize from main sketch like for buildDAC option ( AudioOutputI2S(0, **true**); )

```
/*
  AudioOutputI2S
  Base class for I2S interface port
  
  Copyright (C) 2017  Earle F. Philhower, III

  This program is free software: you can redistribute it and/or modify
  it under the terms of the GNU General Public License as published by
  the Free Software Foundation, either version 3 of the License, or
  (at your option) any later version.

  This program is distributed in the hope that it will be useful,
  but WITHOUT ANY WARRANTY; without even the implied warranty of
  MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
  GNU General Public License for more details.

  You should have received a copy of the GNU General Public License
  along with this program.  If not, see <http://www.gnu.org/licenses/>.
*/

#include <Arduino.h>
#ifdef ESP32
  #include "driver/i2s.h"
#else
  #include <i2s.h>
#endif
#include "AudioOutputI2S.h"

#define ESP32PDM

AudioOutputI2S::AudioOutputI2S(int port, bool builtInDAC, int use_apll)
{
  this->portNo = port;
  this->i2sOn = false;
  this->builtInDAC = builtInDAC;
#ifdef ESP32
  if (!i2sOn) {
    if (use_apll == APLL_AUTO) {
      // don't use audio pll on buggy rev0 chips
      use_apll = APLL_DISABLE;
      esp_chip_info_t out_info;
      esp_chip_info(&out_info);
      if(out_info.revision > 0) {
        use_apll = APLL_ENABLE;
      }
    }
    i2s_config_t i2s_config_dac = {
#ifdef ESP32PDM
      .mode = (i2s_mode_t)(I2S_MODE_MASTER | I2S_MODE_TX | I2S_MODE_PDM),
#else
      .mode = (i2s_mode_t)(I2S_MODE_MASTER | I2S_MODE_TX | (builtInDAC ? I2S_MODE_DAC_BUILT_IN : 0)),
#endif
      .sample_rate = 44100,
      .bits_per_sample = I2S_BITS_PER_SAMPLE_16BIT,
      .channel_format = I2S_CHANNEL_FMT_RIGHT_LEFT,
#ifdef ESP32PDM
      .communication_format = (i2s_comm_format_t)(I2S_COMM_FORMAT_I2S | I2S_COMM_FORMAT_I2S_MSB),
#else
      .communication_format = (i2s_comm_format_t)((builtInDAC ? 0 : I2S_COMM_FORMAT_I2S) | I2S_COMM_FORMAT_I2S_MSB),
#endif
      .intr_alloc_flags = ESP_INTR_FLAG_LEVEL1, // high interrupt priority
      .dma_buf_count = 8,
      .dma_buf_len = 64,   //Interrupt level 1
      .use_apll = use_apll // Use audio PLL
    };
    Serial.printf("+%d %p\n", portNo, &i2s_config_dac);
    if (i2s_driver_install((i2s_port_t)portNo, &i2s_config_dac, 0, NULL) != ESP_OK) {
      Serial.println("ERROR: Unable to install I2S drives\n");
    }
    if (builtInDAC) {
      i2s_set_pin((i2s_port_t)portNo, NULL);
      i2s_set_dac_mode(I2S_DAC_CHANNEL_BOTH_EN);
    } else {
      SetPinout(26, 25, 22);
    }
    i2s_zero_dma_buffer((i2s_port_t)portNo);
  } 
#else
  (void) use_apll;
  if (!i2sOn) {
    i2s_begin();
  }
#endif
  i2sOn = true;
  mono = false;
  bps = 16;
  channels = 2;
  SetGain(1.0);
  SetRate(44100); // Default
}

AudioOutputI2S::~AudioOutputI2S()
{
#ifdef ESP32
  if (i2sOn) {
    Serial.printf("UNINSTALL I2S\n");
    i2s_driver_uninstall((i2s_port_t)portNo); //stop & destroy i2s driver
  }	
#else
  if (i2sOn) i2s_end();
#endif
  i2sOn = false;
}

bool AudioOutputI2S::SetPinout(int bclk, int wclk, int dout)
{
#ifdef ESP32
  if (builtInDAC) return false; // Not allowed

  i2s_pin_config_t pins = {
    .bck_io_num = bclk,
    .ws_io_num = wclk,
    .data_out_num = dout,
    .data_in_num = I2S_PIN_NO_CHANGE
  };
  i2s_set_pin((i2s_port_t)portNo, &pins);
  return true;
#else
  (void) bclk;
  (void) wclk;
  (void) dout;
  return false;
#endif
}

bool AudioOutputI2S::SetRate(int hz)
{
  // TODO - have a list of allowable rates from constructor, check them
  this->hertz = hz;
#ifdef ESP32
  i2s_set_sample_rates((i2s_port_t)portNo, AdjustI2SRate(hz)); 
#else
  i2s_set_rate(AdjustI2SRate(hz));
#endif
  return true;
}

bool AudioOutputI2S::SetBitsPerSample(int bits)
{
  if ( (bits != 16) && (bits != 8) ) return false;
  this->bps = bits;
  return true;
}

bool AudioOutputI2S::SetChannels(int channels)
{
  if ( (channels < 1) || (channels > 2) ) return false;
  this->channels = channels;
  return true;
}

bool AudioOutputI2S::SetOutputModeMono(bool mono)
{
  this->mono = mono;
  return true;
}

bool AudioOutputI2S::begin()
{
  return true;
}

bool AudioOutputI2S::ConsumeSample(int16_t sample[2])
{
  MakeSampleStereo16( sample );

  if (this->mono) {
    // Average the two samples and overwrite
    uint32_t ttl = sample[LEFTCHANNEL] + sample[RIGHTCHANNEL];
    sample[LEFTCHANNEL] = sample[RIGHTCHANNEL] = (ttl>>1) & 0xffff;
  }
  uint32_t s32 = ((Amplify(sample[RIGHTCHANNEL]))<<16) | (Amplify(sample[LEFTCHANNEL]) & 0xffff);
#ifdef ESP32
#ifndef ESP32PDM
  if (builtInDAC) {
    int16_t l = Amplify(sample[LEFTCHANNEL]) + 0x8000;
    int16_t r = Amplify(sample[RIGHTCHANNEL]) + 0x8000;
    s32 = (r<<16) | (l&0xffff);
  }
#endif
  return i2s_push_sample((i2s_port_t)portNo, (const char *)&s32, 0);
#else
  return i2s_write_sample_nb(s32); // If we can't store it, return false.  OTW true
#endif
}

bool AudioOutputI2S::stop()
{
#ifdef ESP32
  i2s_zero_dma_buffer((i2s_port_t)portNo);
#endif
  return true;
}
```
but for final/selectable version we have to think about second parameter from
```
AudioOutputI2S(int port, bool builtInDAC, int use_apll)
```
because bool isn't enough for modes (external)I2S/BUITINDAC/PDM
it should be similar to APPL enum like:
```
public:
AudioOutputI2S(int port=0, int output_mode=EXTERNAL_I2S, int use_apll=APLL_DISABLE);
enum : int { EXTERNAL_I2S = 0, DAC_BUILT_IN = 1, PDM = 2 };
.
.
protected:
int output_mode;
```

earlephilhower, 2020-08-08:
The ESP32 port was contributed a couple years back, so this should be OK.  Any specific issues, please open a new bug listing them.

closed 2020-08-08 by earlephilhower