earlephilhower/ESP8266Audio #35: Problem with I2S Delta-Sigma DAC

Eduardo-F, 2018-01-23:
Hillo Earle,
Thank you for your great job.

To play a MP3 file from PROGMEM, I'm using a ESP32 board connected to an external DAC on pins 26(BCK), 25( LCK) and 22 (DIN). 
It works very fine and very good quality.

The problem is when I tried the I2S Delta-Sigma DAC. Nothing sound is played.
About hardware: the external DAC is removed, and I follow your schematic to drive a speaker (Base is connected to pin 25 in ESP32).

On Arduino IDE the following code compile without errors:

#include <Arduino.h>
#include "AudioFileSourcePROGMEM.h"
#include "AudioGeneratorMP3.h"
#include "AudioOutputI2SNoDAC.h"
#ifdef ESP32
    #include <WiFi.h>
#else
    #include <ESP8266WiFi.h>
#endif

// enigma.mod sample from the mod archive: https://modarchive.org/index.php?request=view_by_moduleid&query=42146
#include "fileMP3.h"

AudioGeneratorMP3 *mp3;
AudioFileSourcePROGMEM *file;
AudioOutputI2SNoDAC *out;

void setup()
{
  WiFi.mode(WIFI_OFF); //WiFi.forceSleepBegin();
  Serial.begin(115200);
  delay(1000);
  file = new AudioFileSourcePROGMEM(fileMP3_mp3, sizeof(fileMP3_mp3) );
  out = new AudioOutputI2SNoDAC();
  mp3 = new AudioGeneratorMP3();
  mp3->begin(file, out);
}

void loop()
{
  if (mp3->isRunning()) {
    if (!mp3->loop()) mp3->stop();
  } else {
    Serial.printf("MP3 done\n");
    delay(1000);
  }

the code is the same than used with DAC, only the AudioOutputI2SNoDAC object is used instead of the AudioOutputI2SDAC.

What am I doing wrong?
Could you help me?

earlephilhower, 2018-01-23:
Howdy,

> To play a MP3 file from PROGMEM, I'm using a ESP32 board connected to an external DAC on pins 26(BCK), 25( LCK) and 22 (DIN). It works very fine and very good quality.
> The problem is when I tried [...]

What you'll need to do is connect the same DIN (22)pin that was going to your external I2SDAC to the transistor base, not pin (25).  You may want to double-check the I2SNoDac.cpp sources as there's a hardcoded configuration specified there (which will be changed at some point soon to configurable options).  You can find the pin mappings there, just in case it's not the same as the I2SDAC (it should be, but I've not tested it personally).

-EFP3

Eduardo-F, 2018-01-24:
Hello Earle,

Thanks for your reply.
I have connected the DIN (22)pin to the transistor base, now the speaker sounds like a continuous data stream (as an old data modem), but nothing similar to the music :(

Regarding I2SNoDac.cpp, I only see :

i2s_pin_config_t pin_config = {
    .bck_io_num = 26, //this is BCK pin
    .ws_io_num = 25, // this is LRCK pin
    .data_out_num = 22, // this is DATA output pin
    .data_in_num = -1   //Not used

It is OK.

Any suggestion?

gamecompiler, 2018-01-24:
Please let us know what "connection" possibilities you have. Because for example mine has an external mute port which you shouldnt let it float. On some breakout ports are a lot of pins which may have to be connect to gnd for example if not connected internally.

I had the problem that my DAC breakout board had an extra 5V to 3.3V regulator on Board (Sounds excactly like you deschribed - like an old modem when connected wrong). You may also want to check that. 

What kind of I2S Dac do you have?

Sorry for my bad english.

Eduardo-F, 2018-01-24:
Hi,
Thanks for your reply.
It works fine when I use an external DAC (https://es.aliexpress.com/item/Interface-I2S-PCM5102-DAC-Decoder-GY-PCM5102-I2S-Player-Module-For-Raspberry-Pi/32807703240.html).

My problem is when ONLY a speaker is connected throug a transistor (changing the output with AudioOutputI2SNoDAC();

About your english, I'm sure is better than mine ;-)

earlephilhower, 2018-01-24:
Hi @Eduardo-F .  I've not tried the NoDac version on the ESP32 yet, but I run it lots on the ESP8266.  So I think there may be some tweaking on the 32 left to do.

I see 3 differences in the setup structs from dac to nodac.  Can you cut-n-paste the dac.cpp version over the nodac for the "i2s_config_t i2s_config_dac" struct?  The missing "I2S_COMM_FORMAT_I2S" may cause it to do something weird to the bitstream.

Everything else, on quick inspection, seems reasonable.

Eduardo-F, 2018-01-24:
Hi Earl,

AudioOutputI2SDAC.cpp:

int i2s_num = 0; // i2s port number
i2s_config_t i2s_config_dac = {
     .mode = (i2s_mode_t)(I2S_MODE_MASTER | I2S_MODE_TX),
     .sample_rate = 36000,
     .bits_per_sample = I2S_BITS_PER_SAMPLE_16BIT,
     .channel_format = I2S_CHANNEL_FMT_RIGHT_LEFT,
     .communication_format = (i2s_comm_format_t)(I2S_COMM_FORMAT_I2S | I2S_COMM_FORMAT_I2S_MSB),
     .intr_alloc_flags = ESP_INTR_FLAG_LEVEL1, // high interrupt priority
     .dma_buf_count = 8,
     .dma_buf_len = 64,   //Interrupt level 1
     .use_apll = true // Use audio PLL
    };

AudioOutputI2SNoDAC.cpp:

i2s_config_t i2s_config = {
     .mode = (i2s_mode_t)(I2S_MODE_MASTER | I2S_MODE_TX | I2S_MODE_DAC_BUILT_IN),
     .sample_rate = 44100,
     .bits_per_sample = I2S_BITS_PER_SAMPLE_16BIT, /* the DAC module will only take the 8bits from MSB */
     .channel_format = I2S_CHANNEL_FMT_RIGHT_LEFT,
     .communication_format = (i2s_comm_format_t)I2S_COMM_FORMAT_I2S_MSB,
     .intr_alloc_flags = 0, // default interrupt priority
     .dma_buf_count = 8,
     .dma_buf_len = 64,
     .use_apll = 0
    };

earlephilhower, 2018-01-24:
Err, I meant to cut-n-paste that bit fromt he DAC to NODAC in your own code and recompile. :)

The settings being slightly different may be the problem.  We know DAC sends the bits out the right way.

earlephilhower, 2018-01-24:
I just pushed a version of I2SNoDac which uses the exact same bits to setup the I2S interface as the DAC version.  I also wired my ESP32 up during lunch to a 3904 + speaker and tested it, and it's running fine in the example I've tried (PlayAACFromProgmem).  Can you pull this version, re-check your wiring, and rebuild/run?

tueddy, 2018-01-24:
Hi Earle,

you are the audio pin guy!
Can confirm now the NoDAC work's also on ESP32! On Pin22 i get output. Tested with StreamMP3FromHTTP:

...Connecting to WiFi
Connected
Running for 5201 ms...
STATUS(buffer) '2' = 'Refilling buffer'
STATUS(mp3) '565' = 'Decoding error 'bad main_data_begin pointer' at byte offset 0'
STATUS(mp3) '565' = 'Decoding error 'bad main_data_begin pointer' at byte offset 246'
Running for 6245 ms...
METADATA(ICY) 'StreamTitle' = 'Stevie Wonder With Paul Mccartney - Ebony & Ivory'
Running for 7280 ms...

Stevie Wonder sings just soooooo sloooow and has a terrible sound (against an external DAC). Maybe the sample rate is wrong or is it of missing SPI RAM buffer? Sounds a bit of wrong speed of my 90ties BRAUN Atelier P1 vinyl record player..

Anyway, good work!
Best Dirk

earlephilhower, 2018-01-24:
In the DAC.cpp file, line 43, try changing ".use_apll =1" to ".use_apll = 0"  and rebuilding/running.

Could be there's some add'l setup needed to get the audio PLL synchronized and warmed up which the SDK doesn't take care of automatically.  This change will tell it to just divide the system clock, which will be off some small %age (1-2%, not usually an issue).

tueddy, 2018-01-24:
Changing use_apll seems to make no hearable difference due to bad quality. Sounds like it's playing in half speed or just too slow. Using latest ESP32 git revision. tested with predefined sample URL "/http://streaming.shoutcast.com/80sPlanet?lang=en-US", MP3 stream stops after 15 seconds (buffer underflow) and cannot start again until reset, using a local URL here in germany works "http://rb-bremeneins-live.cast.addradio.de/rb/bremeneins/live/mp3/64/stream.mp3" but still too slow..

earlephilhower, 2018-01-25:
Can you folks help narrow down the failing case?  Does DAC and NODAC give the same slow output?  Do the samples that aren't HTTP related (Play*FromSPIFFS/PROGMEM) play OK?  Is the sound lower frequency (like playing that record player too slow) or is it stuttery where the pitch seems OK but the playback takes 2x as long because each sgement is played 2x, etc.

Eduardo-F, 2018-01-25:
Hi Earle,
I just compiled and run the PlayAACFromProgmem example.
The getting results are the same in my ESP32 board:

Ussing:
#include "AudioOutputI2SDAC.h"
AudioOutputI2SDAC *out;
out = new AudioOutputI2SDAC();
and my external DAC + speaker it works fine.

When I replace the previous with:
#include "AudioOutputI2SNoDAC.h"
AudioOutputI2SNoDAC *out;
out = new AudioOutputI2SNoDAC();
with transistor + speaker it sounds like a data stream.

earlephilhower, 2018-01-25:
@Eduardo-F That's the one I actually tried first, and for me it seemed fine, except for possibly something weird about the playback rate like @tueddy said (but I've not verified it, and it's subtle not 1/2 speed or something).

Your noise problem may be power supply related.  You could try adding a large electrolytic between GND and +5V, and a 1K resistor between pin 22 and the transistor base to help clear things up.  With the higher power needs of the ESP32, that noise may be way higher than the 8266.

tueddy, 2018-01-25:
Hi,

I'll try it later this evening, i assume the low samplerate comes from the http connection (I use no external buffer Ram)..
Dirk

Eduardo-F, 2018-01-25:
Hi Earle,

Now it WORKS FINE. 

I decided download your library again (because I did some changes to do tests) and now everything works fine (DAC, NoDAC, AAC and MP3 also). Really I don't know what the problem was.

About the capacitor and resistor, the noise is reduced, but works fine without them.

I'm appreciate very much your time and job.

Thank you very much.

tueddy, 2018-01-25:
Hi,
tested again with PlayAACFromPROGMEM, DAC/NoDAC both plays with correct speed. 
Best Dirk

earlephilhower, 2018-01-25:
Thanks to you @Eduardo-F and @tueddy for the updates!

So iteems like it's isolated to the web streaming where this is not quite right.  The I2S interface is sending out the bits at the right speed, but we're just not feeding it bits fast enough...

closed 2018-01-26 by Eduardo-F