earlephilhower/ESP8266Audio #38: stereo mp3 stream: every sample gets played twice

h3ndrik, 2018-01-26:
Using AudioGeneratorMP3 and AudioOutputI2S connected to a MAX98357A
stereo streams sound distorted and only playing at half speed. (mono is okay)

For example if you compare the two streams (first is stereo, second is mono):
"http://streaming.shoutcast.com/80sPlanet?lang=en-US"
"http://swr-swr1-bw.cast.addradio.de/swr/swr1/bw/mp3/64/stream.mp3"

Code:
```
  file = new AudioFileSourceICYStream(URL);
  file->RegisterMetadataCB(MDCallback, (void*)"ICY");
  buff = new AudioFileSourceBuffer(file, preallocateBuffer, preallocateBufferSize);
  buff->RegisterStatusCB(StatusCallback, (void*)"buffer");
  out = new AudioOutputI2S(I2S_NUM_0, false);
  out->SetPinout(12, 13, 23);
  mp3 = new AudioGeneratorMP3(preallocateCodec, preallocateCodecSize);
  mp3->RegisterStatusCB(StatusCallback, (void*)"mp3");
  mp3->begin(buff, out);
```

earlephilhower, 2018-01-26:
Howdy, and thanks for the fixes.

It looks like I have a regression in the Buffer class.  I recently added an optimization for when an error occurred in the stream to skip ahead faster to the next frame sync.  I could swear I ran it fine on the PC in my test suite, but maybe not.

Anyway, can you please remove the buffer from the chain (be sure to update to " mp3->begin(file, out);") and give it a whirl on the ESP32?  I do my work on the ESP8266 and right now I'm able to run the HTTPStream example w/o a buffer and it's coming out just fine w/aDAC in stereo now (except for IP hiccups every 15 secs to minute).

That would confirm that the buffer is where the regression is.

Thx
-EFP3

h3ndrik, 2018-01-27:
it's not the buffer. behavior is the same both with and without buffer.
(sorry i forgot to mention: this is on an esp32.)

i'll do some more tests tomorrow. maybe i can unplug an esp8266 somewhere or test with a stereo file from spiffs.

thanks for this library!

h3ndrik, 2018-01-28:
update: PlayMP3FromSPIFFS plays fine. ICYStream 80sPlanet does not.

gamecompiler, 2018-01-28:
Got the same problem. The ICYStream is really really laggy. When listening to 80sPlanet on my computer everything work fine.

Maybe it works with a connected SPI RAM as Buffer better?
(Funny fact - because you changed the title i was unable to comment over email)

h3ndrik, 2018-01-29:
sorry for that.
i don't think a too small buffer is the issue here. first of all it works at correct speed on an esp8266 - also without buffer.
and there are streams which do work.

but im unable to pinpoint the location of the problem. maybe the sample format is set up incorrectly? or it's the way the samples are written?
and i've still to debug if the i2s has samples available all the time.
but why would it work with some audio sources and not work with other ones?

i swapped icystream with httpstream, swapped the mad decoder with the helix one and fiddled a little bit with the way the i2s is set up, but i get the same result each time.

earlephilhower, 2018-01-29:
I think there's a different mechanic going on with the Arduino WiFiClient+TCP backend on the ESP32 vs. the ESP8266, and that this is causing a lot of the issues seen here.  I've also been told the APLL can also be busted even in silicon that says it's Rev 1, and if so that would account for any slowdown.

Another thing is the streamer reports a long string on errors, but it does not have the "jump to next MP3 frame" I added in my local branch.  And those errors cascade (error at byte 501...error at byte 502...etc.) now so the Serial.println() or even the WebRadio's strcpy() spends tons of time just outputting that instead of sending I2S data.  So you get hiccups/repeated data (on the ESP8266 it silences on underflow, on the ESP32 it seems not to do so).

It definitely needs more investigation!

h3ndrik, 2018-01-31:
interesting. i'll have another try with the apll after this;)

yes, seems the read from the streamer is causing this.

i've added a few debug printfs to measure the number of samples the i2s writes in the mp3 loop and most of the time that number is small. so the buffer has still samples left. and after a new frame is decoded - which triggers a read() in icystreamer - there are lots of samples written.
i think the i2s buffer is completely consumed at the time the readInternal() returns.

i added a huge buffer to the input and then thought it couldn't be the input. but as i see, everytime we read from the buffer, the buffer also reads from its source. which makes the buffer kind of pointless... if it isn't able to return its buffered content while the source is taking too long to return something.

h3ndrik, 2018-01-31:
and it's not the string copying but the actual readBytes itself.

icy's readInternal() takes about 70-170 µs to get to the final

`int ret = stream->readBytes(reinterpret_cast<uint8_t*>(data), len);`

and this takes like 330000 µs !

earlephilhower, 2018-01-31:
Great debugging.  That was what I was suspecting.

We refill the buffer in *nonblocking* mode after any user read, but from your debug it seems like there is a different mechanic going on here.

Maybe try replacing the "len" in that line with "(nonblocking?stream->available():len)"?

h3ndrik, 2018-01-31:
found it! just strip the "Bytes" from the mentioned line and it works fine.

i think readBytes(...) is the one from the Stream class and copies the bytes one by one whereas read(...) with the same signature is the one from WiFiClient and just reads a chunk of data in a few µs.

(note: my first measurements seem to be a little exaggerated, either i had other wifi problems or it was a zero too much, but too long anyways)

now i just have to find out what the esp8266 does and prepare a pull request.

commit 05a78dce (2018-02-01): Make stream reads orders of magnitude faster (fixes #38) (#44)

Fix by @h3ndrik 
* Use correct read for stream data on the ESP32 (fixes #38)
* Adjust types for read()
closed 2018-02-01 by earlephilhower