earlephilhower/ESP8266Audio #41: PlayMP3FromSPIFFS crashes

lucalcin, 2018-01-27:
Hi,
the example **PlayMP3FromSPIFFS** crashes when it calls the function:

out = new AudioOutputI2SNoDAC();

I'm using a Wemos D1 Mini, latest ESP8266Audio library, ESP8266 2.4.0
This is the stack:

Exception (0):
epc1=0x4000e25d epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000
ctx: cont 
sp: 3fff0360 end: 3fff05a0 offset: 01a0
>>>stack>>>
3fff0500:  00000007 00000007 00000001 4022da15  
3fff0510:  00000000 3ffeef28 3fff1a0c 4020d34a  
3fff0520:  00000000 00000000 3fff1a0c 402105c4  
3fff0530:  00000000 3ffeef28 3fff1a0c 4020d498  
3fff0540:  00000000 3ffeef28 3ffef4ac 4020c3cd  
3fff0550:  3ffe8cc0 00000000 000003e8 feefeffe  
3fff0560:  3fff18ec 3fff192c 3fff1864 3fff18bc  
3fff0570:  feefeffe feefeffe feefeffe 3ffef56c  
3fff0580:  3fffdad0 00000000 3ffef564 4020f278  
3fff0590:  feefeffe feefeffe 3ffef580 40100710  
<<<stack<<<
 ets Jan  8 2013,rst cause:2, boot mode:(1,6)
 ets Jan  8 2013,rst cause:4, boot mode:(1,6)
wdt reset

What I wrong?

renegadeandy, 2018-01-30:
I also get this problem

earlephilhower, 2018-02-01:
https://github.com/earlephilhower/ESP8266Audio/commit/f1916051ecd515b48c031102e6ca6efeda8cdc8f

Should have fixed this.  An uninitted variable (which for some reason I was always lucky and came up as 0 on my runs!) caused the i2s interface not to be created, and then the code tried to set the playback rate with a NULL variable.

earlephilhower, 2018-02-01:
Actually, there were a couple other class members not being initted properly.  Those were just fixed, so pull the new head if you can and try again.

bigujun, 2018-02-14:
I was having a similar problem and found it to be the Lame mp3 codec, recoded the mp3 file with FFmpeg and the problem was solved.

earlephilhower, 2018-02-14:
@bigujun Can you attach a failing MP3?  I can run it locally and check if it crashes w/the git head, and if so it's much easier to debug these things with a known, repeatable failing case.  The ones I've thrown at it haven't had issues, but I couldn't say how they were encoded.

bigujun, 2018-02-15:
@earlephilhower I have attached two mp3 files and the crash log, 

1. pno-cs-56kbps-mono-Audacity(CRASH).mp3
2. pno-cs-56kbps-mono-Reencoded.mp3

The first mp3 crashes and was exported with Audacity, the second is the same audio reencoded with soundKonverter and works fine.
Every audio exported with Audacity seems to crash.
I'm using PlayMP3FromSPIFFS with I2SNoDAC example and esp12E.

Tanks for your work, it's a great lib. 

[pno-cs.zip](https://github.com/earlephilhower/ESP8266Audio/files/1726214/pno-cs.zip)

earlephilhower, 2018-02-15:
Thanks for the upload.  I'm able to reproduce it as a watchdog timer error, which unfortunately makes it harder to tell really where it's gone bad, but indicates an infinite loop.

earlephilhower, 2018-02-15:
This was an ID3 decode error.  Audacity sent an extended header, while the other one did not.  Just tested and both versions now pass.  Please check with your other MP3s.  Thx!
-EFP3

commit 01d9daeb (2018-02-15): Fix ID3v2 decoding, avoid infinite loop

The ID3 reader was incorrectly skipping extended headers and was
cutting off the last 10 bytes of the last ID3 entry. Correct per
http://id3.org specs.

Fixes #41
closed 2018-02-15 by earlephilhower