earlephilhower/ESP8266Audio #42: Software NoDac Crashes

renegadeandy, 2018-01-30:
I am trying the No DAC implementation, exactly as per the example : https://github.com/earlephilhower/ESP8266Audio#software-i2s-delta-sigma-dac-ie-playing-music-with-a-single-transistor-and-speaker

It crashes with the following error:

> 
> Exception (0):
> epc1=0x4000e25d epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000
> 
> ctx: cont 
> sp: 3fff0250 end: 3fff0460 offset: 01a0
> 
> >>>stack>>>
> 3fff03f0:  00000007 00000007 00000001 4022d08d  
> 3fff0400:  3ffeedfc 00000000 3fff14ac 4020cb42  
> 3fff0410:  00000000 00000000 3fff14ac 4020fc8c  
> 3fff0420:  3ffeedfc 00000000 3fff14ac 4020cc90  
> 3fff0430:  feefeffe 00000000 3fff14ac 4020c1c6  
> 3fff0440:  3fffdad0 00000000 3ffef430 4020eab0  
> 3fff0450:  feefeffe feefeffe 3ffef440 40100710  
> <<<stack<<<
> 
>  ets Jan  8 2013,rst cause:2, boot mode:(3,6)
> 
> load 0x4010f000, len 1384, room 16 
> tail 8
> chksum 0x2d
> csum 0x2d
> v4ceabea9
> ~ld
> 
>  ets Jan  8 2013,rst cause:4, boot mode:(3,6)
> 
> wdt reset
> load 0x4010f000, len 1384, room 16 
> tail 8
> chksum 0x2d
> csum 0x2d
> v4ceabea9
> ~ld
> rl l⸮⸮| ⸮l⸮|⸮l⸮b|⸮⸮⸮⸮r⸮b⸮b⸮⸮nn⸮lnn⸮⸮⸮bp⸮⸮lrlrlp⸮n⸮⸮l⸮⸮bn⸮|l⸮⸮b⸮⸮nn⸮ l⸮⸮l`⸮nnl`nr⸮⸮⸮nb⸮lr⸮⸮nb⸮l⸮⸮⸮b⸮l`⸮⸮n⸮
> Exception (0):
> epc1=0x4000e25d epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000
> 
> ctx: cont 
> sp: 3fff0250 end: 3fff0460 offset: 01a0
> 
> >>>stack>>>
> 3fff03f0:  00000007 00000007 00000001 4022d08d  
> 3fff0400:  3ffeedfc 00000000 3fff14ac 4020cb42  
> 3fff0410:  00000000 00000000 3fff14ac 4020fc8c  
> 3fff0420:  3ffeedfc 00000000 3fff14ac 4020cc90  
> 3fff0430:  feefeffe 00000000 3fff14ac 4020c1c6  
> 3fff0440:  3fffdad0 00000000 3ffef430 4020eab0  
> 3fff0450:  feefeffe feefeffe 3ffef440 40100710  
> <<<stack<<<
> 
>  ets Jan  8 2013,rst cause:2, boot mode:(3,6)
> 
> load 0x4010f000, len 1384, room 16 
> tail 8
> chksum 0x2d
> csum 0x2d
> v4ceabea9
> ~ld
> 
>  ets Jan  8 2013,rst cause:4, boot mode:(3,6)
> 
> wdt reset
> load 0x4010f000, len 1384, room 16 
> tail 8
> chksum 0x2d
> csum 0x2d
> v4ceabea9
> ~ld

Once it crashes, the esp resets, and it fails again! Help!

earlephilhower, 2018-02-01:
Can you please decode your stack trace?  The examples aren't very good and don't check for things like the file not being found or stream not connecting and will crash like this.  W/o the EspExceptionDecode output it's not possible to see if this is related to the I2S or not.

FWIW I just tested a NoDac HTTP stream for some ESP32 fixes (to make sure ESP8266 wasn't affected) and it ran absolutely fine for me, so I'm not quite convinced it's the NoDac crashing here...

Thx

renegadeandy, 2018-02-01:
Hi,

This is the decoded stack trace:

`Decoding 7 results
0x4022d08d: i2s_begin at /Users/user/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.0/cores/esp8266/core_esp8266_i2s.c line 233
0x4020cb42: AudioOutputI2S::SetRate(int) at /Users/user/Documents/Arduino/libraries/ESP8266Audio/src/AudioOutputI2S.cpp line 120
0x4020fc8c: AudioOutputI2SNoDAC::SetOversampling(int) at /Users/user/Documents/Arduino/libraries/ESP8266Audio/src/AudioOutputI2SNoDAC.cpp line 48
0x4020cc90: AudioOutputI2SNoDAC::AudioOutputI2SNoDAC(int) at /Users/user/Documents/Arduino/libraries/ESP8266Audio/src/AudioOutputI2SNoDAC.cpp line 35
0x4020c1c6: setup at /Users/user/Documents/Arduino/testsound/testsound.ino line 15
0x4020eab0: loop_wrapper at /Users/user/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.0/cores/esp8266/core_esp8266_main.cpp line 57
0x40100710: cont_norm at /Users/user/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.0/cores/esp8266/cont.S line 109
`

renegadeandy, 2018-02-01:
It is definitely crashing on this line:

 out = new AudioOutputI2SNoDAC();

earlephilhower, 2018-02-01:
Have you pulled the latest code (w/in about 6 days)?  There was an unitialized boolean (i2s_on) in the constructor on the ESP32 which would cause it to skip the i2s_begin() call, leading to a crash like this since the i2s structs weren't initted when it tried to set the frequency in the constructor.

https://github.com/earlephilhower/ESP8266Audio/commit/f1916051ecd515b48c031102e6ca6efeda8cdc8f

renegadeandy, 2018-02-01:
Doing a git log in the directory i cloned the library into shows:

`commit e7c2aba2aa41c118d46b8069872f463440549c1f (HEAD -> master, origin/master, origin/HEAD)
Author: h3ndrik <hendrik+dev@xd0.de>
Date:   Fri Jan 26 16:51:49 2018 +0100

    don't use audio pll on buggy rev0 chips (#40)
    
    as ESP32_MP3_Decoder does

commit f1916051ecd515b48c031102e6ca6efeda8cdc8f
Author: h3ndrik <hendrik+dev@xd0.de>
Date:   Fri Jan 26 16:50:43 2018 +0100

    initialize member variables (#39)
    
    if AudioOutputI2S is instanciated in a function (like in the examples) the member variables are not initialized to a default value.

`

So I think I have this. Yes. This latest version has my problem in it.

earlephilhower, 2018-02-01:
Perfect.  Can you paste your exact source code as well?  It's a lot easier to catch these things if I can reproduce your environment exactly.  If it's playing a WAV or MP3, I obviously don't need that and will sub in something equivalent.

renegadeandy, 2018-02-01:
Sure:

`#include <Arduino.h>
#include "AudioFileSourceSPIFFS.h"
#include "AudioGeneratorMP3.h"
#include "AudioOutputI2SNoDAC.h"

AudioGeneratorMP3 *mp3;
AudioFileSourceSPIFFS *file;
AudioOutputI2SNoDAC *out;
void setup()
{
  Serial.begin(115200);
  delay(1000);
  SPIFFS.begin();
  Serial.printf("1\n");
  file = new AudioFileSourceSPIFFS("/w1.mp3");
  Serial.printf("2\n");
  out = new AudioOutputI2SNoDAC();
  Serial.printf("3\n");
  mp3 = new AudioGeneratorMP3();
  Serial.printf("4\n");
  mp3->begin(file, out);
  Serial.printf("5\n");
}

void loop()
{
  if (mp3->isRunning()) {
    if (!mp3->loop()) mp3->stop();
  } else {
    Serial.printf("MP3 done\n");
    delay(1000);
  }
}`

On an ESP8266E, ESP-12-E. 160 Mhz,4M, v2 prebuilt, MSS 1460, unstable.

earlephilhower, 2018-02-01:
Hmmm... I just replaced the MP3 w /pno-cs.mp3 and ran without any incident at all here.  I'm running GIT head which has 2 commits from @h3ndrik for speeding up HTTP streams (but no changes to anything in this sketch).  Multiple runs give the same successful result:

````
1
2
3
4
5
MP3 stop, len==0
MP3 done
MP3 done
...
````

Only other thing I can think of is there's something off in the official 2.4.0 release of the Arduino package.  Can you get the nightly build for it, or just git clone the esp8266/Arduino repo?  There were some changes (correctness, not crash-related IIRC) from 2.4.0 to now) in the I2S code.

The line mentioned in the i2s crash is in the middle of resetting the hardware, not even a memory access, so I'm not so sure about the validity of the trace...this is not uncommon, unfortunately, as the stack dump is more aspirational than definitive. :(

I'm running the same settings as you, but git head of the Arduino SDK, on a ESP-12 (wemos d1 mini pro), @160MHZ, Debug-level=none.

earlephilhower, 2018-02-01:
That said, I did see a couple spots where I might have just been getting lucky on my own tests.  I've pushed a commit just now that sets some add'l default values which might cause what you're seeing.

Can you pull git head and give it another go on the failing system?

renegadeandy, 2018-02-01:
OK! At version:

`eb67766 (HEAD -> master, origin/master, origin/HEAD) Add more default values to I2S output`

I hear my audio track! It is seriously crackily, and very very quiet but it's there.

Is the fuzzy audio to be expected? How can I fix that?  What about cranking the volume up?
I removed the 270 ohm resistor on base which somebody suggested , this made it ever so slightly louder but no clearer.

earlephilhower, 2018-02-01:
Not sure what you mean by fuzzy.  The 1T amplifier sound for most folks, myself included, is actually amazingly good.  If there's lots of power suply noise you may get a high pitched buzz in concert with the audio output.  Some decoupling caps near the transistor and speaker can help minimize this (professional delta-sigma dacs also integrate a LPF on the output as well to minimize any higher F harmonics).

As for volume, there's a couple things to check:  Is the xistor EBC wired properly?  W/o the beta-gain from the B the resultant output will be very low, and swapping E and C could do it.  Also, make sure you've got the full 5V into the speaker, not a 3.3V source.  Higher V = bigger amplitude (esp. since w/the BJT transistor you can't go below 0.7V).  An 8-ohm or 4-ohm speaker will sound better than a 32-ohm or piezo due to the Ron of the xistor.

There's also a out->SetGain(f); call.  Pass in a float between 0.1 and 4.0 and it'll pre-amplify before shifting out.  This handles sources which weren't normalized properly before transcoding.

renegadeandy, 2018-02-01:
by fuzzy i mean i can hear hissing, popping and crackling from the speaker.

```
Pins CBE from left to right flat side facing out are:

C-> Negative Terminal
B -> I2S Data
E -> ESP Ground

```
This produces quiet audio.

```
Swapping C & E   : 
C -> ESP Ground
E -> Negative Terminal
```

This produces a tiny bit louder audio but the xistor gets seriously hot, like awful awful hot. Ive popped it back to normal(The xistor still gets awful hot, suppose that's normal here)

Using `out->SetGain(1.0);` resulted in 'default low volume audio'.

Using `out->SetGain(2.0);` resulted in slightly (x 1)  clearer audio, maybe a tiny bit louder than 1.0

Using `out->SetGain(3.0);` resulted in slightly (x 2) clearer audio, maybe a tiny bit louder than 2.0

Using `out->SetGain(4.0);` resulted in tiny crackles resembling next to nothing from the original audio.

renegadeandy, 2018-02-04:
Any ideas?

earlephilhower, 2018-02-05:
What exact xistor do you have?  Is is possible you've got a 3906 PNP and not a 3904 NPN, which would leave the xistor never really turning off?  There should not be much power dissipated in the transistor, and it shouldn't be getting very hot.

closed 2020-08-08 by earlephilhower