earlephilhower/ESP8266Audio #43: No playback with ESP32 from SPIFFS

rfestag, 2018-01-31:
I'm having issues getting this library to play MP3 files on my ESP32 (Adafruit Feather, and am connecting it to a Teensy Prop Shield amp, which accepts i2s). My understanding of the schematics implies that I should be using DAC1 (that is routed to the same pin that goes to the prop shield). However, it seems like I'm not even getting to the point where that should matter. It seems like the library is having a hard time reading the file at all. My code is:

`
#include <Arduino.h>
#include <WiFi.h>
#include <SPI.h>
#include <SPIFFS.h>
#include "AudioFileSourceSPIFFS.h"
#include "AudioFileSourceID3.h"
#include "AudioGeneratorMP3.h"
#include "AudioOutputI2S.h"

AudioGeneratorMP3 *mp3;
AudioFileSourceSPIFFS *poweron;
AudioFileSourceSPIFFS *hum;
AudioFileSourceID3 *poweronid3;
AudioFileSourceID3 *humid3;
AudioOutputI2S *out;

void MDCallback(void *cbData, const char *type, bool isUnicode, const char *string)
{
  (void)cbData;
  Serial.printf("ID3 callback for: %s = '", type);

  if (isUnicode) {
    string += 2;
  }
  
  while (*string) {
    char a = *(string++);
    if (isUnicode) {
      string++;
    }
    Serial.printf("%c", a);
  }
  Serial.printf("'\n");
  Serial.flush();
}
void setup() {
  Serial.begin(115200);
  delay(1000);      // sanity check delay
  while (!Serial) ; // wait for serial port open

  SPIFFS.begin();
    Serial.printf("Listing directory: %s\n", "/");

    File root = SPIFFS.open("/");
    if(!root){
        Serial.println("Failed to open directory");
        return;
    }
    if(!root.isDirectory()){
        Serial.println("Not a directory");
        return;
    }
    File file = root.openNextFile();
    while(file){
            Serial.print("  FILE: ");
            Serial.print(file.name());
            Serial.print("  SIZE: ");
            Serial.println(file.size());
            file = root.openNextFile();
     }

  poweron = new AudioFileSourceSPIFFS("/POWERON.mp3");
  poweronid3 = new AudioFileSourceID3(poweron);
  poweronid3->RegisterMetadataCB(MDCallback, (void*)"ID3TAG");
  out = new AudioOutputI2S(0, false);
  mp3 = new AudioGeneratorMP3();
  mp3->begin(poweronid3, out);
  Serial.println("Done Setup");
}
void loop() {
  //Update sound
  if (mp3->isRunning()) {
    if (!mp3->loop()) mp3->stop();
  }
  else {
    Serial.println("MP3 Done");
    delay(1000);
  }
}
`
Which produces output like this:

`
Listing directory: /
  FILE: /POWERON.mp3  SIZE: 24987
  FILE: /SWING1.mp3  SIZE: 12934
  FILE: /HUM.mp3  SIZE: 275054
  FILE: /CLASH3.mp3  SIZE: 8502
  FILE: /CLASH2.mp3  SIZE: 17161
  FILE: /SWING5.mp3  SIZE: 13508
  FILE: /SWING3.mp3  SIZE: 15702
  FILE: /POWEROFF.mp3  SIZE: 17453
  FILE: /CLASH5.mp3  SIZE: 8501
  FILE: /CLASH1.mp3  SIZE: 10326
  FILE: /LOCKUP.mp3  SIZE: 75683
  FILE: /SWING2.mp3  SIZE: 11633
+0 0x3ffca67c
Done Setup
MP3 stop, len==0
MP3 Done
...
`
The MP3 stop, len==0 happens very quickly, sooner than I might have expected for the file being played. So, that leaves me with two questions:
1) Has anyone else had success with ESP32 and SPIFFS?
2) How do I tell the library to output to DAC1 instead of DAC0?

I know ESP32 support is relatively new/buggy. I'd be happy to help wherever I can, I'm just not sure where to start. Thanks!

earlephilhower, 2018-01-31:
Howdy,

SPIFFS on the ESP32 has not been tested by me, sorry, but it's pretty standard and run lots on the ESP8266.  Is there a special way to upload to SPIFFS?  The IDE plugin I'm using says the ESP32 isn't supported, only the ESP8266?!

As a sanity, since it looks like you know your way around the IDE the simplest check is to create a SPIFFS object in your sketch, then "char x; while (file->read(&x, 1)) Serial.printf("%02X", (int)x));" to dump it as a hex stream.  A quick check of the start/end should verify if it's working properly.

For DAC1, that's been requested in another issue but not yet done.  The "bool internalDac" option will transform into an enum and be DAC0/DAC1/PDM.  You can hack it right now by editing the I2S.cpp file and changing the i2s_config_t to point to DAC1 instead of DAC0.

Thx
-EFP3

rfestag, 2018-01-31:
Thanks for the quick reply. I was able to get this plugin to work: https://github.com/me-no-dev/arduino-esp32fs-plugin. 

I did some quick sanity checks, and it looks like I was wrong...the file does seem to be read, but it acts like the playback is almost immediate. I'm also not seeing exactly how to tell it which DAC to output to. Is that the portNum parameter? It looks like I can't use the built-in DAC on DAC1 (only DAC0...if I set portNum to 1, I get an error at runtime). If I tell it not to use built-in DAC, it doesn't seem to matter which portNum I specify. I don't see anything specifically on the i2s_config_t I can set to tell it which DAC to use. Am I  missing something?

earlephilhower, 2018-01-31:
I'm in the middle of some ESP8266 SSL debug, so haven't tried this, but I believe the line
> i2s_set_dac_mode(I2S_DAC_CHANNEL_BOTH_EN);
sends values out on both dacs (right and left channels).  There's an enum for each channel you can replace it with IIRC.

rfestag, 2018-02-01:
It turns out I was being dumb, and thought the device I was connecting to took i2s (based on some related samples, but not specific to the prop shield). Apparently I am talking directly to an amp, not an i2s device.

I'm mostly interested in this library for the MP3 decoding. I'm not seeing one, but is there an output that supports direct output an amp via the DAC, but not as I2S? The NoDAC seems to be the opposite...I2S over a non-DAC pin?

earlephilhower, 2018-02-02:
The I2S/I2sNoDAC output drives an internal i2s circuit.

On the ESP8266 this goes right out to a set of pins that you can then use to either (a) run a real I2S DAC, or (b) run a 1-transistor amplifier connected to that same set of pins using the I2SNoDAC class (via a software delta-sigma DAC algorithm) to give an equivalent 8-12 bit DAC performance.

On the ESP32 you have (a) and (b), but also (c) Use the I2S block to drive an on-chip 8-bit DAC which is connected to DAC0/DAC1 pins.  So to drive a real analog amp, use the I2S with internal DAC.  There's a constructor parameter (2nd boolean IIRC) to enable this.

rfestag, 2018-02-02:
Gotcha, makes sense. Yea, I did see that parameter (and have been using it in my latest code), I just assumed I was misunderstanding something.

I've done a bit more digging, and it turns out hat there is somehow an issue reading the files from SPIFFS. When I tested earlier, I just assumed it was taking a long time to print the file, so I took the fact that there was output when reading the file as "good enough". Unfortunately, it looks like attempts to read 10 bytes within the AudioFileSourceID3::read is actually getting 0 bytes. Attemps to just read byte-by-byte (as in the example you had above) just show repeating "EC" over and over. 

I added some debugging to verify the files exist, are available and have non-zero sizes  before reading them. All of those seem to imply there should be data to read. What is strange is that the example code you show above runs forever,(always returning a non-zero length), but the read in AudioFileSourceID3 always returns 0. 

In any case, it seems like I'll have to dig more into why the SPIFFS data is wrong...maybe it is just an issue with the plugin and how the files were transferred. Perhaps a better option, for now, is to come up with an HTTP sketch that downloads the files to SPIFFS.

rfestag, 2018-02-02:
Update: I went back to the plugin, and it turns out there is some incompatibility introduced in the ESP32 libraries after 12/19/2017. I reverted back to the last commit on that day, and things are sort of working (I'm getting sound playing, and the ID3 metadata callbacks are actually running now). So its good to know I'm not crazy. I think my issues are more like #47 now (playback seems to be too fast). I think we can close this issue (the core problem is resolved, now the problem is simply the lack of internal DAC support on ESP32).

earlephilhower, 2018-02-02:
@rfestag That's both funny and sad.  I hope you've opened a bug in the Arduino-esp32 repo about it!

I'll close this and we'll track on #47.

closed 2018-02-02 by earlephilhower

rfestag, 2018-02-27:
Yea, here are the steps I followed  (temporary hack until @me-no-dev updates the esp32fs-plugin):

```
cd ~/Arduino/hardware/espressif/esp32
#I determined the checkout by running `git log`, and finding the last commit on 12/19/1027
git checkout 70d0d4648766cea047613062ed50a9ecfc9de31f
```

You can find more information in this ticket, and others that reference it:
https://github.com/me-no-dev/arduino-esp32fs-plugin/issues/6

rfestag, 2018-02-27:
Actually, I looked at it again this morning, and a better option (for now) is to keep the arduino-esp32 code at latest, and insetad manually replace the mkspiffs tool with 0.2.2. Choose the appropriate download for your system from here (I used mkspiffs-0.2.2-arduino-esp32-linux64.tar.gz):
https://github.com/igrr/mkspiffs/releases/tag/0.2.2

Untar it and replace the mkspiffs tool located here:
~Arduino/hardware/espressif/esp32/tools/mkspiffs

I just tried and it works.

earlephilhower, 2018-02-27:
Thanks, @rfestag .  When I get back from traveling I'll update the readme with this info.  

Do you know if there is an open bug in there repo for this?  It seems like they have a non working SPIFFS.

rfestag, 2018-02-28:
No worries. There are actually a number of tickets (all closed, but none actually resolving the problem). For whatever reason, the maintainers of arduino-esp32 removed the mkspiffs tool, and expect you to download it and put it in the right place yourself (without documenting that). I have a discussion with them right now seeing if we can find a better way. Anyone new to it would have no way of knowing what to do without looking at closed issues.

https://github.com/espressif/arduino-esp32/issues/1022

earlephilhower, 2018-03-01:
Got it.  I've added some lines to the README.md which point to the pull request that seems to fix it.

rfestag, 2018-03-04:
Update: The pull request was  merged today. I just pulled it in, re-uploaded my files, and reloaded my sketch, and it appears to work. The proper solution now is:

```
cd ~/Arduino/hardware/espressif/esp32 #Or wherever you have it installed
git pull #Update to the latest
cd tools
python get.py #On my system, I have python3 installed by default, so I had to run python2.7 get.py
#Re-upload files using the new mkspiffs that is installed
#Then reload your sketch
````

earlephilhower, 2018-03-05:
Thanks for the update, I'll change the README.md right now...

gamecompiler, 2018-03-05:
How do i use mkspiffs for the Arduino environment? What settings should i set for compilation?  Like `SPIFFS_OBJ_NAME_LEN`. Thats where i cant keep up.

rfestag, 2018-03-06:
@gamecompiler, you can use the arduino-esp32fs-plugin (https://github.com/me-no-dev/arduino-esp32fs-plugin). Just install it as described, and make sure you have the latest mkspiffs (should be 0.2.2). Then you should be able to add a data directory in your sketch directory, and click Tools->ESP32 Sketch Data Upload plugin in the Arduino IDE to upload the files. If you also had to update your arduino-esp32, you may also need to re-upload your sketch.

rickyanwar, 2021-01-15:
[image]
[image]

where I plugin pin for esp32 ,I using rx0 rx2 is not working, in serial monitor i have no errors, using LM386 and speaker 0.5 watt

playground, 2021-04-17:
I would like to be able to write the response from “https://translate.google.com/translate_tts?ie=UTF-8&tl=en_US&client=tw-ob&q=hello%20world” to SPIFFS as mp3 file and be able to play it back.  Do you have an example how to save the mp3 content to SPIFFS?

I have tried this but audio is not playing
```
  http.begin(tts.c_str());
  // Send HTTP GET request
  int httpResponseCode = http.GET();
  if (httpResponseCode>0) {
     Serial.print("HTTP Response code: ");
     Serial.println(httpResponseCode);
     int len = http.getSize();
     uint8_t buff[len] = {0};
     WiFiClient *stream = http.getStreamPtr();
   
     while(http.connected() && (len > 0 || len == -1)) {
        size_t size = stream->available();

        if(size) {
            int c = stream->readBytes(buff, ((size > sizeof(buff)) ? sizeof(buff) : size));
            delay(1);
         }
        delay(1);
     } 
     File file = SPIFFS.open("/helloworld.mp3", "w"); 
     file.write(buff, c);
     file.close();  
   }
   else {
     Serial.print("Error code: ");
     Serial.println(httpResponseCode);
   }
   http.end();
```

rfestag, 2021-04-17:
Saving the HTTP response to SPIFFS really is out of scope for this issue (and the repo in general), you may be better served asking the Arduino ESP32 forums for help with that. At a minimum, you'll probably get more traction from a new issue as opposed to responding to a closed one.

All of that said, a few thoughts that may help:
1. Is there a reason you need to save the MP3 to SPIFFS? If you simply want to play back the response (not save it for future use), you could look at the AudioFileSourceHTTPStream (although you may need to switch to an http URL instead of https, I can't speak to whether this that particular source supports https).
2. I'm not sure your code would work as expected. If I recall correctly, each call to `readBytes` will return the number of bytes read. Ignoring the fact that I think `c` is out of scope by the time it gets down to your `file.write`, it may only represent the number of bytes from the very last read. not the total bytes. 
   1. Given all of that, I suspect you'd be better served to simply write straight to SPIFFS each set of bytes. This would mean having potentially a smaller buffer, reading some number of bytes into it, and immediately writing that to SPIFFS. Rinse and repeat until you have read all bytes.

playground, 2021-04-17:
Hi @rfestag thanks for your response.  

There is cut/paste error in the above code.  

Nonetheless the issue I'm seeing is that response I get from http.GET request is different from the one from Postman.   The response from Postman does work.

[image]

response from http.GET
[image]

The output from readBytes don't match up with output from Postman.  Do you see if there is problem with the way it is reading the bytes?  I will try ESP32 forums too