earlephilhower/ESP8266Audio #47: ESP32 / M5Stack

alanroche2015, 2018-02-01:
Hi,

Wondering if you might have some pointers about a problem i'm having playing WAV and MP3 on a ESP32,  I think i have the audio working (or at least some sound) but it appears to be playing way to fast. Would you have an pointers as to what it might be or where i should start looking ? 

Thanks 

here is my code i've been testing with..

`
#include <Arduino.h>
#ifdef ESP32
#include <WiFi.h>
#include "SPIFFS.h"
#else
#include <ESP8266WiFi.h>
#endif

#include "AudioFileSourceID3.h"
#include "AudioGeneratorMP3.h"
#include "AudioOutputI2SNoDAC.h"
#include "AudioOutputI2S.h"
#include "AudioFileSourceSD.h"

#include "AudioGeneratorWAV.h"

#include <M5Stack.h>

AudioGeneratorWAV *mp3;
AudioFileSourceSD *file;
AudioOutputI2S *out;
AudioFileSourceID3 *id3;

void setup()
{
M5.begin();

M5.Speaker.setVolume(3);

Serial.begin(115200);
delay(1000);
//SPIFFS.begin();
Serial.printf("Sample MP3 playback begins...\n");
file = new AudioFileSourceSD("/slow.wav");

out = new AudioOutputI2S(0,true);

mp3 = new AudioGeneratorWAV();
mp3->begin(file,out);

}

void loop()
{
M5.update();
if (mp3->isRunning()) {
M5.update();

if (!mp3->loop()) {
  mp3->stop();
}
} else {
Serial.printf("MP3 done\n");
delay(1000);
}
}
`

earlephilhower, 2018-02-01:
Could very well be.  I've not hooked up anything to the internal DAC yet as I don't have a handy amplifier near my desk.

Does it sound like the pitch is shifted (i.e. it's playing all the samples, but far too fast), or is the pitch correct but it's only, say, playing the first 5 seconds of a 20 second clip?

Pitch shift probably indicates that I need to adjust the internal i2s_rate.  Only playing the beginning of a longer one is something more fundamental.

alanroche2015, 2018-02-01:
I think it’s the shift rate, the audio file I have been testing with fades out at the end and I can definetly hear it fading out.  If I try a longer file.. say 3 minutes it plays for about 25 seconds and fades out at the end. If I use a shorter file it plays for maybe 5 seconds or so. 

  I also tried to slow the file down on the computer and play it thinking it might speed it up to a normal rate but that didn’t sound right at all right.  

Anything you think I can try ? I could also give you the audio file and record the the actual output from the M5 if that would help ?

earlephilhower, 2018-02-02:
Edited the title to reflect what it seems the problem is now.

rfestag, 2018-02-03:
The issue for me was the use of APLL. I changed the config i2s_config_t in AudioOutputI2S.h:

```
    i2s_config_t i2s_config_dac = {
      .mode = (i2s_mode_t)(I2S_MODE_MASTER | I2S_MODE_TX | (builtInDAC ? I2S_MODE_DAC_BUILT_IN : 0)),
      .sample_rate = 44100,
      .bits_per_sample = I2S_BITS_PER_SAMPLE_16BIT,
      .channel_format = I2S_CHANNEL_FMT_RIGHT_LEFT,
      .communication_format = (i2s_comm_format_t)((builtInDAC ? 0 : I2S_COMM_FORMAT_I2S) | I2S_COMM_FORMAT_I2S_MSB),
      .intr_alloc_flags = ESP_INTR_FLAG_LEVEL1, // high interrupt priority
      .dma_buf_count = 8,
      .dma_buf_len = 64//,   //Interrupt level 1
      //.use_apll = use_apll // Use audio PLL
    };
```

And the following sketch works:
```
#include <Arduino.h>
#include <SPIFFS.h>
#include <FS.h>
#include <WiFi.h>
#include "AudioFileSourceSPIFFS.h"
#include "AudioFileSourceID3.h"
#include "AudioGeneratorMP3.h"
#include "AudioGeneratorMP3a.h"
#include "AudioGeneratorWAV.h"
#include "AudioOutputI2S.h"

AudioGeneratorMP3 *mp3;
AudioFileSourceSPIFFS *mp3Test;
AudioFileSourceID3 *mp3TestId3;
AudioOutputI2S *out;

#define AMP_EN 13

void MDCallback(void *cbData, const char *type, bool isUnicode, const char *string)
{
  (void)cbData;
  Serial.printf("ID3 callback for: %s = '", type);

  if (isUnicode) {
    string += 2;
  }
  
  while (*string) {
    char a = *(string++);
    if (isUnicode) {
      string++;
    }
    Serial.printf("%c", a);
  }
  Serial.printf("'\n");
  Serial.flush();
}
void setup() {
  Serial.begin(115200);
  delay(2000);      // sanity check delay
  pinMode(AMP_EN, OUTPUT);
  digitalWrite(AMP_EN, HIGH);  // enable access to AMP on Teensy Prop Shield

  if (!SPIFFS.begin()) {
    Serial.println("Failed to mount SPIFFS");
  }
  Serial.printf("Listing directory: %s\n", "/");

  File root = SPIFFS.open("/");
  if(!root){
    Serial.println("Failed to open directory");
    return;
  }
  if(!root.isDirectory()){
    Serial.println("Not a directory");
    return;
  }
  File file = root.openNextFile();
  while(file){
    Serial.print("  FILE: ");
    Serial.print(file.name());
    Serial.print("  SIZE: ");
    Serial.println(file.size());    
    file = root.openNextFile();
    //uint8_t x; while (file.read(&x, 1)) Serial.printf("%02X", x);
  }
  

  mp3Test = new AudioFileSourceSPIFFS("/pno-cs.mp3");
  
  mp3TestId3 = new AudioFileSourceID3(mp3Test);
  mp3TestId3->RegisterMetadataCB(MDCallback, (void*)"ID3TAG");
  out = new AudioOutputI2S(0, true);
  out->SetOutputModeMono(true);
  out->SetGain(0.3);
  
  mp3 = new AudioGeneratorMP3();
  mp3->begin(mp3TestId3, out);
  Serial.println("Done Setup");
}
int currentLed = 0;
void loop() {
  //Update sound
  if (mp3->isRunning()) {
    if (!mp3->loop()) {
      mp3->stop();
    }
  }
  else {
    Serial.println("Done");
    delay(1000);
  }  
}
```

I do have an issue of a high pitch hum as well on top of the signal (which is why I lower the gain...otherwise I get a lot of clipping at higher frequencies).

earlephilhower, 2018-02-03:
@alanroche2015  Can you confirm @rfestag 's findings?  Just edit the I2S.cpp file and replace ".use_apll =1" with ".use_apll=0" and re-run.

If so, I'll just drop the whole thing.  The Audio PLL seems to be very flakey, or at least its integration with the Arduino SDK.

alanroche2015, 2018-02-06:
Hi all

Worked perfectly, I just followed the above changes and everything played at the correct speed. I can hear a small ‘hum’ when not playing audio but I don’t think that’s anything to do with the software.

Thanks for all the help.

earlephilhower, 2018-02-06:
Wonderful, thanks for the feedback!  I'm going to drop the APLL by default...it's been nothing but trouble.

commit 0bb85903 (2018-02-06): handle esp32 apll issues (#51)

* make use_apll user-configurable (fixes #37)

* disable apll per default (fixes #47)
closed 2018-02-06 by earlephilhower