earlephilhower/ESP8266Audio #48: esp32: send silence on error?

h3ndrik, 2018-02-03:
The ESP32 starts stuttering once the i2s dma buffer gets empty. how should we handle this?

we could i2s_zero_dma_buffer() in order to gain some time (Pushes zero-byte samples into the TX DMA buffer, until it is full.) and to play silence until anything else is written. fortunately the i2s driver offers an event queue (currently not configured) which could report this. but where in the code would we poll for an buffer empty event? the output is not called regularly, only if there is data availabe.

another way would be to push silence down the pipeline. either via a new SetSilence() in AudioOutput? or do it sample by sample like i tried here (works most of the time, but not always):

https://github.com/h3ndrik/ESP8266Audio/commit/417e98123459f4db4aa6126af53593b6ab577217

h3ndrik, 2018-02-03:
update: idea of introducing a new method which doesn't affect other AudioOutputs something like this:

https://github.com/h3ndrik/ESP8266Audio/commit/b0c0911c0c8431ea66bb3770043686192507402a (see branch silenceonerror3)

also writes enough zeros to prevent the next decoding error;) but should probaby be moved to where the underflow is detected.

earlephilhower, 2018-02-03:
That's a good idea and needed for the ESP32.  The 8266's code will zero out buffers on underflow automatically, but the 32's code just repeats the existing data.  

Have you tried it with success?  As part of the MP3 decode, when a frame error occurs I could swear it zeros out the frequency coefficients (which will end up zeroing out the time-series data when generated) anyway, no?

h3ndrik, 2018-02-05:
i think the main issue is that there is nothing being propagated to the i2s either way. so the i2s buffer will underflow.

I'm still trying to make this work. It works fine for the initial decoding errors when starting a stream. But i still have to handle buffer underflows and check for other side effects.

h3ndrik, 2018-02-05:
buffer underflows are a bit of a problem as the buffer blocks for too long and we have to handle this before the blocking.
i think first of all the second (redundant) blocking read in AudioFileSourceBuffer "Still need more, try direct read from src" has to be removed. The next iteration with "Fill up completely before returning any data at all" will do that anyways, doesn't it?
That way the blocking read happens one buffer.loop() later, and the decoder will run again before that happens. The decoder could then poll if buffer.filled was set to false and do the zeroing. for example in the Input() function of the AudioGeneratorMP3 (right before the next read).

I'm currently unable to code that properly but my very messy first approach seems to be working. i'll try and find a proper way to implement that.

earlephilhower, 2018-02-05:
Why not just have us read client::available() bytes in the fil-up read?  I thought I was doing that already, but I guess not.  Only reading that amount guarantees non-blocking while still getting all the presently available data into the buffer for the next call.

earlephilhower, 2018-02-05:
The basic problem is that this library is not the right place to insert those 0s to I2S.  The SDK should really do this in its IRQ handler, just like the ESP8266 does.  Otherwise it's really nothing but a race condition waiting to fail. :(

h3ndrik, 2018-02-05:
alright. that seams reasonable. there is an i2s interrupt and as i understand the I2S_TX_REMPTY_INT bit in I2S_INT_RAW_REG is set on underrun.

But I'm simply not able to allocate the interrupt. seems it should be something like setting the intr_alloc_flag for the i2s_driver_install to shared and then set up another shared interrupt:
`esp_err_t err = esp_intr_alloc(ETS_I2S0_INTR_SOURCE+(i2s_port_t)portNo, ESP_INTR_FLAG_SHARED | ESP_INTR_FLAG_IRAM, i2s_isr, NULL, &i2s_intr_handle);`

but this always fails with an ESP_ERR_NOT_FOUND. doing it before the i2s_driver_install() works, but makes the i2s_driver_install() fail "E (14120) I2S: Register I2S Interrupt error".
non-shared interrupts also don't work but overwriting(?) the default isr with more low level functions seems to work. (example for i2s irq code [here](https://github.com/cnlohr/esp32-cnlohr-demo/blob/master/main/i2s_stream_out.c))

i'm officially stuck here.
as you pointed out sending an arbitrary amount of silence down the pipeline (in case something might be wrong) is obviously not the best way to do it.
but atm i prefer living with the occasional hiccups over spending more time with debugging this.
in case anyone can point me in the right direction... i'd be happy to implement this.

earlephilhower, 2018-02-06:
Seems like a bug in their code.  On the ESP8266 the i2s DMA/IRQ code is part of the Arduino core.  Is it the same on the ESP32?  I'd look myself, but I'm still deep into completely rewriting the SSL support on the 8266.

h3ndrik, 2018-02-07:
well, thanks for all the good advice. makes my life much easier. (1<<8) is more than an uint8, right? there is a downcast to uint8_t which would throw my ...FLAG_SHARED passed to the i2s_driver_install away. i believe that's indeed the "bug in their code". i filed a pull request (https://github.com/espressif/esp-idf/pull/1589)

to answer the question: i think both, the esp_intr_alloc() and i2s_driver_install() are from the esp-idf ( esp_intr_alloc.h resp. driver/i2s.h ) the headers in arduino's tools/sdk/include/ are there to pull the code from the idf, right?
can't find any i2s code in arduino-esp32, at all.

cranphin, 2018-03-29:
Note that they do detect underflows: https://github.com/earlephilhower/ESP8266Audio/issues/48
But they don't do much then :)
Wouldn't it be too late anyways? the dma is probably already pulling the next (wrong) buffer, you might get a pop even if you zero the buffers then?
Better would be to clear each buffer once it's marked as read I guess? if that's possible at all :)
(I guess just before they put it back in the queue of empty buffers)

cranphin, 2018-03-29:
Ha, yeah, should do the trick :)
Issue made ;)
https://github.com/espressif/esp-idf/issues/1789

garageeks, 2019-01-08:
Hi guys, apparently the fix has been pushed to the latest release (1.0.1-rc2), however some I2S structs have been changed and it doesn't work with ESP8266Audio anymore. Any clue on how to adapt AudioOutputI2S library? Below the compiler errors
`C:\Program Files (x86)\arduino-1.8.7\libraries\ESP8266Audio-master\src\AudioOutputI2SNoDAC.cpp: In member function 'virtual bool AudioOutputI2SNoDAC::ConsumeSample(int16_t*)':

C:\Program Files (x86)\arduino-1.8.7\libraries\ESP8266Audio-master\src\AudioOutputI2SNoDAC.cpp:94:8: warning: 'int i2s_write_bytes(i2s_port_t, const void*, size_t, TickType_t)' is deprecated [-Wdeprecated-declarations]

   if (!i2s_write_bytes((i2s_port_t)portNo, (const char *)dsBuff, sizeof(uint32_t) * (oversample/32), 0))

        ^

In file included from C:\Program Files (x86)\arduino-1.8.7\libraries\ESP8266Audio-master\src\AudioOutputI2SNoDAC.cpp:23:0:

C:\Users\nic\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.1-rc2/tools/sdk/include/driver/driver/i2s.h:274:5: note: declared here

 int i2s_write_bytes(i2s_port_t i2s_num, const void *src, size_t size, TickType_t ticks_to_wait) __attribute__ ((deprecated));

     ^

C:\Program Files (x86)\arduino-1.8.7\libraries\ESP8266Audio-master\src\AudioOutputI2SNoDAC.cpp:94:8: warning: 'int i2s_write_bytes(i2s_port_t, const void*, size_t, TickType_t)' is deprecated [-Wdeprecated-declarations]

   if (!i2s_write_bytes((i2s_port_t)portNo, (const char *)dsBuff, sizeof(uint32_t) * (oversample/32), 0))

        ^

In file included from C:\Program Files (x86)\arduino-1.8.7\libraries\ESP8266Audio-master\src\AudioOutputI2SNoDAC.cpp:23:0:

C:\Users\nic\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.1-rc2/tools/sdk/include/driver/driver/i2s.h:274:5: note: declared here

 int i2s_write_bytes(i2s_port_t i2s_num, const void *src, size_t size, TickType_t ticks_to_wait) __attribute__ ((deprecated));

     ^

C:\Program Files (x86)\arduino-1.8.7\libraries\ESP8266Audio-master\src\AudioOutputI2SNoDAC.cpp:94:103: warning: 'int i2s_write_bytes(i2s_port_t, const void*, size_t, TickType_t)' is deprecated [-Wdeprecated-declarations]

   if (!i2s_write_bytes((i2s_port_t)portNo, (const char *)dsBuff, sizeof(uint32_t) * (oversample/32), 0))

                                                                                                       ^

In file included from C:\Program Files (x86)\arduino-1.8.7\libraries\ESP8266Audio-master\src\AudioOutputI2SNoDAC.cpp:23:0:

C:\Users\nic\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.1-rc2/tools/sdk/include/driver/driver/i2s.h:274:5: note: declared here

 int i2s_write_bytes(i2s_port_t i2s_num, const void *src, size_t size, TickType_t ticks_to_wait) __attribute__ ((deprecated));

     ^

C:\Program Files (x86)\arduino-1.8.7\libraries\ESP8266Audio-master\src\AudioOutputI2S.cpp: In constructor 'AudioOutputI2S::AudioOutputI2S(int, int, int, int)':

C:\Program Files (x86)\arduino-1.8.7\libraries\ESP8266Audio-master\src\AudioOutputI2S.cpp:72:5: sorry, unimplemented: non-trivial designated initializers not supported

     };

     ^

C:\Program Files (x86)\arduino-1.8.7\libraries\ESP8266Audio-master\src\AudioOutputI2S.cpp:72:5: sorry, unimplemented: non-trivial designated initializers not supported

C:\Program Files (x86)\arduino-1.8.7\libraries\ESP8266Audio-master\src\AudioOutputI2S.cpp:72:5: sorry, unimplemented: non-trivial designated initializers not supported

C:\Program Files (x86)\arduino-1.8.7\libraries\ESP8266Audio-master\src\AudioOutputI2S.cpp: In member function 'virtual bool AudioOutputI2S::ConsumeSample(int16_t*)':

C:\Program Files (x86)\arduino-1.8.7\libraries\ESP8266Audio-master\src\AudioOutputI2S.cpp:192:10: warning: 'int i2s_write_bytes(i2s_port_t, const void*, size_t, TickType_t)' is deprecated [-Wdeprecated-declarations]

   return i2s_write_bytes((i2s_port_t)portNo, (const char*)&s32, sizeof(uint32_t), 0);

          ^

In file included from C:\Program Files (x86)\arduino-1.8.7\libraries\ESP8266Audio-master\src\AudioOutputI2S.cpp:23:0:

C:\Users\nic\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.1-rc2/tools/sdk/include/driver/driver/i2s.h:274:5: note: declared here

 int i2s_write_bytes(i2s_port_t i2s_num, const void *src, size_t size, TickType_t ticks_to_wait) __attribute__ ((deprecated));

     ^

C:\Program Files (x86)\arduino-1.8.7\libraries\ESP8266Audio-master\src\AudioOutputI2S.cpp:192:10: warning: 'int i2s_write_bytes(i2s_port_t, const void*, size_t, TickType_t)' is deprecated [-Wdeprecated-declarations]

   return i2s_write_bytes((i2s_port_t)portNo, (const char*)&s32, sizeof(uint32_t), 0);

          ^

In file included from C:\Program Files (x86)\arduino-1.8.7\libraries\ESP8266Audio-master\src\AudioOutputI2S.cpp:23:0:

C:\Users\nic\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.1-rc2/tools/sdk/include/driver/driver/i2s.h:274:5: note: declared here

 int i2s_write_bytes(i2s_port_t i2s_num, const void *src, size_t size, TickType_t ticks_to_wait) __attribute__ ((deprecated));

     ^

C:\Program Files (x86)\arduino-1.8.7\libraries\ESP8266Audio-master\src\AudioOutputI2S.cpp:192:84: warning: 'int i2s_write_bytes(i2s_port_t, const void*, size_t, TickType_t)' is deprecated [-Wdeprecated-declarations]

   return i2s_write_bytes((i2s_port_t)portNo, (const char*)&s32, sizeof(uint32_t), 0);

                                                                                    ^

In file included from C:\Program Files (x86)\arduino-1.8.7\libraries\ESP8266Audio-master\src\AudioOutputI2S.cpp:23:0:

C:\Users\nic\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.1-rc2/tools/sdk/include/driver/driver/i2s.h:274:5: note: declared here

 int i2s_write_bytes(i2s_port_t i2s_num, const void *src, size_t size, TickType_t ticks_to_wait) __attribute__ ((deprecated));

     ^`

dagnall53, 2019-02-04:
@h3ndrik @earlephilhower I can add to this a little: I am in the middle of making my Software ESP32 compatible. I am using the NoDAC (1 Transistor) option, and at first got confused by the update to the library which removed the  AudiooutputDAC Cpp/h code and somehow affected NoDAC. The issue was that the original AudioutputDAC.cpp code remained in my copy of the library, causing problems. However deleting the two "DAC" files made everything work again for "NoDAC" (although I have not tried a real DAC yet with ESP32). 
I understand that the changes you made to work with ESP32 affected the I2S code, but I do get a lot of deprecation warnings (which lead me to this thread). 
When you get time, perhaps you can recast the code to use different functions to avoid the deprecation warnings? Presumably if they are deprecated, there are some alternatives?.

The sounds are working well (including the mixer), but I have heard some faults seemingly during startup, but they are rare and I have not tracked them down. Mostly they correct themselves. I have some sounds that end and leave the speaker quiet.others that leave a very slight audible clicking. If I was to guess, I think that if the sounds do not end on "0", the last signal outlevel is being repeatedly replayed. (but this is only a guess). 

All the best
Dagnall
 

  
[Deprecation warnings.txt](https://github.com/earlephilhower/ESP8266Audio/files/2827512/Deprecation.warnings.txt)

garageeks, 2019-06-26:
Hi to all, an update from my previous attempt. Using the latest commit on esp32 1.0.2 version, have set a 64kbit buffer and removed `mp3->RegisterStatusCB(StatusCallback, (void*)"mp3");` cause the serial output flooding makes buffering slow. 
It works brilliantly, but still make noise like a 56k modem when starting audio stream or buffer underrun occurs. 
It would be awesome if there is a way to silence output on these occasions.

garageeks, 2019-06-27:
I tried the flag _tx_desc_auto_clear = true_ , seems the noise when buffering changed, but it's still nasty :(
Please see an example https://youtu.be/0N4wqma4CY0
What can we do to improve it? I could mute the amplifier when buffering, but a software solution would be preferable

earlephilhower, 2020-08-08:
Code was submitted which fills the ESP32 I2S buffers w/0s on stop(), which should more or less handle this.  If not, please open a new bug w/some more info and a way to repro it.  Additional PRs always welcome!

closed 2020-08-08 by earlephilhower