earlephilhower/ESP8266Audio #49: No playback on NoDAC on esp-12-e

renegadeandy, 2018-02-04:
Trying to run the NoDac example in the readme fails on the esp-12-e if you are flicking the onboard led.

This onboard LED seems to use GPIO pin 2.

According to the board docs : http://www.kloppenborg.net/images/blog/esp8266/esp8266-esp12e-specs.pdf pin 2 also handles : GPIO2; UART1_TXD, however I believe it also is used for i2s, hence the potential issue.

Can the led be driven from another pin?

Is this a board limitation?

Can the i2s be driven on different pins to avoid this issue?

earlephilhower, 2018-02-05:
Howdy,

The I2S pins are hardwired, there is no changing them allowed on the ESP8266.  

That said, I'm using Wemos D1 Mini Pros which also use an ESP-12 which also has the Tx LED.

Do you have the right pin?  It's actually the Rx that's the data out, not Tx.

renegadeandy, 2018-02-05:
Yes sorry got it the wrong way round, but this is definitely an issue.

closed 2020-08-08 by earlephilhower