earlephilhower/ESP8266Audio #50: Question : Replay the same MP3 again ??

itsjustvenky, 2018-02-05:
@earlephilhower thank you so much for your excellent library.

I have a question regarding replaying the same MP3 file from SPIFFS.
Once the mp3 is played and then mp3->stop() will get called.

Now again I tried using "  mp3->begin(id3, out);", the mp3 is not getting played again.
Is this the correct way of replaying a mp3 with same settings (file, i2s..etc)

Please let me know. Thanks.

itsjustvenky, 2018-02-05:
Ignore this, I found the answer.

closed 2018-02-05 by itsjustvenky

jivido, 2018-02-23:
Hi!
I'm having the same issue. Could you perhaps tell what you've done to solve it?
I fixed it by re-loading the file (creating a new AudioFileSource object...) at a recall. Is that it?
Thanks

earlephilhower, 2018-02-23:
That's the correct way to handle this, yes.  Create a new file object, connect it to a new mp3decoder, and you can use the old i2s output object.

Rich968, 2023-03-02:
I'm playing a short mp3 file from Program Memory rather than from SPIFFS. I'm also using an external I2S DAC. I have not found it neccessary to delete and recreate new objects to replay the file. I simply re-open the file and call mp3 begin() again ie: file->open(sound_mp3,sizeof(sound_mp3));
then:    mp3->begin(file, out);    (Re-opening the  file is needed because mp3->stop() seems to close the file as part of it's routine) This all seems to work fine. Without getting further into the details within the library - are there any memory allocation issues that are not handled correctly with this approach?