earlephilhower/ESP8266Audio #53: Audio playback using NoDAC causes WiFi / Udp server to 'stop'

renegadeandy, 2018-02-05:
Hi, this issue is reproducible and happens consistently. It is using the NoDAC output, as I have no DAC to test it out on that Output option.

My setup is:

ESP-12-E
v1.4 Open Source
160MHz

This project reproduces the problem, please check it out from git : https://bitbucket.org/renegadeandy/audio_bug/

`git clone https://renegadeandy@bitbucket.org/renegadeandy/audio_bug.git`

This application connects to WiFi and launches a UDP Multicast server joining group 231.3.3.7 on port 1337.
**Make sure to change your wifi ssid and password.** It then plays w1.mp3 in the music folder (everything is included in the repo)

Audio playback begins.

Using [Packet Sender](https://packetsender.com/) A tool for sending packets , I send packeets to the multicast group as per:

[image]

When a packet arrives, the DiscoverMe class sends a printf. Sending packets quickly after startup works ok:

> Connecting to VM2111119 .. connected
> DC::Starting up DiscoverMe listening for UDP datagrams...
> DC::Now listening to Multicast Group IP: 231.3.3.7l From IPAddress: 192.168.0.53
> Started playing audio
> DC::Received 16 bytes from IP:192.168.0.41 on port:52888
> DC::UDP packet contents:This is a test!!
> DC::Received 16 bytes from IP:192.168.0.41 on port:52888
> DC::UDP packet contents:This is a test!!
> MP3 stop, len==0

I then leave the ESP8266 running the same sketch for around 7 - 11 minutes, after the audio is played and a couple of packets are received.

Then I try to send another Multicast packet, it is never received by my application - no **DC::Received 16 bytes from IP:192.168.0.41 on port:52888** style message ever appears on my serial monitor. No errors, nothing, the main loop() continues on its merry way.

Reset the ESP8266, and the audio plays, some packets are received for the first couple of minutes, then , boom, packets are not received any longer.

Removing this audio library from the sketch, allows the UDP Server to run, stably, for any amount of time - drawing the conclusion that the problem is associated with this library. It may not be, but what else could it be?

Thoughts?

earlephilhower, 2018-02-06:
Can you remove the wire to your 1T amp and run your test?  I just did a circuit simulation and the current through the I2S output pin to the xistor is higher than I though (by about 10x per spice!) which may be wreaking havoc with your WiFi internal circuits.

As TCP is built on top of UDP, and myself and others have run TCP streaming for hours at a time, I think the likelihood of this being a code problem is pretty low.

renegadeandy, 2018-02-06:
1T amp?

earlephilhower, 2018-02-06:
The transistor base.  Just disconnect it so the I2S is sending bits, but nobody is receiving them.

renegadeandy, 2018-02-06:
Ok tried that. Problem still occurs.

renegadeandy, 2018-02-08:
Any ideas on this?

earlephilhower, 2018-02-09:
Can you see if it's even reporting still being connected to WiFi on every 100 loops() or so?  There were reports I saw where WiFi could become disconnected in certain circumstances.  These were fixed in git head, so if you are getting disconnected try git head instead of 2.4.0 release.

renegadeandy, 2018-02-09:
Added the loop wifi checker, which suggests WiFi is connected fine -- updated the git repo with this new check.

I never get the WiFi Not Connected message and the DiscoverMe just stops responding.

Remove Audio from the mix and it works perfectly.

renegadeandy, 2018-02-14:
I am using a 2N3904 -- mistakes do happen!

How do you wire up the resistor? I could try that...what other changes do you have compared to me....most odd.

dagnall53, 2018-02-14:
2N3904 has 200mA collector current rating which is perhaps marginal? 
If you have an 8 ohm speaker and say 5V for the speaker power, the transistor will need to pass 625mA, and the higher the voltage on the speaker, the more current... Having said that, I had a BC107 working fine. But you would be better with a 1A capable transistor, especially if you have a higher voltage power supply. I would not use the 3v3 rail, as taking pulses from this is asking for trouble from interference.

When I say add a 1K resistor, I mean you should connect one end to RX/D9 and the other to the transistor Base. You will also need a reasonable capacitor (see Earle's notes) across V+ to Gnd to help support the the power rails during the pulses that the transistor takes. 
                                                             
                                                    Collector -- Speaker --- V+
ESP9266/RX-----Resistor------Base                                Capacitor
                                                    Emitter------------------Gnd

Check the data sheets to make sure you have the correct pinout for the transistor. There are a couple of common pinouts, and not all TO-92 style transistors use the same pinout. 
Good luck
Dagnall

earlephilhower, 2018-02-14:
Thanks for chiming in, @dagnall53.  I've been spending what little free time I've got on other things recently and been neglecting the audio project.

If you really want to play it safe, the 220-1K base resistor could have a DC blocking cap in the base path.  A LPF is also technically required for proper operation of sigma-delta DACs, since they shift noise to higher frequencies, but in practice the HF aliasing hasn't been noticeable for these "toy" setups.

What's odd is losing the WiFi even w/o the transistor connected.  I can't see any way for the code itself to do this, because if it was corrupting memory it'd do way worse than just making UDP disappear.  Possibly power supply issues, with the code causing higher current draw than idle?

dagnall53, 2018-02-14:
Cheers!, just trying to help in the hope you can get around to playing with the mixer!
:-)

earlephilhower, 2020-08-09:
I think this one's going to remain a mystery.  Might be timing related (WiFi dropping), might be power supply/noise, might be something else, but there's no clear path forward.

closed 2020-08-09 by earlephilhower