earlephilhower/ESP8266Audio #54: AudioFileSourceHTTPStream : Playing MP3 shutters

itsjustvenky, 2018-02-06:
Board : ESP-12E (4MB Flash)

I am trying to play a mp3 file (15KB) served by a local webserver in the internal network.
But while playing the mp3, the playback shutters a lot. If I use the "AudioFileSourceBuffer", then I couldn't even hear the playback properly.
`
"mp3->begin(file, out);" 
`

Using "file" directly in the "begin(), the shuttering is not that much, but still its there.
But If I try to play the same mp3 from SPIFFS (same code) the playback is very smooth.
I have even tried to pre-allocate buffer
`
void *preallocateCodec = NULL;
const int preallocateCodecSize = 29192; // MP3 codec max mem needed
preallocateCodec = malloc(preallocateCodecSize);

mp3 = new AudioGeneratorMP3(preallocateCodec, preallocateCodecSize);
`

and it never helped. So any solution on how we can fix this ?? Thank you.

`
#define FS_NO_GLOBALS
#include <Arduino.h>
#include <ESP8266WiFi.h>
#include <ESP8266HTTPClient.h>
#include <ESP8266Spiram.h>
#include <FS.h>
#include <SD.h>
#include <ESP8266WiFi.h>
#include "AudioFileSourceSPIFFS.h"
#include "AudioGeneratorMP3.h"
#include "AudioOutputI2S.h"
#include "AudioFileSourceHTTPStream.h"

char ssid[] = "wifiname";
char password[] = "passwd";

AudioGeneratorMP3 *mp3;
// AudioFileSourceSPIFFS *file;
AudioOutputI2S *out;
AudioFileSourceHTTPStream *file;

void setup()
{
  Serial.begin(115200);

  Serial.printf("Connecting to %s ", ssid);
  WiFi.begin(ssid, password);
  while (WiFi.status() != WL_CONNECTED)
  {
    delay(500);
    Serial.print(".");
  }
  Serial.println(" connected");

  /**
    Loading a mp3 file from SPIFFS works without any shutter
     The playback is smooth
  **/
  // file = new AudioFileSourceSPIFFS("/test/hello.mp3");

  // Mp3 is loaded from a internal webserer on network
  file = new AudioFileSourceHTTPStream("http://192.168.0.120/test.mp3");

  /**
    Using buffer is causing more shutter in playback
  **/
  // buff = new AudioFileSourceBuffer(file, 2048);

  out = new AudioOutputI2S();
  out->SetOutputModeMono(true);
  out->SetGain(0.5);
  out->SetRate(22050);

  mp3 = new AudioGeneratorMP3();
  
  // Directly using "file" instead of "buff"
  // This is causing less shutters in playback
  mp3->begin(file, out);
}

void loop() {
   if (mp3->isRunning()) {
    if (!mp3->loop()) mp3->stop();
  }
}
`

earlephilhower, 2018-02-06:
Preallocating the buffer only helps keeping memory from fragmenting, it won't affect performance.

Make sure you are using MSS=1460 ("Higher Bandwidth") for TCP streaming.  At 576 ("Lower Memory") each MP3 compressed frame needs to be broken into *3* separate TCP packets, and that kills performance.

olivierLefebvre96, 2019-05-03:
@itsjustvenky Same problem for me.
I have the same DAC as you with a webserver on it that I use to play a music in streaming with HTTP.

After some tests, I've concluded that my board doesn't have enough CPU core frequency (up to 160Mhz).
At 80 Mhz, the stuttering is far worse.
Without the webserver, the stream is nearly perfect.

Here's my sketch (school project): https://github.com/iot-nova/wifi-speaker