earlephilhower/ESP8266Audio #57: Problem with starting stream/playback in loop

erhapp, 2018-02-12:
Hello

I'm having some trouble starting the playback of a mp3 stream from within loop(). The code crashes every time the stream is started in this way.

Starting the same stream within setup() works flawlessly.

I attached a script showing the problem. The variable "startInLoop" switches between setup (0) and loop (1)

What am I doing wrong? Any help would be greatly appriciated.

kind regards

Erik

[StreamSetupLoop.zip](https://github.com/earlephilhower/ESP8266Audio/files/1715815/StreamSetupLoop.zip)

h3ndrik, 2018-02-12:
I think with the `if (mp3->isRunning())` your accessing mp3 while it's still NULL.
You should move the startInLoop initialization code to the beginning of the loop and only access mp3 after that, or check if mp3 is set up before calling isRunning: `if (mp3 && mp3->isRunning())`

earlephilhower, 2018-02-12:
Agree with @h3ndrik , you're just reading *mp3 before mp3 was allocated.    Be sure to check for mp3 in both of the if()s.

closed 2018-02-12 by earlephilhower

erhapp, 2018-02-12:
Indeed. My mistake. Thank you very much for taking to help me out.