earlephilhower/ESP8266Audio #59: decoding error

GolPesar132, 2018-02-17:
I have this error in decoding mp3.

`STATUS(mp3) '257' = 'Decoding error 'lost synchronization' at byte offset 6050'`

what is the problem. should i change the mp3 file properties?

earlephilhower, 2018-02-17:
The MP3 decoder is seeing something it doesn't like in the bitstream.  It's not a fatal error, and most times the decoder will recover.

Is it every file with an error, or only one specific one?  If every one, how were they generated (tool/etc)?  Could you attach a sample?  It's always easier to do stuff when you can reproduce a failure on demand.

GolPesar132, 2018-02-18:
every file that i test has this problem. 

i dont know what this mean:
>  If every one, how were they generated (tool/etc)?

for example i use this url:
[http://mohammadmokhalled.ir/mp3/1.MP3](url)

closed 2020-08-08 by earlephilhower