earlephilhower/ESP8266Audio #60: Regarding issue 58 clarity

renegadeandy, 2018-02-21:
https://github.com/earlephilhower/ESP8266Audio/issues/58

I was linked to http://www.esp8266.com/wiki/lib/exe/fetch.php?media=pin_functions.png

But this still doesnt answer the question. This table shows some pins..... but which match ;
LRCK : ???
DATA ;???
BLK ; ???

It is *not* obvious to me looking at the table which pin matches these inoputs, please be specific with the mapping on the esp-12-e

bigujun, 2018-02-21:
DATA  ->  GPIO3/RX0
LRCK ->  GPIO2
BCLK ->   GPIO15

earlephilhower, 2018-02-22:
Thanks, @bigujun.  @renegadeandy you really need to xref that spreadsheet and your own specific board as it seems every board vendor renames or even renumbers the board pins vs. the controller pins.

closed 2018-02-22 by earlephilhower