earlephilhower/ESP8266Audio #61: MP3 playback delay / skip on Wemos D1 Mini Pro

jivido, 2018-02-26:
Hi! I really like the library :)
But there's a strange thing... I tried to (re)play a mp3 on a Wemos D1 Mini Lite, which worked great. 
Then I tried it on the Wemos ~ Pro, becaus of the bigger flash size. (Both on 160 MHz)
Then things behaved different. I heard a gap in the audio, at different places, at different bitrates... (Every time, same place)
Here's a picture:
[image]
MP3's are converted WAV's with ffmpeg: `ffmpeg -i filename.wav -b:a 320000 newfilename.mp3`
Here's the code (mostly the mp3-SPIFFS-example).

```
#include <Arduino.h>
#include "AudioFileSourceSPIFFS.h"
#include "AudioGeneratorMP3.h"
#include "AudioOutputI2S.h"

AudioGeneratorMP3 *mp3;
AudioFileSourceSPIFFS *file;
AudioOutputI2S *out;
bool beat = true;

void setup() {
  // put your setup code here, to run once:
  Serial.begin(115200);
  delay(100);
  SPIFFS.begin();
  delay(100);

  file = new AudioFileSourceSPIFFS("/a.mp3");
  out = new AudioOutputI2S();
  mp3 = new AudioGeneratorMP3();
  mp3->begin(file, out);
}

void loop() {
  // put your main code here, to run repeatedly:
  if (mp3->isRunning()) {
    if (!mp3->loop()){
      mp3->stop(); 
      delete file;
      delete mp3;
      mp3 = new AudioGeneratorMP3();
    }
  } else {
    Serial.printf("mp3 done\n");
    if(beat){
      Serial.println("a");
      file = new AudioFileSourceSPIFFS("/a.mp3"); // 320 kbps
      beat = false;
    } else{
      Serial.println("b");
      file = new AudioFileSourceSPIFFS("/b.mp3"); // 128 kbps
      beat = true;
    }
      delay(5000);
      mp3->begin(file, out);
  }
}
```

Curious!

earlephilhower, 2018-02-26:
So, you're seeing no problem on the mini pro (which is what I'm using now), but on the pro you're getting a hiccup.  That looks like underflow, and I believe it's all related to SPIFFS.  When an underflow occurs, the samples in the buffer are all 0'd, which is what you're seeing there.

SPIFFS can be exceedingly slow in some cases.  I found that for my first audio generator, the MOD player, which needs to read blocks of sampled instruments over-and-over to play the tune, it was taking something like 40% of my code's time doing reads.  For the MIDI player, it was even worse, as the sample set was much larger.  I'm not knocking SPIFFS, it does some pretty cool things and seems to have functionality nailed...it's just slow on the ESP8266.

To counter slow SPIFFS, I started working on ESP8266FastRomFS, which gave me a 3x reduction in time on MIDI when I was using it, but I got sidetracked and haven't finished it.  It works, has an uploader, but is not fully tested in the way a filesystem should be.

As an alternative, could you try an SD card on the Pro (which, honestly, I haven't but other folks have)?  While it's no speed demon, I think it's somewhat faster than SPIFFS and not subject to the (random-ish) slowdowns.

jivido, 2018-02-26:
Hi, thanks for your response!
No problem with the D1 mini Lite (512k SPIFFS), having problems with the D1 mini Pro (15M SPIFFS).
I actually have the D1 SD-card shield here somewhere, I'll give that a try tomorrow!

earlephilhower, 2018-02-26:
One word of warning: I've been told that the SD shield uses some of the I2S bus for chip-select.  It's configurable somehow, though, so you can move it to another pin.  For my own brief testing I just used a generic microSD board w/5(?) pins that I simply wire-wrapped to the proper pins on my D1 Mini.

jivido, 2018-02-27:
Using a SD card, and an AudioFileSourceSD object, solved the problem, 128 kbps at 44.1 kHZ works fine!
I used the Wemos SD Shield on a breadboard, with D8 (GPIO15) on the shield wired to D2. 
In setup() I put `SD.begin(D2);`

jivido, 2018-02-27:
I'll close this issue, because it's a SPIFFS-size vs. speed related problem?

closed 2018-02-27 by jivido