earlephilhower/ESP8266Audio #62: Is there an indication of the pin assignment for DAC?

JohannesMTC, 2018-03-02:
How would you connect an "Adafruit I2S 3W Class D Amplifier Breakout - MAX98357A" to an ESP wroom 32?

This is described in detail for "nonDAC". But for a DAC that would have to be stated. According to Espressiff's documentation, the outputs could indeed be set freely. But where did the author of this libery link?

earlephilhower, 2018-03-02:
It's really up to you.  Lots of people like to use pin 22 as DIN, pin 25 as LRCLK, and pin 26 as BCLK/SCLK.

closed 2018-03-02 by earlephilhower

JohannesMTC, 2018-03-03:
I am convinced that the mistake lies with me.
Nevertheless, I am now using the example "MixerSample" to run for days.
Serial monitor shows that the two waves are started and stopped.
Nevertheless, the speaker is mute.
I think I've already made 50 visits on how the DAC is connected.
Do I have to insert something somewhere in this example, that the ESP32 knows where to connect?
If so, where and how?