earlephilhower/ESP8266Audio #63: ESP32 AudioI2S causes relay switch to 'buzz'

renegadeandy, 2018-03-02:
Hi!

I have an ESP32 board, the oooouuuu development board. I have the following sketch, where relayPin1 is attached to a relay board(https://www.amazon.co.uk/gp/product/B06XKQQXKW/ref=oh_aui_detailpage_o01_s00?ie=UTF8&psc=1)

When I run the sketch, the sampleaac.h audio plays, but as soon as it has finished, the relay which is powered (sending low to relayPin1) makes a buzzing noise. Removing the ESP8266Audio library from the sketch stops the buzz. Help!

The code for the project which recreates the issue can be found at this git repo:
https://bitbucket.org/renegadeandy/audio_bug_buzz

Grab it with:

`git clone https://bitbucket.org/renegadeandy/audio_bug_buzz`

earlephilhower, 2018-03-02:
Sounds like you have a crosstalk issue. Try doing an AudioOutputI2S->stop() to disable the interface and clocks.

renegadeandy, 2018-03-02:
Aha - that solves it, thank you. I assume the Audio from SPIFFS now works on ESP32 since they fixed the SPIFFS support?

earlephilhower, 2018-03-02:
I've not done any ESP32 work recently, but others say it does work if you apply a pull request that's been floundering for a while now or get an older SPIFFs toolset/GIT tag.  Check the issues here for the link, or the README.md.  Unfortunately it seems like the ESP32-Arduino is a little less responsive than the ESP8266-Arduino team (I think the IDK is really pushed for the ESP32 so that probably explains it).

renegadeandy, 2018-03-02:
What is the IDK?

earlephilhower, 2018-03-02:
Meant to say IDF.  Their build platform.  Was thinking last word as "Kit" not "Foundation"

closed 2018-03-03 by renegadeandy