earlephilhower/ESP8266Audio #64: Audio Playback from SPIFFS fails sometimes with specific filenames

renegadeandy, 2018-03-03:
Hi,

I have an ESP32, running the AudioOutputI2S working well. However, I have found that for some reason when certain files are to played back with the my set of file names, more commonly than not, no audio is played back at all.

Please find the project which recreates this situation here: https://bitbucket.org/renegadeandy/audio_bug_buzz

Grab it with:

`git clone https://bitbucket.org/renegadeandy/audio_bug_buzz`

If you take the sketch, rename all audio files, 1.mp3, 2.mp3, 3.mp3 and adjust the sfx.h accordingly,  audio playback works every time! Why is this?

earlephilhower, 2018-03-04:
There's a 31-char total limit for SPIFFS filenames, and no directory support (it just prepends the path to the name, again w/a 31 char total limit).  Looks like you're probably running into it.  Remove the subdirs, shorten the names, and all should be fine.

renegadeandy, 2018-03-04:
I didn't know that limitation, fantastic thank you for explaining it!

closed 2018-03-04 by renegadeandy

JohannesMTC, 2018-03-05:
I found a solution for reading from the SPIFFS on an ESP32.

I use the "ESP32 Data Upload Tool."
https://github.com/me-no-dev/arduino-esp32fs-plugin

After the "ESP32 Sketch Data Upload" open the example "SPIFFS-Test"
Compil it and upload it.
Then open the Serial Monitor.

Using the reset button, restart the ESP32 until the output on the serial monitor shows no error.

Then you can compile and upload your own sketch.

However, once you load up additional data, you need to re-upload and run the example SPIFFS Test.

earlephilhower, 2018-03-05:
Actually, @JohannesMTC , it looks like the SP32-Arduino project just yesterday committed a PR which fixed it.  I'm going to update the README.md, but see the end of https://github.com/earlephilhower/ESP8266Audio/issues/43 for the details.

JohannesMTC, 2018-03-05:
Thanks for the answer.

My comment was about using the "Sketch Data Upload" to upload the files. Here is still the error, but can be eliminated with above-described method.
For my project I need about 24 files. Applying all of these using "make" would be quite a hassle.

If there is an update for the "Sketch Data Upload" that fixes this error, I will contact you again.