earlephilhower/ESP8266Audio #65: Audio Playback causes relay switch to "fade" and cause relay to "buzz" whilst audio is playing

renegadeandy, 2018-03-03:
I have an ESP32 which is playing back audio through I2S using this library nicely.

However when MP3 audio is playing, from SPIFFS, when I have my relay being driven (by sending LOW on pin 21) I get 2 problems. 

1. The relay LED dims oddly during the Audio playback, and returns to normal after audio stops
2. The relay makes buzz noises during the Audio playback, and returns to normal after audio stops.

I am not sure why this happens, ultimately, I want a solution where these effects are not present. This problem seems related but is not the same as this issue: https://github.com/earlephilhower/ESP8266Audio/issues/63 because this new issue happens DURING audio playback, but not after. The issue which is now fixed caused noise AFTER audio playback, but not DURING.

The sketch which recreates this issue can be found at this repository:https://bitbucket.org/renegadeandy/audio_bug_relay

Check it out with:

`git clone https://bitbucket.org/renegadeandy/audio_bug_relay`

earlephilhower, 2018-03-04:
This and the other problems all point to a bad power supply.  You're going to need to find a different way of powering the chip.  Try a USB 2.4A charger or a powered USB 3.0 hub(normally they have higher current capabilities than cheaper 2.0 models).

renegadeandy, 2018-03-04:
I have tried 2 other ways of powering the board, with exactly the same results.

earlephilhower, 2018-03-04:
Sorry, you must have some circuit issues then.  If a relay is making noise when you're playing sounds, that's because the power rails into it are oscillating, or the control line is marginal.  Either way, it's a HW, electrical issue and not SW.  Check your schematic, make sure you're not sharing any connections with the analog portion (i.e. it's not even really possible to drive a relay w/the ESP8266's pins, you need optoisolators and probably even xistors to drive those isolators as many require more mA than the 8266 can safely provide), add decoupling caps wherever possible, etc.

renegadeandy, 2018-03-05:
Admiteddly the relay and my https://www.banggood.com/PCM5102PCM5102A-DAC-Decoder-Board-I2S-32Bit-384K-For-Raspberry-Pi-Red-Core-Player-p-1259752.html?rmmds=myorder board are connected to the same 5v rail. This could be the reason? I am not sure how to actually solve it though? Chunky capacitor on the way to powering the Relay?

renegadeandy, 2018-03-05:
This is my best effort of drawing the circuit as it stands, which recreates the problem:

[image]

earlephilhower, 2018-03-05:
I think you need to check the current requirements for that relay module.  The ones I've used need >10mA to drive the optoisolator, which is beyond chip spec for the ESP32.  Unlike the Atmel AVR chips in the classic Arduino setup, which are built for driving outputs and can source or sink a lot of current (and are 5V tolerant to boot), the ESP32 is intended for more cerebral tasks. :)  Some sort of 74xx series buffer is required to safely drive that relay isolator...

renegadeandy, 2018-03-05:
The relay is a SRD-O5VDC-SL-C, Not sure what the current needed to drive that is, datasheet is hard to read, not sure what my voltage code is? 05?
[
Datasheet](http://datasheetcafe.databank.netdna-cdn.com/wp-content/uploads/2015/10/SRD-05VDC-SL-C-Datasheet.pdf)

earlephilhower, 2018-03-05:
I hope your board doesn't actually connect the input to the relay but uses an optoisolator.  If not, go and recycle it and save yourself much trouble...it's plain unsafe if you're thinking of switching AC voltages.

The datasheet also says it needs to source/sink 70-80mA, which is so far above the safe limit of the ESP8266 and even AVR chips it's not funny.

Even optoisolators need 10+mA simply because you need to light up a little LED in the package to switch on the high-side diode.  Driving 1 relay may be marginal.  Driving 2+ is definitely out of spec.

renegadeandy, 2018-03-05:
I hope your board doesn't actually connect the input to the relay but uses an optoisolator. If not, go and recycle it and save yourself much trouble...it's plain unsafe if you're thinking of switching AC voltages.

**I am not sure I follow you. I am indeed driving the input to the relay low to activate the relay switch - it is designed for an arduino!!!! Why would this mean I should recycle it?  And yes, I plan on switching AC voltages.**

The datasheet also says it needs to source/sink 70-80mA, which is so far above the safe limit of the ESP8266 and even AVR chips it's not funny.

**Agreed if that's the case, but the chip is marketted as an arduino relay, how wrong were they? Please see the board here : https://www.amazon.co.uk/gp/product/B06XKQQXKW/ref=oh_aui_detailpage_o01_s00?ie=UTF8&psc=1**

Even optoisolators need 10+mA simply because you need to light up a little LED in the package to switch on the high-side diode. Driving 1 relay may be marginal. Driving 2+ is definitely out of spec.

**This board may then use optoisolators. If this is the case, how can I drive them from the ESP32 in a safe way? It's confusing for me when we talk about driving them, because I am sending LOW to drive them anyway, not sure I understand how you drive something LOW with current ~ 10ma.**

earlephilhower, 2018-03-05:
The board you linked does use isolators, but they require more current than you can drive on the ESP chips.  Even though you are grounding the input, what that just means is all that 10mA+ current needs get to ground *through your chip.*  So high or low doesn't matter, you're either sourcing or sinking, and they both are beyond ESP specs.

For a real *AVR Arduino* (Uno, Mega) you can probably drive several relay isolators just fine.  But those are 100% different chips, built for high IO tolerances and low CPU loads.  You're using an *ESP* chip, not any real Arduino board, and only using the Arduino software toolchain.  You need to get a high current drive buffer chip to safely sink that current.  You should check the forums at www.esp8266.com for any suggestions from other folks trying to what you're doing.

A board w/o isolators could, under fault conditions drive those 120-240VAC directly to the chip and all connected parts.  That could, if you're lucky, only cause a fire.  If you're unlucky and pressing a button on the board when it happens, it could kill.  Fuses and circuit breakers are there to protect the house wiring, not people, and often don't trip in electrocutions because the current required to maim or kill is so much smaller than the 10-20A they trigger on.

I *strongly recommend* stopping your idea of switching AC with this board until you get a professional opinion and advice on your design, construction, and packaging.  Being new to electronics isn't a bad thing, but when playing with mains voltage small things can be fatal.  There are also commercial/local electrical codes you need to obey if you don't want to end up losing insurance coverage should it go pear shaped.  Here in the US there's the NEC and UL, I'm not quite sure what's there in the UK but I'm sure there are equivalents.

renegadeandy, 2018-03-06:
</lecture>

renegadeandy, 2018-03-08:
I am well aware that option 4 is dangerous, but I understand what you are saying. I do actually understand what I am doing.

Is there a specific ESP32 API for changing the current for control bits?

renegadeandy, 2018-03-08:
Having done more research this specific relay is a SRD-05VDC-SL-C which I believe correlates to a 5v high sensitivity coil. That means each pin will draw 70mA to drive the coil, obviously this is way beyond the spec for the ESP32 - hence my question.

earlephilhower, 2018-03-08:
Your best bet is the ESP32 boards @ espressif about the API for setting that drive strength.  Worst case you will end up setting the register bits directly.

The relay drive current is immaterial if you're using the isolator, which is itself a switch.  You just need to make sure the led inside has enough current to fully open the photodiode on the other side, and that it's not getting any current when you want it off (or vice-versa depending on if it's inverted logic or not).

earlephilhower, 2018-03-09:
You should also do the circuit analysis on the optoisolator path.  I have a feeling that you're not truly turning off that LED even when you set the output to high...that could cause relay clatter as the electromagnet would be somewhere in the middle of its drive strength.

renegadeandy, 2018-03-09:
When connecting the ESP32 pin directly to the relay signal pin, during the music the ESP pin flicks between 1.4v and 1.8v

After the music stops, it settles at 49mV.

The 5v for the relay sits at 4.68v whilst the music plays, and returns to 4.71v when the music stops.

earlephilhower, 2018-03-10:
Well, there you have it then.  Are you sure you didn't connect the I2S LRCLK, MCLK, or DOUT pin to that relay???  If not, you've got massive coupling noise to an I2S signal and need to redo wiring completely.

renegadeandy, 2018-03-10:
I absolutely definitely did not mix the wires up. Definitely, GPIO pin 22 is relay, 5 ,17 & 16 are for i2s. What next! When you say redo the wiring, what do you mean?

earlephilhower, 2018-03-10:
You have a clock or data pin driving either directly or via coupling that relay pin since 1/2 (50% highj, 50% low) 3.3v = 1.6V == exactly the voltage you're seeing.  If you had an oscilloscope I bet you'd see a nice clock waveform on the pin, either MCLK or LRCLK, or your actual I2S bitstream.  There's nothing in the SW this is related to, so I'm going to close this.  Your best bet is to find an ESP32 forum and publish your circuit and hope someone there can point you in the right direction.  Good luck!

closed 2018-03-10 by earlephilhower

renegadeandy, 2018-03-10:
If i disconnect the ESP32 from the relay, and test the voltage on Pin 22, it indeed sits between 1.4v and 1.8v whilst audio is playing.

When audio stops, it sits at 0v.

If I change the ESP32 pin to 13 for the relay low, then this problem doesn't happen. What is it about pin 22 that causes this?!

I can't work it out? Please see this github project which recreates the issue :
`git clone https://renegadeandy@bitbucket.org/renegadeandy/audio_bug_relay.git`

Note to prove it works, just change the:
`relayPin1 = 22;`
to
`relayPin1=13;`
and connect voltmeter to pin 13 to test.