earlephilhower/ESP8266Audio #66: MP3 and ACC does not work correctly but WAV is fine

abdollahpour, 2018-03-04:
Thanks for your great library!
I'm using ESP-12F （3M SPIFFS, 160Mhz). I just use the example programs. Seems my board works fine because playing WAV from SPIFFS is fine and sound quality is good with no error. But mp3 quality (from SPIFFS) is terrible, it's just loud random noises with very low music in the background.
Any idea what's the problem?

earlephilhower, 2018-03-04:
Sounds like a power supply issue.  MP3 and AAC decode require much more CPU and so burn more power, which may be introducing noise on your amplifier's rails.

Also, for AAC, only non-SBR is supported.  There isn't enough RAM on the 8266 to do it, but normally only web radio uses this extension since it reduced BW.  The -32 can and does support it fully, thanks to its much larger RAM.

abdollahpour, 2018-03-05:
I have another board using ILI9341 and a couple of more modules and it still works good with WAV, this one just has audio output and an ESP but still have the problem.
I'll also check it with ESP32 and let you know if it has the same result or not.
Thanks for the answer.

earlephilhower, 2018-03-05:
One other thing could be that you are having i2s timing issues and the bitstream is sorted by 1 but in reception.

You could start by adding a out->SetGain(0.4); to reduce the output volume. If that fixes it you could try adding a resistor on the DIN pin.

closed 2020-08-08 by earlephilhower