earlephilhower/ESP8266Audio #68: Trying to play file on loop

rfestag, 2018-03-05:
I'm working on a project where I want to play a certain file on a loop, but when certain buttons are pressed I play a different file (then go back to the loop sound). I have code that kind of works, but I run into a couple of issues:

1. There is a distinct click each time I change to a different file for playback
2. After a little while, my call to mp3->begin() fails, which is immediately followed by a `Guru Meditation Error: Core  1 panic'ed (LoadProhibited)`

I'm wondering if there is a way to do this without re-allocating new AudioFileSource* objects each time (my attempt at that using currentFile->open(path) didn't seem to work. I'm also curious what the proper way to handle the begin() failing is (and what could cause it to fail), since I assume that failure is leading to the eventual panic. The code below works for a few  repeats of the file, but eventually I get the panic and it reboots. Interestingly, my other initialization code also fails immediately after this panic, but hitting the reset button allows it to boot properly.

```
# SoundService.h
class SoundService
{
  public:
    void begin(int pin);
    void swing();
    void clash();
    void hum();
    void lockup();
    void extend();
    void retract();
    void tick();
    void setGain(float gain) {out->SetGain(gain);};
  private:
    bool loop = false;
    bool playing = false;
    const char* loopPath;
    AudioGeneratorMP3 *mp3;
    AudioOutputI2S *out;
    AudioFileSourceSPIFFS* currentFile;
    AudioFileSourceID3* currentFileID3;
    void playOnce(const char *file) {
      //if (mp3->isRunning()) {
      //  mp3->stop();
      //}
      if (currentFileID3 != NULL) {
        delete currentFile;
      }
      currentFile = new AudioFileSourceSPIFFS(file);
      if (currentFileID3 != NULL) {
        delete currentFileID3;
      }
      currentFileID3 = new AudioFileSourceID3(currentFile);
      loop = false;
      if (mp3->begin(currentFileID3, out)) {
        Serial.print("Beginning ");
        Serial.println(file);
      }
      else {
        Serial.print("Failed to begin ");
        Serial.println(file);
      }
    }
    void playLoop(const char *file) {
      loopPath = file;
      playOnce(file);
      loop = true;
    }
};
```

```
#SoundService.cpp

void SoundService::begin(int pin) {
  Serial.println("Setting up amplifier and MP3 player...");
  pinMode(pin, OUTPUT);
  digitalWrite(pin, HIGH);  // enable access to AMP
  out = new AudioOutputI2S(0, true);
  out->SetOutputModeMono(true);
  setGain(0.5);
  mp3 = new AudioGeneratorMP3();
  Serial.println("Done");
}
void SoundService::swing() {
  playOnce("/SWING1.mp3");
}
void SoundService::clash() {
  playOnce("/CLASH1.mp3");
}
void SoundService::hum() {
  playLoop("/HUM.mp3");
}
void SoundService::lockup() {
  playLoop("/LOCKUP.mp3");
}
void SoundService::extend() {
  playOnce("/POWERON.mp3");
  playing = true;
}
void SoundService::retract() {
  playOnce("/POWEROFF.mp3");
  playing = false;
}
void SoundService::tick() {
  if (playing && mp3->isRunning()) {
    if (!mp3->loop()) {
      if (loop) {
        playLoop(loopPath);
        tick();
      }
      else {
        hum();
        tick();
      }
    }
  }
}
```

```
#Snippet from sketch
function setup() {
  SoundService sound;
  sound.extend(); #This should play the extend sound once, then start looping on hum.
}
function loop() {
  sound.tick();
}
```

The basic logic seems to work for a random number of loops

earlephilhower, 2018-03-05:
On the ESP32 on i2s underflow or stop, the DMA buffers are not cleared so you end up with stuttering at the end.  On the ESP8266, on underflow the underflowing buffer is 0'd out in the Arduino core.

To avoid that problem, a user submitted a patch which fills the DMA buffers w/0 on a generator->stop().  If this only overwrote free samples, it shouldn't click (but you may have a gap of <10ms), but if this IDF call actually writes 0s over the entire buffers (including bits which were written to the buffer but not sent to the I2S yet) then it might cause clicking.  

You could try commenting that line out in the code to avoid the 1st issue.

Your second one is most likely an out-of-mem error.  If the AudioInput==null or AudioOutput==NULL, begin() can fail (safely), but at that point there's other bits of the ESP32 code that are going to barf since malloc()s are failing.  Can you double-check the free heap on each call?  On the ESP8266 it's Serial.prinf("head: %d\n", ESP.getFreeHeap()); I'm sure there's something like it on the -32.

As the begin() call is going to allocate memory, my guess is your not calling delete mp3 is the culprit.  For now, sorry, there's no reuse of objects, and you need to make a new one...

jurecotic, 2018-03-07:
@rfestag Have you managed to fix the second issue? Would you be willing to share the working code?

rfestag, 2018-03-08:
@jurecotic I actually haven't had a chance to play with either of the suggestions from @earlephilhower . I'm hoping to have some time on Friday. If I do figure it out it, I'll definitely post some working code.

rfestag, 2018-03-08:
Update: Simply adding the print statements above before and after the begin statement seems to have changed the behavior. Previously, I'd get 5-10 loops before I would see the error. Now it has been going for a few minutes and I haven't seen the problem. I'm guessing it is a timing issue? I don't think it will be an out of memory error, because I get the same output on every iteration:

```
...
MP3 stop, len==0
head (before begin): 212304
Beginning /HUM.mp3
head (after begin): 183056
...
```

Regarding the click, were you suggesting not calling generator->stop()? Or finding the code that changed in the commit you mentioned and commenting out the bit that is zeroing it out? I have tried with and without calling stop() when loop() returns false, no difference.

earlephilhower, 2018-03-08:
Comment out this line to stop it from filling buffers w/0.  I don't know if it fills all buffers or just tries to push in 0s:
https://github.com/earlephilhower/ESP8266Audio/blob/9e708af1f229b7fc017459dda56bcf2765a42f5c/src/AudioOutputI2S.cpp#L178

I now see you're changing the input to the same mp3 object and not making a new one.  I didn't build it that way, so I'm (happily) surprised to see it not barf.  You may have some state from the last playback that's causing some weird output at the beginning of the new one.

The simplest way to test this would be to change the output to a SPIFFSWAV which records the samples faithfully, and pulling that to your machine and examining w/audacity or some other tool.  If it's I2S related, this version will be clean.  If it's algorithm related, it'll show up here.

JohannesMTC, 2018-03-08:
I think the problem is quite different.
With the clearing of the IS2 buffer one does not get that under control.
Now write my own function that enables the audio output for the ESP32.

If you want to have a LOOP, it should connect directly to the previous one. Since you read in blocks, the respective File SOURCE class ignores the white space if the file does not exactly match the buffer size.
So you have to reopen the respective file immediately and start bit reading the file to fill the rest of the buffer. Everything else will crack or cause noise.

Since the classes of your libery do not internally include a LOOP function, this will never really go out.
The class would have to return a value indicating that a file is fetish and stopping the functions. Man would have to instruct his own program to monitor this and, if necessary, open a new file.

If you need a mixer now, it will be quite tight in time.

So I just started to write something for myself.

earlephilhower, 2018-03-09:
Oh, good luck then.  One other way would be to alternate between inputs on a mixer, but you need to keep track of playback times which could get complicated.

JohannesMTC, 2018-03-09:
The first attempt yesterday already showed that it does not expect the processing power of an ESP32 to seamlessly play a 16 bit 44100 sample / sec into a loop.
Although I pinned the interface to the core 0 and thus a processor without exception takes care of the output.
The jump in the read function when you arrive at the end of the file just takes too much time. So will the output with 8 bit and 22050 sample / sec realize. This reduces the computational effort to a quarter.
Since the soundfils are usually all very short, I also renounce to any conversion. It is invariably supported 8bit 22015 sample / sec PCM.

For robots or models, which also cause noise, should such an edition be enough? In addition one must fall back often on very small Laudsprecher and so the sound quality is already strongly limited.
Incidentally, I will only support the reading of SPIFFS or SD.
So this will be an audio interface that is only suitable for such purposes.

In any case, I now understand why that always had to fail with your Libery. You have to do a lot of effort to read the header information from the files and thus the output immediately to fit. The luxury you can not afford in an issue, the hapless audiofiles in a row can not afford.

Accordingly, my interface is then not in competition with yours. It should simply be an alternative that responds to the specific needs of modelers. is already strongly limited.
Incidentally, I will only support the reading of SPIFFS or SD.
So this will be an audio interface that is only suitable for such purposes.

In any case, I now understand why that always had to fail with your Libery. You have to do a lot of effort to read the header information from the files and thus the output immediately to fit. The luxury you can not afford in an issue, the hapless audiofiles in a row can not afford.

Accordingly, my interface is then not in competition with yours. It should simply be an alternative that responds to the specific needs of modelers.

rfestag, 2018-03-09:
I tried commenting out the dma zero function. There is a  slight difference, but the issue still exists. Iwas actually wondering something along the lines of what @JohannesMCT suggested. Perhaps keeping track in the ID3 source of where the end of the header is, and add an option to tell the generator to replay. That would seek to the beginning of the data and continue playback.

earlephilhower, 2018-03-09:
FWIW, SPIFFS can become very slow for reading and file operations on the ESP8266.  Not sure if the same issue happens on the -32, but pre-opening and just Seek()'ing is likely to help.

@rfestag's idea is actually quite elegant, if I can figure out a clean way to capture the start position after WAV header/ID3/etc. processing.

@JohannesMTC you make a good point!  May I suggest sticking with 16-bit, not 8-bit, as most I2S boards are natively 16-bit anyway and you'll need to convert 8-bit unsigned to 16-bit signed for each sample to drive them.

FWIW, I'm quite sure the ESP-32 can provide continuous 16bit, 44.1khz stereo output in a loop as long as the code isn't doing something very wasteful (which it **is** when my lib requires you to restart the whole MP3 processing at the end of the cycle).  The MOD tracker I ported does this 100s of times per song on a much slower ESP-8266, but because it works on preconfigured data structures it's smoother than the naive method I've got for looping.

rfestag, 2018-03-09:
I've taken a first stab, and seem to be having minimal success (but the click is still noticeable). Essentially I updated AudioFileSource.h to have a seekToData() function and a dataStart to record the position of data. It would be up to each AudioFileSource to set dataStart after it has parsed the header. Each AudioGenerator would then be updated to check whether they have reached the end of the file, and call seekToData() if necessary.

I've only tried implementing for MP3 so far, and changed the Input()  method as follows:

```
enum mad_flow AudioGeneratorMP3::Input()
{
  int unused = 0;

  if (stream->next_frame) {
    unused = buffLen - (stream->next_frame - buff);
    memmove(buff, stream->next_frame, unused);
    stream->next_frame = NULL;
  }
  if (unused == buffLen) {
    // Something wicked this way came, throw it all out and try again
    unused = 0;
  }

  lastReadPos = file->getPos() - unused;
  int len = buffLen - unused;
  len = file->read(buff + unused, len);
  if (len == 0) {
    if (replayOnEnd) {
      file->seekToData();
      return Input();
    }
    else {
      Serial.printf_P(PSTR("MP3 stop, len==0\n"));
      return MAD_FLOW_STOP;
    }
  }

  mad_stream_buffer(stream, buff, len + unused);

  return MAD_FLOW_CONTINUE;
}
```

Are there other places I should be looking to do the reset? Or is this likely just a case of SPIFFS taking a long time to reset the header and read the next data? If the latter, perhaps it makes sense to cache the first N bytes and use those while waiting for the file to seek?

earlephilhower, 2018-03-09:
SPIFFS isn't multithreaded (at least on the -8266), so you unfortunately can't seek while doing other things. :(

Are you getting any MP3 warnings to the serial port when it loops back (i.e. "lost sync at byte XXX").  I should remove that printout completely as it cascades and you can get 100s of those printouts until it finds a good syncword in the stream.  Comment out the error reporting up the stack and you might eliminate your problem...

rfestag, 2018-03-09:
Nope, no MP3 warnings. When replaying, I see no serial output with the code changes above. For kicks, I'm playing with saving off the first frame and copying from it into the buffer when the file->read() returns fewer bytes than expected (as opposed to waiting for it to return 0 bytes). I'm not getting as obvious of a click (it is more of a short silence). I then wait to reset the position of the file to the end of loop() (after calling loop() on file and output). I'm not sure how useful it is though...in my case, I find that I am successfully reading 136 bytes, then getting 282 bytes from my first frame buffer. That lets me fill out the unused 418 bytes of the 1536 byte buffer. 

Below are my latest updates to the Input() method, in case there is something obviously silly I'm doing (it's worth noting that the Serial.printf() doesn't appear to significantly impact the playback...there is a pause with and without it:

```
enum mad_flow AudioGeneratorMP3::Input()
{ 
  int unused = 0;
  
  if (stream->next_frame) {
    unused = buffLen - (stream->next_frame - buff);
    memmove(buff, stream->next_frame, unused);
    stream->next_frame = NULL;
  }
  if (unused == buffLen) {
    // Something wicked this way came, throw it all out and try again
    unused = 0;
  }
  
  lastReadPos = file->getPos() - unused;
  int readLen = buffLen - unused;
  int len = file->read(buff + unused, readLen);
  if (!hasFirstRead) {
    Serial.printf("Copying first frame\n");
    memcpy(firstRead, buff, buffLen);
    hasFirstRead = true;
  }
  if (readLen != len) {
    if (replayOnEnd) {
      memcpy(buff+unused+len, firstRead, buffLen-unused-len);
      Serial.printf("Copied %d bytes from %d to %d. Already added %d from read. Got %d/%d bytes. All %d bytes should be full\n", buffLen-unused-len, firstRead, buff+unused+len, len, buffLen-unused, readLen, buffLen); 
      len = buffLen-unused;
      resetFile = true;
    }
    else if(len == 0){
      Serial.printf_P(PSTR("MP3 stop, len==0\n"));
      return MAD_FLOW_STOP;
    }
  }

  mad_stream_buffer(stream, buff, len + unused);

  return MAD_FLOW_CONTINUE;
}
```

JohannesMTC, 2018-03-12:
I have now times with the proposed 16 bit signed and 44 kHz a try. 
Unfortunately I can only read one file fast enough from the SPIFFS. With two files I get ready to read problems fast enough from the SPIFFS.

At 44kHz and 16-bit, the files become so large that full buffering, even at 0.25 seconds per loop, is no longer possible.

**Of course you are right that you can not describe the external DAC's with 8 bits.**
Have now made some experiments. 16 bit signed with 22 kHz are no problem.

Will try it now with 8 bit 22 kHz raw files. These should be read quickly enough. The conversion to 16 bit signed I do not see problematical, because the mixer gennerell every single sample must multiply and then be added to the other. 
The reduction of the output to 22 kHz doubles the time the buffer can hold.

The reduction of the input files terminates the input buffer by one quarter.

Let's see if that goes out then?

jurecotic, 2018-03-15:
Any news on this? or if somebody could point help me out with an example of code. Basically I'm working on an altimeter and would like to have it call out specific prerecorded sounds at certain altitudes. And none of the examples are working if. I just hear garbled sound.

earlephilhower, 2018-03-15:
If you're getting garbled sound, it's not related to this.  Here people are trying to seamlessly string multiple sounds w/o having a click or noise between them.  I'd check that you're using the right I2S or I2SNoDAC generator, wiring, and setGain(0.4); to reduce volume if you're clipping due to power issues.

rfestag, 2018-03-18:
Following up on this:
I did try dropping down to a 22kHz sample rate, and it didn't seem to make a difference. In my current code:
1) I have native support for looping. At the end of the file (determined by a read returning less than the number of bytes requested), I seek to the start of the data portion of the file (which was set after the ID3 header was completed). There is a very short (but noticeable) pause in playback, with no click.
2) When changing the file, I am using new SPIFFS and ID3 sources, and a new MP3 generator. This setup causes a noticeable click, regardless of whether the second file is played after the first file has finished.

I feel like if I can resolve the first issue (the small pause), I can basically implement a playlist-like setup to handle seamless playback of consecutive files. That does leave open what the appropriate way to handle stopping playback of one file so another can play to completion. Ideally I would continue the original where it left off, but I'd be content with just restarting it.

Any thoughts on good ways to resolve any of these issues? Would it be worthwhile to submit a pull request for the looping playback, even though it isn't quite seamless yet?

earlephilhower, 2018-03-18:
The delay must be algorithmic, since it's not related to the sample rate. My guess would be that libmad is spitting out one frame's worth of 0s at the beginning of decode.

A simple, if a little inelegant, way around this would be, in the looping output to throw away and not push all 0s after the loop is restarted until a non-0 value is passed in.

rfestag, 2018-03-18:
Makes sense. I started looking at the value of lastSample (before it is sent to ConsumeSample), and it looks like there is some sort of preamble for each file before I see "standard" values (during playback, the vast majority of values are in the thousands (positive or negative). However, the first few values sent are always 0, followed by a section of "low values" (generally double digits, sometimes as high as +-300), then gets into the more "expected" values. I'm assuming that section of low values is the click I'm hearing. Is that expected? Or could some metadata (rate info, whatever else is in an MP3) be interpreted as samples and being sent through?

JohannesMTC, 2018-03-27:
Servus Earle!
Since modelers have different sound output requirements than developers of playback devices, I have now developed a simple library that meets these requirements.
The files must be loaded in RAW format 8 bits signet 22 khz into the SPIFFS. Can then be easily started or looped.
Do not see this as a competitor to your library but as an alternative, which I also logically point out in my readme file.
https://github.com/JohannesMTC/ESP32_MAS

zanjie1999, 2018-04-02:
Use
 
ESP.reset();

you can play again

earlephilhower, 2018-04-02:
Just added a link in the README for your project, @JohannesMTC .  Sorry for the delay.  Best of luck!

JohannesMTC, 2018-04-02:
Thank you very much.

imranaalam, 2020-04-18:
@rfestag what is best way to remove clicking sound while playing a playlist?

closed 2020-08-08 by earlephilhower