earlephilhower/ESP8266Audio #69: M4A file format supported?

tueddy, 2018-03-06:
Hi,

Esp8266Audio works fine here on my ESP-32 kidsplayer project, congrats, good work!
Now i see i have to deal with .m4a files coming from iTunes mediathek.
Stupid question:
Is .m4a supported or what would be the way to play it?

Best
Dirk

earlephilhower, 2018-03-06:
I believe .m4a is simply AAC, so try the AAC decoder.  On the ESP-32 it should be able to handle almost anything (on the 8266 SBR is not supported, leading to odd artifacts on some webradios0.

tueddy, 2018-03-06:
Hi Earle,
thank's for your good support! I already tried AudioGeneratorAAC but no sound comes out. Same sketch works fine with .aac file selected from SPIFFS. Here is the .m4a sample. Looking into binary header tells me it's a real M4A soundfile. It plays fine on Windows/MAC. Any idea what's wrong?

[test_m4a.zip](https://github.com/earlephilhower/ESP8266Audio/files/1786678/test_m4a.zip)

earlephilhower, 2018-03-06:
It seems that M4A is a real MPEG-4 stream container (like you'd get over satellite TV or from a BluRay).

You need to extract the AAC track (which may be the only track) before playing as there's not a MP4 demuxer available on the ESP.  FFMPEG or other utility should be able to do it w/o any quality loss since it's just demuxing, not re-encoding.

tueddy, 2018-03-06:
Hi Earle,
thank's for information, i see m4a can contain various info and only a single part might be AAC audio. demuxing, never heard before.. OK, then i have to prepare m4a to aac before playing. I thought it can be done out of the box. Thanks anywhere!

closed 2018-03-06 by tueddy