earlephilhower/ESP8266Audio #70: Support I2S-ADCs/microphones

renne, 2018-03-09:
Hi,

you do a great job bringing I2S-audio to the ESPs! :+1: 

Please, support the I2S-input for I2S-ADCs and I2S-microphones like [SPH0645LM4H](https://www.adafruit.com/product/3421).

earlephilhower, 2018-03-09:
Actually, the Arduino-ESP8266 core team added I2S to the ESP, I just packaged codecs for use.

Adding input support requires changes in the arduino core.  It's not rocket science, but it's not something I've dedicated any thought to.

What kinds of uses would you have for this?  I've been thinking of grabbing a Google AIY box (deeply discounted @ Microcenter) and repurposing the MEMS microphone, but come up blank with ideas as to what could be done.  40KB of usable RAM and slow SPI writes to a small onboard flash feel limiting.

renne, 2018-03-09:
My idea is a multi-purpose MQTT sensor device with the ESP8266 for temperature, humidity, pressure, air quality, microwave motion detection, brightness, color temperature and a RGBWW LED-driver. Adding an I2S microphone and an I2S amplifier would allow bidirectional audio-streaming with a server for e.g. voice-control and audio-notifications.

earlephilhower, 2018-03-10:
For Pi day, Microcenter is selling the AIY board for $3.14.  I couldn't resist at that price.  It turns out it has a stereo I2S microphone hat in addition to the MAX 1-channel DAC.  https://www.invensense.com/wp-content/uploads/2015/02/ICS-43432-data-sheet-v1.3.pdf  

There's also a project on GitHub recording from I2S:  https://github.com/joextodd/listener

renne, 2018-03-11:
Lucky you!
In Germany we have to pay 30,- EUR. That's 240,- EUR for 8 rooms + 280,- EUR for the 8 RPis.

earlephilhower, 2018-03-21:
Check the latest PR I submitted.  It's got some work to do, but if you've got things wired up already I'd like to hear how it goes.  My wire wrap tool has gone missing and I've got to find it before I can wire up that cheap stereo mike...

h3ndrik, 2018-08-18:
still interested in audio input support?

i'm trying to implement this but i'm not sure how to hook this into esp8266audio so this will actually become useful. Also this is neither a AudioGenerator nor a AudioFileSource. Do i need to inherit from something like that for some use cases?

https://github.com/h3ndrik/ESP8266Audio/tree/AudioInputI2S

File is called AudioInputI2S

earlephilhower, 2018-08-20:
That's a good start, @h3ndrik!  Apologies in advance as I've found most of my free hacking time stuck on improving the Arduino core and not the audio bits.

I actually rewrote the I2S subsystem and figured out the (darn near undocumented) 8266 DMA engine to get input working, but never did anything with it.

I'm trying to figure out how this fits in the chain, logically.  I think it could be an AudioGenerator, just one with a `null` input source.  If I add a AudioOutputUlaw class, for example as compression, or an AudioOutputSocket class, for transmission, and an AudioOutputCallback class for the user's own handling of blocks, it may just fit into the framework with less mess.  

Does that seem like it makes sense?  Any other ideas?

h3ndrik, 2018-08-20:
uh. just filed the PR before i read this.
I could reformat this to inherit from AudioGenerator. For now I implemented nearly the same methods.

Is there already an AudioOutputUlaw class or a network output somewhere?

earlephilhower, 2018-08-21:
Thanks for the updated PR.  There is no AudioOutput(U,A)law, yet, but it's one thing I'll need to add as it could easily give 2:1 compression with minimal CPU and quality loss, which is important for the 8266!

I'm really swamped this week an next with travel and meetings, so I don't know if I will be able to look at it in depth and merge until the 1st week of September, but it's definitely on the to-do list!

I also got a note about doing the same thing for bluetooth a2dp (actually I guess you could do an input and an output mode) on the 32, but haven't looked at it at all.  I think the same framework could work as this, which would help prove the AudioGenerator() interface is a usable model.

h3ndrik, 2018-08-22:
Alright. No need to hurry, I'll always have my own forks and branches.

I'm working on RTP (U,A)Law-streaming in the meantime.

Morgaul, 2019-05-24:
Hi. It will be very usefull to have AudioOutputOPUSclass for voice (VoIP) applications...

dmtulsa, 2019-07-17:
I'm doing a project that requires real time bidirectional audio but it doesn't have to be real high quality and one way video 30fps esp32 >> android / ios and would like to use I2s . Just thought you may want too know what others are looking to do. I know its hard work I'm a retired software engineer. But I;m too old and too far behind to to develop anymore. Not even sure the esp32 can handle that.

jamesboblak, 2021-03-19:
I'm interested in getting voice audio into an I2S microphone/ESP32 and sending it realtime to another ESP32/I2S amp.  I have built a transmitter and receiver with SPH0645 and MAX98357A.  They are connecting to each other via wifi, but I still can't get audio out of the speaker.  Did this tread continue elsewhere?  I'm concerned that it may simply not be possible.  Thank you for your incredible work here!

earlephilhower, 2021-03-19:
@jamesboblak for your use, a simple walkie-talkie, this library really isn't going to help since it's all about decompression.

I'd just use the basic Espressif I2S calls for that...on a UDP  packet receive, just write it to the I2S buffer.  And for I2S in, just resend it via UDP when you get notified that one buffer is available.

frankcohen, 2021-03-19:
On my own project I had to be careful with the ESP32 pin connections to the I2S device. ESP32 uses some of the pins I didn’t expect to support Wifi. I found Andreas video to be very helpful: https://www.youtube.com/watch?v=LY-1DHTxRAk

To see my pinouts, check out the Reflections project wiring guide at https://github.com/frankcohen/ReflectionsOS

-Frank

jamesboblak, 2021-03-19:
Thank you so much for your help - I'm very excited to open these links.
I've been trying to get this going for almost a month - hopefully the
search is almost over :)
Thank you again,
-James

frankcohen, 2021-03-19:
Thanks. I’m working on the ShowRunner part now https://github.com/frankcohen/ReflectionsOS/tree/main/ShowRunner. That shows how I run WAV audio files using ESP8266Audio.

I successfully run a 240x240 pixel MJPEG video to a TFT display at the same time I play a WAV audio file. I’m getting 20 frames per second and the audio sounds good. I had to back down the WAV to 8000 kHz 16 bits mono and converted the MJPet to uncompressed files using ffmeg. 

-Frank

jamesboblak, 2021-03-19:
Frank,

I am so inspired by your work (I was starting to worry the hardware wasn't
up to it).  I'm going to take your suggestions immediately for my one-way
walkie talkie.  The ESP32 pin connections video may have solved one of my
wifi connection issues already!  Are there any other ESP32 networking or
I2S resources you recommend?  The ESP8266 resource is incredible.

Cheers,

-James

frankcohen, 2021-03-19:
There are a ton of ESP32 projects. i recommend you join a selection of them related to your project. The water is warm! :-)

jamesboblak, 2021-03-19:
I've only just discovered github.  This summer I'm starting a coding boot
camp at the local university, but at the moment my coding skills are very
basic.  Thank you for the warm welcome, I hope to see you around in the
repositories (if that's how you say it :)
Cheers,
-James

jamesboblak, 2021-03-21:
We have our two ESP32's connecting to each other and transmitting data, but
after a few seconds the Receiver stops receiving the data, a second later
the Transmitter stops sending data, and both systems reboot.  This loop
continues every 5 seconds or so, and there's no audio in the speaker at
all.  I was able to verify the Receiver circuit with the ESP8266 Homer
Simpson I2S example sketch, but we cannot get any audio out with uor script.

I feel as though we are close!  What is the best way to reach out for
community help?  Is that the purpose of GitHub?  Probably someone could
look at our two scripts and point to a new place to start - we're running
out of ideas.  Sorry to bother you!  -James

jamesboblak, 2021-03-21:
Last question, promise!  Are your services available for hire?  Best, -James