earlephilhower/ESP8266Audio #72: forbidden bitrate value with AudioFileSourceSPIRAMBuffer and I2SDAC

mcuadros, 2018-03-15:
Hi,

I am using this excellent library with a MAX98357A and 23lc1024, the wiring is exactly as described in the readme.

When I am trying to reproduce the MP3 using IS2 and `AudioFileSourceSPIRAMBuffer`, I am getting the error: `Decoding error 'forbidden bitrate value' at byte offset 129537, code: 259`

Apparently, the code is correct and works if I use a standard buffer without SPIRAM.

As MP3, I am using: `http://streaming.shoutcast.com/80sPlanet?lang=en-US` from the example.

Thanks.

```
#include <Arduino.h>
#include <Brick.h>
#include "AudioFileSourceICYStream.h"
#include "AudioGeneratorMP3.h"
#include "AudioOutputI2SDAC.h"
#include "AudioFileSourceSPIRAMBuffer.h"

Brick brick("speaker", ESP.getChipId());

AudioOutputI2SDAC *out;
AudioGeneratorMP3 *mp3;
AudioFileSourceICYStream *file;
AudioFileSourceSPIRAMBuffer *buff;

void stop()
{
  if (mp3 == NULL) {
    return;
  }

  mp3->stop();
  buff->close();
  file->close();
  delete mp3;
  delete buff;
  delete file;
  mp3 = NULL;
}

// Called when there's a warning or error (like a buffer underflow or decode hiccup)
void statusCallback(void *cbData, int code, const char *string)
{
  const char *ptr = reinterpret_cast<const char *>(cbData);

  // Note that the string may be in PROGMEM, so copy it to RAM for printf
  char msg[64];
  strncpy_P(msg, string, sizeof(msg));
  msg[sizeof(msg)-1]=0;

  Serial.printf("WARNING(%s): %s, code: %d\n", ptr, msg, code);
  Serial.flush();
}

const float defaultGain = 0.1;
const float maxGain = 0.5;

void handleAction(char* payload)
{
  stop();
  if (!strcmp(payload, "stop")) {
    Serial.printf("MQTT(action): Stop requested\n", payload);
    return;
  }

  const int plen = strlen(payload);
  char *pch = strtok(payload, " ");
  char *url = pch;
  
  float gain = defaultGain;
  if (plen > strlen(url)) {
    char *data = strtok (NULL, " ");
    gain = atof(data);
  }

  if (gain > maxGain) {
    Serial.printf("WARNING(MP3): Invalid gain given [%d], max. value: [%d]\n", 
      gain, maxGain);

    gain = maxGain;
  }

  out->SetGain(gain);

  Serial.printf("MQTT(action): Playing URL [%s] at gain [%.2f]\n", url, gain);

  file = new AudioFileSourceICYStream(url);
  free(url);

  buff = new AudioFileSourceSPIRAMBuffer(file, 4, 131072);
  //buff = new AudioFileSourceBuffer(file, 2048);
  buff->RegisterStatusCB(statusCallback, (void*)"buffer");
  mp3 = new AudioGeneratorMP3();
  mp3->RegisterStatusCB(statusCallback, (void*)"mp3");
  mp3->begin(buff, out);
} 

void setup()
{
  brick.setActionCallback(handleAction);
  brick.setup();

  out = new AudioOutputI2SDAC();
}

void loop()
{
  brick.loop();

  if (mp3 == NULL) {
    return;
  }

  if (mp3->isRunning()) {
    if (!mp3->loop()) {
      stop();
    }
  }
}
```

earlephilhower, 2018-03-15:
As a sanity check, does it work properly (with possible hiccups) for you without using the external RAM?  If so, you'll want to try and get in touch with @armSeb as he's the one who built the SPI buffer unit.

mcuadros, 2018-03-19:
Yep, without works perfect, some cuts, but this is what I am trying to solve with the memory.

earlephilhower, 2018-03-19:
Well, that's kind of good news.  It means the problem is either the HW or SW involved with the RAM chip.

Something simple came up, are you running the exact same RAM size as default?  If not, make sure you're passing in the correct amount in the constructor as the default is a 128K chip in the example code.

If that's not it, the next easiest thing would be to double-check your wiring and see if you can do a simple read and write test, using another test sketch, one that just fills up and reads out the RAM.  If that doesn't work, it's obviously a HW or protocol issue WRT your RAM chip.  If it does work, then it's something in the code at https://github.com/earlephilhower/ESP8266Audio/blob/master/src/AudioFileSourceSPIRAMBuffer.cpp .

I don't have a RAM ship installed myself, so can't do testing.  However, the author of the SPIRAMBuffer code did do testing and was running a streaming setup himself before the code was committed, so you may want to ping him and see if he's seen something like this.

mcuadros, 2018-03-19:
Yes, my chip is exactly `23lc1024`, so its 1024kbits, so yes is the same size. 

I tried https://github.com/Gianbacchio/ESP8266_Spiram/blob/master/examples/ESP_spiram/ESP_spiram.ino, also I reviewed my connection several times, with different chips, different PIN for the C/S but nothing works. It's very strange, it felts like SPIRAM lib isn't working.

armSeb, 2018-03-19:
Hello,

What is the used ESP8266 board ? You have to plug the SPIRAM chip on the same SPI bus as the flash chip.  The wires must be as short as possible.

Can you send a picture of your wiring ?

earlephilhower, 2018-03-19:
Thanks, @armSeb .  I've added a few 1Mbit SRAMs to my cart @ Arrow, but it's going to be a while before I hit the minimum order limit there to try it myself.

Is there a simple test he can run (example from the spiram lib, maybe) to test functionality?

armSeb, 2018-03-20:
You can write then read a few bytes like a string. You have a sample code to do that. I need to verify again there is no error in my schematic.

earlephilhower, 2018-03-21:
@mcuadros Can you try a simple sketch that dumps a few 100s of bytes into the buffer and then pulls them out and verify the read == the written values?  That'd be a simple HW diagnostic re: the wiring (@20-40MHz it gets very interesting.  Wire wrap's better than protoboard but can still be iffy if lengths are not identical or there's crosstalk).

armSeb, 2018-03-21:
As far I know, I set the SPI RAM speed to 40MHz. You can try to change this value to 20 MHz.

I wired the SPIRAM chip with very short wires, not the same length but very short length.

You can use the library example code to test your hardware: 

https://github.com/Gianbacchio/ESP8266_Spiram/blob/master/examples/ESP_spiram/ESP_spiram.ino

mcuadros, 2018-03-21:
Yep, I tried the example and doesn't work. I will try to wire it in a protoboard, since my current setup is soldered, since I already had the board with DAC mounted. I triple checked the wires, but most like will be an error on my side.

armSeb, 2018-05-15:
Hi,

Did you have any news regarding this issue ?

earlephilhower, 2018-05-16:
Howdy @armSeb!  On my end, I still haven't hit the minimum order amount so haven't tried it.  If @mcuadros can't give an update in a little bit, I think I'll close this issue.  If someone else hits it, we can work with them to debug at that time.

lileouskydive, 2018-06-15:
Hye 

your code don't work because you must change the Chip Select pin when you use I2S and SPIRAM in same time. 

Just change 

buff = new AudioFileSourceSPIRAMBuffer(file, 4, 131072);

by 

buff = new AudioFileSourceSPIRAMBuffer(file, 0, 131072); // 0 is D3 on node MCU 

it works well for me after finding that. 

Bye

armSeb, 2018-08-14:
I tried the latest commits on my esp8266 test board with SPI RAM, audio DAC and latest arduino ESP8266 libraries release. It's pretty stable, good work !

closed 2020-08-08 by earlephilhower