earlephilhower/ESP8266Audio #75: AudioFileSourceHTTPStream still not resilent

mcuadros, 2018-03-19:
I am using `AudioFileSourceHTTPStream` to play MP3 under demand through MQTT, and the 99% of the time the request is performed properly but still, I have some minor cases were the board is reset.

I have plans to perform some stress test, to collect the stack traces of this cases.

Are you planning to work more in the `AudioFileSourceHTTPStream` in a short term? My cpp skills are very limited, so I barely can help.

earlephilhower, 2018-03-21:
I've seen I2S itself simply cause hangs after some time.  Can you try the PlayAACFromPROGMEM sample and let it sit after it finishes for a few minutes?  That one was hanging after I'd finished with the I2S interface after maybe 200 seconds.  The Arduino Core I2S stuff is a little wonky, I think.

mcuadros, 2018-03-21:
Yesterday, I was playing and stopping the same sample (2secs) via HTTP stream during 10hours and went, well. 

I guess that the resets are related with running of heap memory. Tomorrow I am planning to do some other stress tests using different MP3, with different encoding, to test the resilience. 

Since the heap memory in the ESP8266 is that small, any other functionality consuming memory will make crash the board.

Another issue I found is that with the memory allocation written in the WebRadio example, doesn't run in my wemos d1, it simply doesn't play anything, until the buffer allocation is changed to 2k.
https://github.com/earlephilhower/ESP8266Audio/blob/master/examples/WebRadio/WebRadio.ino#L236-L237

I will do some other tests with `PlayAACFromPROGMEM` to see if is able to finish the file.