earlephilhower/ESP8266Audio #80: Hissing background noise with NoDAC

renne, 2018-03-27:
Hi,

I'm still stuck with audio output by NoDAC. The music itself sounds much better than expected ([Linn Classic Radio, MP3 320 Kbits/s](http://89.16.185.174:8004/stream)) and I use StreamMP3FromHTTP.

There is annoying hissing background noise. I tried 32 Ohm headphones or a SN74HC245N or Class-D amplifier with a 4 Ohm Speaker connected to RX-pin via a 330 Ohm or 1k Ohm resistor.

When I tried the lowpass presented at [Audio Hacking with the esp8266](https://janostman.wordpress.com/audio-hacking-with-the-esp8266/) with the example code from [Audio Hacking with the esp8266](https://janostman.wordpress.com/audio-hacking-with-the-esp8266/) the hissing was gone. With the 10nF capacitor connected there is no audio output from StreamMP3FromHTTP. The sine wave example works with the 10nF capacitor connected.

Any one having an idea what's causing the hissing noise?

Superchicken, 2018-03-27:
We're having similar experiences, I also tried following the tutorial you've linked and had the same result.

I think Pulse Density Modulation must not be the method used by this library to output an audio signal on the I2S pin. Otherwise a low-pass filter would work, right? 

A DIY hardware DAC using whatever the proper I2S protocol is is probably more complex.

renne, 2018-03-28:
I tried a [STP16NF06L-POWER MOSFET](http://www.produktinfo.conrad.com/datenblaetter/150000-174999/156110-da-01-en-PWRMOS_60V_16A_STP16NF06L_STM.pdf). It sounds awful and is completely distorted.

earlephilhower, 2018-03-28:
This SW implements a sigma-delta 1-bit DAC.  By increasing the oversampling frequency it pushes noise higher in the F domain.  While you *could* add in a real, tuned LPF to the output stage and then drive a standard amplifier, the physics of the 1-T and most cheap, 4cm, 8 ohm speakers is such that it's not bad without it.

You *cannot* stick the output of this into any sort of traditional amplifier.  It bang-bangs between 0 and 3.3V, with no intermediate values, at high frequency.  Traditional line-in is 1.0V.  The input filtering on your amp is the only reason it's able to make any sense of it.

Think of this 1-T output like the mono ADC+AMP from Adafruit.  https://www.adafruit.com/product/3006 .  All the same caveats.

A single capacitor is not a good LPF, and I'd have to SPICE their circuit to see what a 10nf would do, but it's not the right way.  There are online filter design tools where you can specify a cutoff and order and get component values, and if you do that then you can feed it to a standard amp (after DC blocking).  But at that point, it's easier to get an I2S DAC which already had all these parts (and does the sigma-delta internally!).

@renne, Don't use a power MOSFET, please try a plain 3904 NPN BJT transistor.  The small plastic DIP "D' type package can handle the kinds of low wattages it dissipates in normal operation without even getting warm.

It's not studio quality, but it's not terribly noisy, either, in my and other folks' experiences.  Somewhere around a FM radio with a strong signal if I wanted to be unscientific. Not AM with tons of static.

renne, 2018-03-29:
The hissing sounds like sparks of a Tesla coil. I'll try a BC547B/BC547C on the weekend.

msmcmickey, 2018-04-28:
I'm using this library with an ESP8266-01 to provide audio to a remote Part 15 AM transmitter. I used a 10 mH choke from an old DSL as a simple filter to remove unwanted signal above the audio range. Prior to adding the choke, I had a lot of issue with noise in the audio. The choke pretty much eliminated the problem. Video of results: https://www.youtube.com/watch?v=TzoXVvrIzeg&feature=youtu.be

closed 2020-08-08 by earlephilhower