earlephilhower/ESP8266Audio #81: Support TI TAS5756m I2S power amplifier IC

renne, 2018-03-28:
The [TI TAS5756M](http://www.ti.com/product/TAS5756M/technicaldocuments) is an I2S power amplifier IC commonly used on Raspberry Pi audio amplifier (p)HATs. It is controlled via I2C and provides quite good sound quality.

As my IQAudIO Pi-DigiAmp+ was quite expensive I have compiled a list of currently available implementations:

- [DIY (Phyx PCB Layout)](https://github.com/phyx-be/RPI_HAT_AMP)
- [JustBoom Amp Zero pHAT for the Raspberry Pi Zero](https://www.justboom.co/product/justboom-amp-zero-phat/) ~29,- €
- [HiFiBerry Amp2](https://www.hifiberry.com/shop/boards/hifiberry-amp2/) ~45,- €
- [IQAudIO Pi-DigiAmp+](http://iqaudio.co.uk/hats/9-pi-digiamp-0712411999650.html) ~65,- €
- [JustBoom Amp HAT for the Raspberry Pi](https://www.justboom.co/product/justboom-amp-hat/) ~69,- €

earlephilhower, 2018-03-28:
Those are pretty hard to come by in the US, unfortunately.  The chips themselves are only in the $5USD qty 1 range, but I suppose the analogs to go along with it and support 40W output (and I love using the PCB copper as the heatsink in these hats, very cool!) are pricey.

If you're handy with C++, I can point you in the direction you'd need to go to get these working.

It looks like these are I2C DACs that need to be programmed to handle specific formats (as opposed to the lower-cost ones which figure it out from WCLK and apply proper filters/etc. automatically) and to control speaker gain.

If so, then what you need to do is create and AudioOutput class that instantiates the standard I2S class and passes through data writes (ConsumeSample), but which wraps calls like SetFreq() or SetGain() and converts those into I2C commands to fire to the chip (and also pass the required ones like SetFreq() or SetChannels() along to the internal I2S object to ensure they are obeyed).  If there's a C sample that plays a WAV using raw I2C on the RPI, it should have everything you need.  You can add add'l member functions for things not envisioned in the standard one, like maybe "SetMute()" or "SetAudioTransform()."

That said, I think you may be connecting a gold-plated amp to an FM radio.  While it's possible to send 24-bits/sample, we're only doing 16 to save memory (24bpp uses 2x the RAM, gives 1/2 the buffer time before you get underflow).  The I2S clock generator is only accurate to within +/- 2-3%.  libMAD has been compiled to favor speed over bit accuracy.  And so on.  Probably not worth the grief...

renne, 2018-03-29:
Yesterday I got a ESP32. ;-)
I stumbled upon the [STM STA333W](http://www.st.com/resource/en/datasheet/sta333w.pdf). I ordered a [13$ power-amplifier at Aliexpress](https://www.aliexpress.com/item/DC16V-20W-20W-CM108-USB-I2S-STA333W-pure-digital-power-amplifier-board-With-USB-sound-card/32816241272.html). It will be delivered in 4-8 weeks. It's basically a [CM108](http://www.qsl.net/om3cph/sb/CM108_DataSheet_v1.6.pdf) soundcard with a [STA333W](http://www.st.com/resource/en/datasheet/sta333w.pdf) attached to the I2S output. I'll try to get to the I2S traces.
For now I'll get the [Adafruit I2S MEMS Mikrofon-Breakout - SPH0645LM4H](https://learn.adafruit.com/adafruit-i2s-mems-microphone-breakout/)  and the [Adafruit I2S 3W Class D Amplifier Breakout - MAX98357A](https://learn.adafruit.com/adafruit-max98357-i2s-class-d-mono-amp).

renne, 2018-04-28:
I received the [13$ power-amplifier at Aliexpress](https://www.aliexpress.com/item/DC16V-20W-20W-CM108-USB-I2S-STA333W-pure-digital-power-amplifier-board-With-USB-sound-card/32816241272.html) today. It's not as linear as my AVR with room/speaker correction but sounds quite well. Instead of a STA333W it uses a [STA333BW](http://www.st.com/resource/en/datasheet/sta333bw.pdf) which can be configured either as stereo amplifier + stereo line-out or 2.1 amplifier. The I2C registers are documented.

closed 2020-08-08 by earlephilhower