earlephilhower/ESP8266Audio #82: Static noise using AudioGeneratorMP3 and AudioOutputI2SDAC

mcuadros, 2018-04-01:
Hi,

Using an ultra low quality MP3, produce randomly static noise, different restart of the same board with the same code, produce the static noise.

The file it self is a 440Hz tone, with the following format: `MPEG ADTS, layer III, v2,  32 kbps, 22.05 kHz, Monaural`

https://gist.github.com/mcuadros/ca53b9cbf38c998dddabb597d2abb24c

```
#include <Arduino.h>
#include "AudioGeneratorMP3.h"
#include "AudioOutputI2S.h"
#include "AudioFileSourcePROGMEM.h"
#include "sampleaac.h"

AudioFileSourcePROGMEM *in;
AudioGeneratorMP3 *mp3;
AudioOutputI2S *out;

void setup()
{
  Serial.begin(115200);
  
  in = new AudioFileSourcePROGMEM(tone_mp3, sizeof(tone_mp3));
  mp3 = new AudioGeneratorMP3();
  out = new AudioOutputI2S();
  out->SetGain(0.1);
  out->SetOutputModeMono(true);

  mp3->begin(in, out);
}

void loop()
{
  if (mp3->isRunning()) {
    mp3->loop();
  } else {
    Serial.printf("MP3 done\n");
    delay(1000);
  }
}
```

Thanks for your continuous efforts.

earlephilhower, 2018-04-01:
Can't seem to reproduce it, sorry.  I just tried it on my own Wemos D1 and on the host-based tester.  Both played fine (like the old TV emergency broadcast warning) through multiple uploads and after hitting the reset button several times.

There does seem to be something fishy in the end-of-mp3 detection this hits and I seem to end up with a loop where it prints MP3 stop, len==0 without actually stopping, but that produces a clicking-type sound (due to underflow) and it's repeatable, not static at all.

kosso, 2018-06-01:
I'm getting the 'clicking' sound while the audio is playing (from a live radio stream). Are there any values I can increase to deal with this?

earlephilhower, 2018-06-01:
Sounds like TCP underflow.  You should try increasing the buffer, @kosso.  It's minimized on the ESP8266 since there is so little space available, but having a much larger one is possible on the ESP32.

closed 2020-05-11 by earlephilhower