earlephilhower/ESP8266Audio #83: No playback when "cloning example"

mc9625, 2018-04-04:
I've modified the "PlayMP3FromSPIFFS" example file, changing the AudioOutputI2SNoDAC -> AudioOutputI2S . Everything works fine.

As soon as I copy the example code within another project, I get no sound at all.

Now my project is rather complex, so I think that paste everything here would result confusing. 
Basically I'm working with an ESP8266, with an ADAFRUIT I2SDAC. The project use WifiManager and other libraries (mainly fastled.h). 

If I do run the modified example code it works, as soon as I try to import the code into my own project, i get no sound.

The problem is I have no clue on how to troubleshoot this issue. The serial monitor doesn't give me any hint. I can clearly see the "Sample MP3 playback begins..." message, and even the "MP3 done" at the end, but no sound.

I've included all the code from the example exactly how it is in the original code, respecting the setup and loop and include parts.

I know this request may seems to vague as it is now, but if you could help me reduce the "field" I could post some of the involved code, or at least look for some kind of error message or something.

Thanks

earlephilhower, 2018-04-04:
Howdy,

It could be an out of memory error, or a lack of SPIFFS upload (FWIW I think I've seen the IDE not duplicate data/* files). 

For out of memory, check the return value of mp3->begin().   If it's false, that's the most likely reason (assuming the file exists).  My samples are not very good in terms of checking return values or cleaning up memory after-the-fact (i.e. delete mp3, etc.), unfortunately, and I assume folks know about this.

mc9625, 2018-04-04:
Hi, 

the mp3-begin() doesn't report any value.

I'm sure the file is there because if I simply upload the example code, it plays.

Here is the code I've inserted in the "setup" part of my project (it's exactly as in the example code, I've simply modified the AudioOutputI2S on both code, and changed the name of the mp3 file, again the example code works fine, my project doesn't).

`void setup()
{
  WiFi.mode(WIFI_OFF); 
  Serial.begin(9600);
  delay(1000);
  SPIFFS.begin();
  Serial.printf("Sample MP3 playback begins...\n");
  file = new AudioFileSourceSPIFFS("/quite-impressed.mp3");
  id3 = new AudioFileSourceID3(file);
  id3->RegisterMetadataCB(MDCallback, (void*)"ID3TAG");
  out = new AudioOutputI2S();
  mp3 = new AudioGeneratorMP3();
  mp3->begin(id3, out);
}

void loop()
{
  if (mp3->isRunning()) {
    if (!mp3->loop()) mp3->stop();
  } else {
    Serial.printf("MP3 done\n");
    delay(1000);
  }
}`

earlephilhower, 2018-04-05:
mp3->begin returns a bool.  Add a check and dump the value, it's probably returning false because of OOM.

closed 2020-08-08 by earlephilhower